home-assistant/operating-system #2143: 8.X 9.X problems with multiple bluetooth adapters

50494554524F, 2022-09-18:
### Describe the issue you are experiencing

when i add an asus bt-400 to the raspberry it can be recognized (not always) and go on hci1, ok.

if i try to reboot the host then it goes on hci0, so things are messed up.

if i accept to use it on hci0 bluetooth devices turn unavailable after a while, i.e. a switchbot bot.

(could be useful reinstalling the os from scratch?!)

thanks

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.X 9.X

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. boot without the asus bt-400
2. insert the asus bt-400
3. if recognized, select hci1 from the bluetooth integration
4. reboot the host
5. bluetooth integration now gives an error

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

agners, 2022-09-19:
Linux does not have a guaranteed order of devices, the device names get assigned whatever comes first, and it seems that in your system that ends up to be random :disappointed: 

If you don't use the internal Bluetooth, probably disabling it is the best approach. You have to edit the `config.txt` file in the first partition of the SD card, add the following at the end of that file:

```
dtoverlay=disable-bt
```

50494554524F, 2022-09-19:
> Linux does not have a guaranteed order of devices, the device names get assigned whatever comes first, and it seems that in your system that ends up to be random 😞
> 
> If you don't use the internal Bluetooth, probably disabling it is the best approach. You have to edit the `config.txt` file in the first partition of the SD card, add the following at the end of that file:
> 
> ```
> dtoverlay=disable-bt
> ```

it's not recognized after reboot, the two adapters conflicts... this is a bug to me... btw, thanks for the tip (i'm not a linux expert yet), i'm going to try to disable the internal bluetooth

ioctl2, 2022-10-16:
> Linux does not have a guaranteed order of devices, the device names get assigned whatever comes first, and it seems that in your system that ends up to be random 😞

Perhaps it would be possible to create sticky names on first run / device install?

closed 2023-01-21 by github-actions[bot]