home-assistant/operating-system #2144: RPi 3 Upgrade from 8.5 to 9.0 fails

Uthrom, 2022-09-19:
### Describe the issue you are experiencing

Trying to upgrade HAOS from 8.5 to 9.0 from GUI seems to time out.

When running it from command line, I get the following:

➜  backup ha os update --no-progress --log-level debug
DEBU[0000] Adding homedir to searchpath                  homedir=/root
INFO[0000] No configfile found                          
DEBU[0000] Debug flags                                   apiToken=f8daa3debc3beebcd62b3ef5b3ee751064150d456e345144f4364dd5d6f781ade7352c242e00345f84bee288ee274f884a1359ffbfd7af44 cfgFile= endpoint=supervisor logLevel=debug noProgress=true rawJSON=false
DEBU[0000] os update                                     args="[]"
DEBU[0000] [GenerateURI]                                 base=supervisor command=update section=os
DEBU[0000] [GenerateURI] Result                          uri="http://supervisor/os/update" url="http://supervisor/os/update" url(string)="http://supervisor/os/update"
DEBU[2305] Response                                      body="{\"result\": \"error\", \"message\": \"Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/9.0/haos_rpi3-64-9.0.raucb: Response payload is not completed\"}" fields.time=38m25.219007174s headers="map[Content-Length:[192] Content-Type:[application/json; charset=utf-8] Date:[Sun, 18 Sep 2022 23:38:06 GMT] Server:[Python/3.10 aiohttp/3.8.1]]" received-at="2022-09-18 18:38:06.549179766 -0500 CDT m=+2305.234044017" request="&{POST http://supervisor/os/update HTTP/1.1 1 1 map[Accept:[application/json] Authorization:[Bearer f8daa3debc3beebcd62b3ef5b3ee751064150d456e345144f4364dd5d6f781ade7352c242e00345f84bee288ee274f884a1359ffbfd7af44] User-Agent:[go-resty/2.7.0 (https://github.com/go-resty/resty)]] <nil> 0x2a4a30 0 [] false supervisor map[] map[] <nil> map[]   <nil> <nil> <nil> 0x4000144040}" status="400 Bad Request" statuscode=400
Error: Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/9.0/haos_rpi3-64-9.0.raucb: Response payload is not completed
➜  backup 

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. run the command: ha os update --no-progress --log-level debug
2. Wait
3. Observe above error message
...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-09-18 17:24:05 INFO (MainThread) [supervisor.api.middleware.security] /core/restart access from a0d7b954_ssh
22-09-18 17:24:05 INFO (SyncWorker_2) [supervisor.docker.interface] Restarting ghcr.io/home-assistant/raspberrypi3-64-homeassistant
22-09-18 17:24:59 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
22-09-18 17:25:02 INFO (MainThread) [supervisor.host.info] Updating local host information
22-09-18 17:25:05 INFO (MainThread) [supervisor.host.services] Updating service information
22-09-18 17:25:06 INFO (MainThread) [supervisor.host.network] Updating local network information
22-09-18 17:25:08 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-09-18 17:25:09 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-09-18 17:25:43 INFO (MainThread) [supervisor.addons] Add-on 'a0d7b954_nodered' successfully updated
22-09-18 17:25:43 INFO (SyncWorker_6) [supervisor.docker.interface] Cleanup images: ['ghcr.io/hassio-addons/node-red/aarch64:13.3.2']
22-09-18 17:25:43 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-09-18 17:25:45 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-09-18 17:25:51 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-09-18 17:25:51 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/fredrikbaberg/hassio-addons repository
22-09-18 17:25:51 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
22-09-18 17:25:51 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
22-09-18 17:25:52 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
22-09-18 17:26:02 INFO (MainThread) [supervisor.store] Loading add-ons from store: 75 all - 0 new - 0 remove
22-09-18 17:26:02 INFO (MainThread) [supervisor.store] Loading add-ons from store: 75 all - 0 new - 0 remove
22-09-18 17:28:40 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
22-09-18 17:32:48 INFO (MainThread) [supervisor.backups.manager] Found 7 backup files
22-09-18 17:32:48 INFO (MainThread) [supervisor.backups.manager] Found 7 backup files
22-09-18 17:33:02 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-09-18 17:33:10 INFO (MainThread) [supervisor.backups.manager] Removed backup file 5bb36f2e
22-09-18 17:33:10 INFO (MainThread) [supervisor.backups.manager] Removed backup file ca8d9472
22-09-18 17:33:10 INFO (MainThread) [supervisor.backups.manager] Removed backup file 5472a50b
22-09-18 17:33:10 INFO (MainThread) [supervisor.backups.manager] Removed backup file 6cb06c96
22-09-18 17:33:10 INFO (MainThread) [supervisor.backups.manager] Removed backup file 0a8ea333
22-09-18 17:33:11 INFO (MainThread) [supervisor.backups.manager] Found 2 backup files
22-09-18 17:33:16 INFO (MainThread) [supervisor.backups.manager] Found 2 backup files
22-09-18 17:33:17 INFO (MainThread) [supervisor.backups.manager] Found 2 backup files
[... 182 lines cut ...]
```

### System Health information

> Optional Copy the full System Health in this text area.
> Can be found in the [Configuration panel -> Settings -> Info](https://my.home-assistant.io/redirect/info/). Use the copy icon on top right and choose For GitHub.

I can't find the icon in question.

### Additional information

running wget https://github.com/home-assistant/operating-system/releases/download/9.0/haos_rpi3-64-9.0.raucb from the commandline does appear to download the file in question after a 302 redirect:

➜  ~ wget https://github.com/home-assistant/operating-system/releases/download/9.0/haos_rpi3-64-9.0.raucb
--2022-09-18 18:46:56--  https://github.com/home-assistant/operating-system/releases/download/9.0/haos_rpi3-64-9.0.raucb
Resolving github.com (github.com)... 140.82.114.4
Connecting to github.com (github.com)|140.82.114.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/115992009/b3e59470-dbe7-40ff-8e51-48100b2bd01d?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220918%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220918T234657Z&X-Amz-Expires=300&X-Amz-Signature=789a8a3ddd15654f18f34399d39ce6a62a228e39cd276d427fb83aeb6d9b1e26&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=115992009&response-content-disposition=attachment%3B%20filename%3Dhaos_rpi3-64-9.0.raucb&response-content-type=application%2Foctet-stream [following]
--2022-09-18 18:46:58--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/115992009/b3e59470-dbe7-40ff-8e51-48100b2bd01d?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220918%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220918T234657Z&X-Amz-Expires=300&X-Amz-Signature=789a8a3ddd15654f18f34399d39ce6a62a228e39cd276d427fb83aeb6d9b1e26&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=115992009&response-content-disposition=attachment%3B%20filename%3Dhaos_rpi3-64-9.0.raucb&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 123381132 (118M) [application/octet-stream]
Saving to: 'haos_rpi3-64-9.0.raucb'

Uthrom, 2022-09-19:
Using wget, both form my machine and the RPi, works (as shown in the additional information).

I tried the upgrade several times throughout the day, as it took almost 30 minutes to fail each time.

agners, 2022-09-19:
Is internet connectivity considered good according to `ha network info` (`host_internet` and `supervisor_internet`?

Uthrom, 2022-09-19:
> Is internet connectivity considered good according to `ha network info` (`host_internet` and `supervisor_internet`?

➜  ~ ha network info | grep internet
host_internet: true
supervisor_internet: true

I'm kicking off another update, I'll report back what happens.