home-assistant/operating-system #2147: HA OS Hyper-V Boot Issue on VM

PablaV, 2022-09-19:
### Describe the issue you are experiencing

Was having issues boot up my VM running HA OS every since the v9.0 update. It would either get stuck on ` a start job is running for docker application container engine` forcing me to reboot the VM then eventually it would boot using Slot B Rescue Shell and then get stuck here.
[image]

This has happened twice to me, both times I have had to disable Hyper-V reboot the host then enable Hyper-V reboot then HA would boot up fine. Attached some logs of the Host hopefully provides some more insight, 

Host: Windows 11 machine
Virtual Box v6.1.38
Home Assistant: 2022.9
HA Core: 9.0

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Shutdown VM
2. Attempt to start VM
3. Disable Hyper-V and reboot host
4. Enable Hyper-V and reboot host
5. Successfully start VM

### Anything in the Supervisor logs that might be useful for us?

```txt
None
```

### Anything in the Host logs that might be useful for us?

```txt
None
```

PablaV, 2022-09-20:
Just a small update, seems like the trick of disabling then enabling Hyper-V did not do the trick this morning. What seems to be more consistently working is constantly power cycling the VM until it boots up correctly. I will likely revert back to Core V8.X until a fix is released

agners, 2022-09-21:
@ioctl2 recently improved OVA slightly, maybe the latest dev build from https://os-builds.home-assistant.io/10.0.dev20220919/ works?

PablaV, 2022-09-21:
Never installed a test build, how would I go about that?

ioctl2, 2022-09-22:
> Never installed a test build, how would I go about that?

Since it looks like you are using VirtualBox with the Hyper-V API backend: There is an [`ova`](https://os-builds.home-assistant.io/10.0.dev20220919/haos_ova-10.0.dev20220919.ova) appliance file that you can download via the [link that @agners provided](https://os-builds.home-assistant.io/10.0.dev20220919/). As a first step, you can import it into VirtualBox (File -> Import Appliance) without replacing your existing VM to test whether it works in your setup in isolation from the existing VM.

You should be aware that running VirtualBox on a Hyper-V-enabled host is an [experimental feature](https://docs.oracle.com/en/virtualization/virtualbox/6.1/admin/KnownIssues.html#ExperimentalFeatures) in VirtualBox 6.1 and, in more general terms, comes with a [performance penalty](https://docs.oracle.com/en/virtualization/virtualbox/6.1/admin/AdvancedTopics.html#hyperv-support).

PablaV, 2022-09-22:
> Since it looks like you are using VirtualBox with the Hyper-V API backend: There is an [`ova`](https://os-builds.home-assistant.io/10.0.dev20220919/haos_ova-10.0.dev20220919.ova) appliance file that [...]

So I was able to successfully boot this dev build up fine. How would I update my existing HA VM to this build? I am not concerned about losing anything at this point since I am temporarily running HA on VMware until I can figure this all out. 

As for Hyper-V I did read that it is experimental, but I had it disabled and it was still causing issues. Only when these issues came about I tried enabling/disabling in hopes that it would help.

Merangle, 2022-11-14:
Hello 

My request for the issue #78705 has been closed and transfered to this issue. 
I'm still stucked each time I have to restart my Synology. 
How can I help ? 
Thanks

PablaV, 2022-11-14:
Unfortunately I was not able to get things working on VirtualBox. I ended up switching to VMware Workstation 16 which booted up with no issues. After that I just ended up purchasing an ODROID since I didn't wanna run into any future issues with VM. One thing that worked for me was constantly booting up the VM then restarting it everytime it hangs, eventually it boots up. From there create and download a back up just in case you need to switch hosts.

closed 2023-05-22 by github-actions[bot]