home-assistant/operating-system #2148: Fresh install of HAOS 9 fails first boot

aaronouthier, 2022-09-21:
### Describe the issue you are experiencing

After writing the latest image (as of this writing) direct to a 500 GB HDD, and rebooting, system hangs with message:

Waiting for root device PARTUUID=8d3d53e3-6d49-4c38-aff6859e82fd…

I’ve tried this same procedure on 2 other machines with the same issue. Same UUID also.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Download image for x64 generic, and verify checksum.
2. Uncompress image and write to blank 500 GB HDD via Linux Live boot disk.
3. Reboot system, and wait for ha prompt.
4. Be annoyed when system hangs for 3 hours without ha prompt.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Unable to grab logs, due to unbootable systems.
```

### Anything in the Host logs that might be useful for us?

```txt
No.
```

agners, 2022-09-21:
Likely a driver missing. What type/make of system is this? What type of HDD?

aaronouthier, 2022-09-21:
HP Compaq 100 SFF PC. Circa early 2012. 500 GB SATA HDD.
I’ll check the MoBo chipset in the morning when I’m up. It’s almost 5 AM now.

aaronouthier, 2022-09-21:
So, the drivers are definitely present, as I can see ata1 and ata2 messages, including SATA link up 1.5 Gbps. However, both the SATA HDD (ATA1] and the DVD RW drive (ATA2) are showing identical errors, including failed to IDENTIFY and qc timeout errors.

Although A driver seems to be present, it may be buggy. I’m going to try switching the BIOS to emulate PATA drives, instead of SATA.

aaronouthier, 2022-09-21:
Booting from a live boot disc, I have listed all UUIDs of the drive with blkid.

Looks like someone tried to specify the UUID by hand, and missed part of the UUID when creating the image!

The only partition UUID that comes remotely close is is the 3rd partition, type squashfs, with a UUID of:
8d3d53e3-6d49-4c38-8349-aff6859e82fd

The PARTUUID= statement is missing 4 hex digits “8349” before the last dash!

aaronouthier, 2022-09-21:
NVM!
I’m the bozo that typed the UUID by hand in my earlier post. UUID seems fine.

aaronouthier, 2022-09-21:
Ok. I’ve now tried other OSs with similar results. Time to visit the recycler…

closed 2022-09-21 by aaronouthier