home-assistant/operating-system #2151: HAOS 9.0 - DHCP not assigning the proper IP when USB modem plugged in

Bieniu82, 2022-09-21:
### Describe the issue you are experiencing

HA installed with SMS integration via GSM modem.
After OS9.0 update HA is not provided the IP address with USB modem plugged in upon rebooting the device.
With the modem plugged out and rebooted it works fine, after that, having the modem plugged in again - it recognizes the modem properly, the integration works fine.
After that, an attempt to restart the device results with lost IP again.

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Reboot the device.

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't access HA over network.
```

### Anything in the Host logs that might be useful for us?

```txt
Can't access HA over network.
```

agners, 2022-09-24:
Did that work before HAOS 9.0?

Bieniu82, 2022-09-24:
> Did that work before HAOS 9.0?

precisely, first reboot after 9.0 upgrade it happened

closed 2022-12-30 by github-actions[bot]