home-assistant/operating-system #2153: when I upgrade  to Operating System 9.0 sata disk on ODROID-XU4 not detect

MihaT2021, 2022-09-23:
### Describe the issue you are experiencing

before Operating System 9.0  I was able to use ssd disk. After I update o.s. to 9.0 it did not boot . I was able to install on SD card again (o.s. 9.0 ) but disk is not veasible.

from terminal:
[core-ssh ~]$ df
Filesystem           1K-blocks      Used Available Use% Mounted on
overlay               29741340  11146824  17346944  39% /
/dev/mmcblk0p4        29741340  11146824  17346944  39% /media
/dev/mmcblk0p4        29741340  11146824  17346944  39% /config
devtmpfs                964396         0    964396   0% /dev
tmpfs                  1014060         0   1014060   0% /dev/shm
/dev/mmcblk0p4        29741340  11146824  17346944  39% /data
/dev/mmcblk0p4        29741340  11146824  17346944  39% /addons
/dev/mmcblk0p4        29741340  11146824  17346944  39% /backup
/dev/mmcblk0p4        29741340  11146824  17346944  39% /ssl
/dev/mmcblk0p4        29741340  11146824  17346944  39% /share
/dev/mmcblk0p4        29741340  11146824  17346944  39% /run/audio
tmpfs                   405624      3608    402016   1% /run/dbus
/dev/mmcblk0p4        29741340  11146824  17346944  39% /etc/asound.conf
/dev/mmcblk0p4        29741340  11146824  17346944  39% /etc/hosts
/dev/mmcblk0p4        29741340  11146824  17346944  39% /etc/resolv.conf
/dev/mmcblk0p4        29741340  11146824  17346944  39% /etc/hostname
tmpfs                  1014060         0   1014060   0% /dev/shm
/dev/mmcblk0p4        29741340  11146824  17346944  39% /etc/pulse/client.conf
tmpfs                  1014060         0   1014060   0% /proc/asound
devtmpfs                964396         0    964396   0% /proc/keys
devtmpfs                964396         0    964396   0% /proc/timer_list
tmpfs                  1014060         0   1014060   0% /proc/scsi
tmpfs                  1014060         0   1014060   0% /sys/firmware

[core-ssh ~]$ lshw
core-ssh
    description: ARMv7 Processor rev 3 (v7l)
    width: 256 bits
    capabilities: smp
  *-core
       description: Motherboard
       physical id: 0
     *-memory
          description: System memory
          physical id: 0
          size: 1980MiB
     *-cpu:0
          physical id: 1
          bus info: cpu@0
          size: 800MHz
          capacity: 1400MHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm cpufreq
     *-cpu:1
          physical id: 2
          bus info: cpu@1
          size: 800MHz
          capacity: 1400MHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm cpufreq
     *-cpu:2
          physical id: 3
          bus info: cpu@2
          size: 800MHz
          capacity: 1400MHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm cpufreq
     *-cpu:3
          physical id: 4
          bus info: cpu@3
          size: 800MHz
          capacity: 1400MHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm cpufreq
     *-cpu:4
          physical id: 5
          bus info: cpu@4
          size: 2GHz
          capacity: 2GHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm cpufreq
     *-cpu:5
          physical id: 6
          bus info: cpu@5
          size: 2GHz
          capacity: 2GHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm cpufreq
     *-cpu:6
          physical id: 7
          bus info: cpu@6
          size: 2GHz
          capacity: 2GHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm cpufreq
     *-cpu:7
          physical id: 8
          bus info: cpu@7
          size: 2GHz
          capacity: 2GHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm cpufreq
  *-mmc0
       description: MMC Host
       physical id: 1
       logical name: mmc0
     *-device
          description: SD Card
          product: USD00
          vendor: Unknown (173)
          physical id: b368
          logical name: /dev/mmcblk0
          version: 1.0
          date: 06/2021
          serial: 306317747
          capabilities: sd
  *-mmc1
       description: MMC Host
       physical id: 2
       logical name: mmc1

Disk is ok I put it to windows  and it is weasible..

alo so Led on odroid for disk is on (disk is connected).

please help.

### What operating system image do you use?

odroid-xu4 (Hardkernel ODROID-XU4)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.ugrade to os. 9.0
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
.
```

### Anything in the Host logs that might be useful for us?

```txt
.
```

### System Health information

.

agners, 2022-09-24:
You are connecting the SATA SSD through USB 3.0 correct? Is the USB port working otherwise (e.g. USB Zigbee dongle or another device?). When downgrading to 8.5, does that make the USB SATA SSD work again?

MihaT2021, 2022-09-24:
I use ODROID-HC1 
https://www.hardkernel.com/shop/odroid-hc1-home-cloud-one/

As you see this odroid have sata connector onbord. I do not use USB disk.

The basic difference:

1.) When I download image and install on SDcard :
https://github.com/home-assistant/operating-system/releases/download/9.0/haos_odroid-xu4-9.0.img.xz. 

Disk SSD SATA is not recognize on home assistent

2.) When I download image 8.5 and install on SDcard.
https://github.com/home-assistant/operating-system/releases/download/8.5/haos_odroid-xu4-8.5.img.xz

Disk SSD SATA is recognize on home assistent.

agners, 2022-10-03:
Unfortunately, I don't have such a device here, so I can test :(

We only changed minor kernel versions, so it is a bit surprising to me that this stopped working.

Can you run `dmesg` on both OS versions and share? That might give a hint what is going on.

MihaT2021, 2022-10-03:
Hire are files with dmsg results:

[dmsg_os_90.log](https://github.com/home-assistant/operating-system/files/9699537/dmsg_os_90.log)
[dmsg_os_85.log](https://github.com/home-assistant/operating-system/files/9699948/dmsg_os_85.log)

MihaT2021, 2022-10-03:
9.0 again:
[dmsg_os_90-v2.log](https://github.com/home-assistant/operating-system/files/9700050/dmsg_os_90-v2.log)

MihaT2021, 2022-10-11:
Today I update to 9.2 .. Sata disk is not recognize.
[os_9.2.log](https://github.com/home-assistant/operating-system/files/9759040/os_9.2.log)

MihaT2021, 2022-10-17:
Hi,

@agners  did you find in my files something usefull?

regards

agners, 2022-10-17:
It seems that HC1 is using an on-board USB to S-ATA adapter, this is from the HAOS 8.5 log (Linux 5.15.60).
```
[    5.428970] usb 4-1: new SuperSpeed USB device number 2 using xhci-hcd
[    5.436950] samsung-i2s 3830000.i2s-sec: DMA channels sourced from device 3830000.i2s
[    5.463189] usb 4-1: New USB device found, idVendor=152d, idProduct=0578, bcdDevice= 1.05
[    5.470041] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    5.477068] usb 4-1: Product: USB to SATA bridge
[    5.481695] usb 4-1: Manufacturer: JMicron
[    5.485708] usb 4-1: SerialNumber: DB00000000013B
```

In HAOS 9.0 (with Linux 5.15.67) this is missing.

As the first line is already missing, it seems the whole controller does not get initialized :confused: 

There was one change, and that was resetting the Ethernet properly (see https://github.com/home-assistant/operating-system/pull/2108). But from what I can tell, Ethernet is totally unrelated with the S-ATA part of that board, so I doubt that this is the culprit. 

I'll create a build with that patch reverted, let's see if that helps fix the problem.

Can you also give the latest nightly a try?
https://os-builds.home-assistant.io/10.0.dev20221013/

MihaT2021, 2022-10-17:
Ok i try it 

 Home Assistant 2022.10.4 Supervisor 2022.10.0 Operating System 10.0.dev20221013 Frontend 20221010.0 - latest 

Disk is not recognize.

[image]

agners, 2022-10-17:
Thanks for testing!

> I can't install ssh terminal.

Hm, that is probably unrelated, but I am not sure.

Can you try this image: https://os-builds.home-assistant.io/haos_odroid-xu4-10.0.dev20221017.img.xz

MihaT2021, 2022-10-17:
[image]

MihaT2021, 2022-10-17:
Home Assistant 2022.10.4 Supervisor 2022.10.0 Operating System 10.0.dev20221017 Frontend 20221010.0 - latest 

 
Same, disk is not found  and also addon ssh terminal have same issue installing.

[supervisor.log](https://github.com/home-assistant/operating-system/files/9802805/supervisor.log)

agners, 2022-10-17:
Ok, that confirms that the only HAOS change is *not* the culprit. It must be a stable kernel regression then.

I've uploaded another image, can you give this a try again?
https://os-builds.home-assistant.io/haos_odroid-xu4-10.0.dev20221017-2.img.xz

MihaT2021, 2022-10-17:
last image it's WORKING!
:-)
disk is recognize.and also addon ssh terminal is working.

 Home Assistant 2022.10.4 Supervisor 2022.10.0 Operating System 10.0.dev20221017 Frontend 20221010.0 - latest 

[v-2.log](https://github.com/home-assistant/operating-system/files/9804152/v-2.log)

agners, 2022-10-17:
Ok, that proves it is a Linux kernel regression between v5.15.61 and v5.15.67. Unfortunately, I had to delete all the development builds from back then :cry: 

Going back in kernel version is not an option. We need to find the exact change. I have found two commit which might explain the problem: 
```
55d328924347 usb: dwc3: disable USB core PHY management
9d6a2d022010 usb: dwc3: fix PHY disable sequence
```

This image uses v5.15.67 with the two suspected commits reverted:
https://os-builds.home-assistant.io/haos_odroid-xu4-10.0.dev20221017-3.img.xz

MihaT2021, 2022-10-18:
[haos_odroid-xu4-10.0.dev20221017-3.img.xz](https://os-builds.home-assistant.io/haos_odroid-xu4-10.0.dev20221017-3.img.xz)

disk is recognize.SATA disk is working. And addon ssh terminal is NOT working.

agners, 2022-10-18:
And one more, this isolates it to a single commit, I assume the culprit is this change

```
55d328924347 usb: dwc3: disable USB core PHY management
```

Can you give this a try?
https://os-builds.home-assistant.io/haos_odroid-xu4-10.0.dev20221018.img.xz

> And addon ssh terminal is NOT working.

I assume that the add-on thing is unrelated to the kernel change, probably some random factor which makes it work sometimes.

Can you try `ha su reload` next time it doesn't work, and see if that helps?

MihaT2021, 2022-10-18:
yes version:
 Home Assistant 2022.10.4 Supervisor 2022.10.0 Operating System 10.0.dev20221018 Frontend 20221010.0 - latest 
is working :)

agners, 2022-10-18:
@MihaT2021 ok, thanks for testing. I'll report it to the upstream developers. For HAOS we can revert the offending commit for now, so this should be fixed in the next release.

agners, 2022-10-20:
@MihaT2021 sorry to bother you again, the upstream patch developer has a possible "proper" fix for this problem, but he is not sure. Could you test this image?

https://os-builds.home-assistant.io/haos_odroid-xu4-10.0.dev20221020.img.xz

MihaT2021, 2022-10-21:
Yes, it is working. Sata disk is recognize.

 Home Assistant 2022.10.5 Supervisor 2022.10.0 Operating System 10.0.dev20221020 Frontend 20221010.0 - latest 

[image]

agners, 2022-10-21:
Hm, I am sorry, I made an error with the above build, that was too late yesterday :cold_sweat: . Can you retry with this version:
https://os-builds.home-assistant.io/haos_odroid-xu4-10.0.dev20221021.img.xz

MihaT2021, 2022-10-21:
Home Assistant 2022.10.5 Supervisor 2022.10.0 Operating System 10.0.dev20221021 Frontend 20221010.0 - latest 

SATA disk is not recognize.

MihaT2021, 2022-10-27:
Home Assistant 2022.10.5 Supervisor 2022.10.0 Operating System 9.3 Frontend 20221010.0 - latest 
https://github.com/home-assistant/operating-system/releases/download/9.3/haos_odroid-xu4-9.3.img.xz

SATA Disk is working. :)

agners, 2022-11-25:
Closing as this is resolved by https://github.com/home-assistant/operating-system/pull/2194.

The problematic patch has also been lined up to the stable kernel updates (Revert "usb: dwc3: disable USB core PHY management" ).

closed 2022-11-25 by agners