home-assistant/operating-system #2154: Upgrade from 8.5 to 9.0 fails on RPi4/64

sihagen, 2022-09-23:
### Describe the issue you are experiencing

After starting the upgrade via the web interface the pi reboots but is not accessible anymore (no ping/ssh/web). When using a monitor the following error message shows up and the system is repeating it endlessly.
Rebooting the Pi without ethernet connected seemed to get it working again to boot into the "old" 8.5 OS version.

[image]

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

See issue description

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

agners, 2022-09-24:
This very much sounds like a broken SD card 😢

sergvanros, 2022-09-24:
The same thing happened to me tomorrow. But I did not connect HDMI monitor. After several reboots RPI starting with 8.5. I tried to update via terminal (ha os update) - again.

sihagen, 2022-09-24:
> This very much sounds like a broken SD card 😢

Thanks for the fast reply. That would be too bad of course, but why is everything else running fine? There are no errors in the logs hinting at disk failures, Homeassistant Updates 2022.9.6 etc. are running smoothly as well.

agners, 2022-10-27:
SD cards can fail in weird ways. Often some writes start to fail, while reads still succeed.

> There are no errors in the logs hinting at disk failures,

Well, the kerne logs above very much point to a SD card failure (or communication error with it).

sihagen, 2022-12-20:
You were right, was a SD card failure - therefore closing this issue.

closed 2022-12-20 by sihagen