home-assistant/operating-system #2161: Intermittent crash on raspberry Pi 3B+

HarvsG, 2022-09-28:
### Describe the issue you are experiencing

*It is possible this is a supervisor issue*

Every few days, I suffer a system crash on raspberrypi 3B+.  My symptoms are as follows:
- Unable to access homeassistant.local
- Automations do not run
- Unable to ssh into the instance via the add-on
- Device still responds to PING
I have only this morning enabled system ssh on port 22222, So I will see if this still works next time. 

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.0 but has been going on in older versions

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Not sure how to reproduce

### Anything in the Supervisor logs that might be useful for us?

```txt
Logs before most recent crash from `# docker logs hassio_supervisor`
### Crash 2022-09-28

22-09-28 00:38:26 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-09-28 01:05:49 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
22-09-28 01:05:49 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
22-09-28 01:07:04 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.NO_CURRENT_BACKUP/ContextType.SYSTEM
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-09-28 01:07:04 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-09-28 01:07:06 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-09-28 01:07:06 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-09-28 01:07:06 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-09-28 01:08:27 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[09:02:21] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
s6-rc: info: service legacy-services successfully started
[09:02:21] INFO: Starting local supervisor watchdog...
22-09-28 09:02:31 INFO (MainThread) [__main__] Initializing Supervisor setup
22-09-28 09:02:31 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
22-09-28 10:02:31 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi3-64
[... 287 lines cut ...]
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:   File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:     await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:   File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:     await service.entity_service_call(
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:   File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 676, in entity_service_call
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:     future.result()  # pop exception if have
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:     await coro
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:   File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 713, in _handle_entity_call
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:     await result
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:   File "/usr/src/homeassistant/homeassistant/components/cast/media_player.py", line 750, in async_play_media
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:     await self.hass.async_add_executor_job(
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:   File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:     result = self.fn(*self.args, **self.kwargs)
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:   File "/usr/local/lib/python3.10/site-packages/pychromecast/quick_play.py", line 77, in quick_play
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:     controller.quick_play(**data)
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:   File "/usr/local/lib/python3.10/site-packages/pychromecast/controllers/media.py", line 547, in quick_play
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]:     raise PyChromecastError()
Sep 26 19:44:19 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]: pychromecast.error.PyChromecastError
Sep 26 19:45:26 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]: 2022-09-26 20:45:26.253 ERROR (Thread-8) [pychromecast.socket_client] [Kitchen display(192.168.0.180):8009] Error reading from socket.
Sep 26 19:45:26 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]: 2022-09-26 20:45:26.261 WARNING (Thread-8) [pychromecast.socket_client] [Kitchen display(192.168.0.180):8009] Error communicating with socket, resetting connection
Sep 26 19:45:56 homeassistant ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2022.9.1/homeassistant[448]: 2022-09-26 20:45:56.295 ERROR (Thread-8) [pychromecast.socket_client] [Kitchen display(192.168.0.180):8009] Failed to connect to service ServiceInfo(type='host', data=('192.168.0.180', 8009)), retrying in 5.0s
Sep 26 19:46:41 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-8cc7b37c2d9aba5ee4fbec56e41b1fd52e7e5a6dd07fc683887e715019503e8c-runc.rLwDDP.mount: Deactivated successfully.
Sep 26 19:47:12 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-8cc7b37c2d9aba5ee4fbec56e41b1fd52e7e5a6dd07fc683887e715019503e8c-runc.ZT3D4O.mount: Deactivated successfully.
```

HarvsG, 2022-09-28:
No memory overuse on my low-memory RPi:
[image]

rexbron, 2022-10-01:
I'm also noticing intermittent system crashes with HA 9.0 and a Rasberry Pi 4 Model B. Will try to provide logs as well.

pvizeli, 2022-10-03:
All crash have `Out of memory`. So yeah, you need upgrade to a model with more memory or uninstall downsize the application/addons

HarvsG, 2022-10-03:
> All crash have `Out of memory`. So yeah, you need upgrade to a model with more memory or uninstall downsize the application/addons

I can only see that one crash has 'out of memory'? And I have now looked at the history graphs for the other crashes and it doesn't seem to correlate with a memory use spike.

Crash 2022-09-26
[image]

Very weirdly, this morning those memory spikes suddenly stopped after being persistent for weeks. No updates were run!
[image]

agners, 2022-10-04:
That is indeed weird, also no new software has been released in the last 8 days or so.

Maybe a device communicating with Home Assistant (which caused the issue), which is no longer online/rebooted/got an update?

HarvsG, 2022-10-04:
[image]
it has returned!

HarvsG, 2022-10-04:
Found the culprit!
[image]

I will disable speedtest and see if I get any more crashes

rexbron, 2022-10-05:
Getting a ton of the following error messages in my logs. 

`#docker log homeassistant 

2022-10-05 06:55:16.212 WARNING (MainThread) [homeassistant.components.binary_sensor] Updating rpi_power binary_sensor took longer than the scheduled update interval 0:00:30
2022-10-05 06:55:26.221 WARNING (MainThread) [bluetooth_auto_recovery.recover] Reset of hci0 timed out after 10!
2022-10-05 06:55:36.228 WARNING (MainThread) [bluetooth_auto_recovery.recover] Reset of hci0 timed out after 10!
2022-10-05 06:55:36.236 ERROR (MainThread) [homeassistant.components.bluetooth.scanner] hci0 (DC:A6:32:65:DC:FE): Failed to restart Bluetooth scanner: hci0 (DC:A6:32:65:DC:FE): Failed to start Bluetooth: [org.bluez.Error.NotReady] Resource Not Ready
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 232, in _async_start
    await self.scanner.start()  # type: ignore[no-untyped-call]
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/scanner.py", line 150, in start
    self._stop = await manager.active_scan(
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 355, in active_scan
    assert_reply(reply)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.NotReady] Resource Not Ready
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 362, in _async_scanner_watchdog
    await self._async_start()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 313, in _async_start
    raise ScannerStartError(
homeassistant.components.bluetooth.scanner.ScannerStartError: hci0 (DC:A6:32:65:DC:FE): Failed to start Bluetooth: [org.bluez.Error.NotReady] Resource Not Ready
2022-10-05 06:55:46.213 WARNING (MainThread) [homeassistant.components.binary_sensor] Updating rpi_power binary_sensor took longer than the scheduled update interval 0:00:30`

closed 2023-01-11 by github-actions[bot]

HarvsG, 2023-01-11:
Stable since removing speedtest