home-assistant/operating-system #2162: HAOS 9.0 VM bluetoothd issue with USB adater

szymucha94, 2022-10-01:
### Describe the issue you are experiencing

After upgrading from 8.5 to 9.0 I've noticed that bluetooth devices connected via USB bluetooth adapter stopped working. I've checked bluetooth service and this is what I got:
[image]

Please urgently fix this. The adapter is a generic BT4.0 CSR and the VM is just standard x86_64 HAOS.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade from 8.5
2. Check bluetooth service status in cli
3. ???

### Anything in the Supervisor logs that might be useful for us?

```txt
Nope. I've checked bluetooth service right away.
```

### Anything in the Host logs that might be useful for us?

```txt
Attached above as screenshot.
```

szymucha94, 2022-10-01:
removal of the file with adapter's MAC, restarting the service and reloading passive bluetooth monitor helped. But this should not be required. Still - please consider fixing this.

Cheers,

agners, 2022-10-03:
> removal of the file with adapter's MAC

So what you are saying `/var/lib/bluetooth/00:1A:7D:DA:71:13` was a file (instead of a directory)? Was that only a one time thing or was it necessary more than once?

I haven't come across such a problem despite upgrading some devices with Bluetooth. In my case, `/var/lib/bluetooth/` only contains directories.

szymucha94, 2022-10-03:
Yes, it was a file. It removed it just once and since then I've restarted multiple times both HAOS and core alone and it still works fine.

agners, 2022-10-03:
From what I understand BlueZ (which provides the bluetooth deemon) writes directories for each controller since a long time. I have no idea how that file came to be there. My best guess currently is some weird file system corruption (altough, feels very unlikely).

I checked all my machines which support Bluetooth and have been running  OS 8.5 before, I haven't noticed a file in that directory.

szymucha94, 2022-10-03:
Nope, seems like I can't. Backup image has been overwritten today during midnight with newest version.
From what I read only folders are expected in /var/lib/bluetooth/ with MAC names, so I guess this corruption must've happened during upgrade process.
This is not hardware corruption tho as ova vdi is stored on btrfs raid1 array and no block-level corruption has been detected by btrfs. That's checked twice a day every day.
Since the problem is gone you can close this ticket, but please keep it in mind if anyone else comes with similar problem. Cheers!

agners, 2022-10-03:
Do you happen to use Proxmox? Unfortunately, we see corruption with it every now and then (e.g. #2136).

agners, 2022-10-03:
Ok. I am closing the issue, and keep in mind if it appears again. Thanks for raising and sorry we couldn't really get to the bottom of it.

closed 2022-10-03 by agners