home-assistant/operating-system #2163: Home Assistant Keeps Restarting - Seg Faults

JonLaliberte, 2022-10-01:
### Describe the issue you are experiencing

I wrote this up a bit over here: https://community.home-assistant.io/t/home-assistant-keeps-crashing-restarting-often/466895/
(I had also created an issue in the core repo, but they sent me here).

Basically, this VM was running great for some time (a year or more).
Over the weekend the host crashed (odd, and I don't know why yet - hasn't happened before nor since).

After this HA was having issues: HA Core keeps restarting and sometimes Supervisor fails.

Sometimes after a while, the OS will lock up completely (can't SSH in, console let me do anything).

I assume there is corruption that occurred somewhere, but it's not clear to me where/what. I'd rather not start from scratch, so I'm trying to figure out what the root cause of the crashing is to hopefully resolve that.

I'm not sure if this is a HA Core or HA OS issue to be honest, but I'm open to providing any logs/doing any testing that may help figure it out.

Home Assistant 2022.9.7
Supervisor 2022.09.1
Operating System 8.5 - NOTE that I was running 9.0, but the last time I rebooted it switched itself back to 8.5 and I figured I'd let it stay there to test if the problem still existed - it does, so although I'm on 8.5 right this moment the issue affects both 8.5 and 9.0.
Frontend 20220907.2 - latest

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

I don't know what's causing it. The system will run for up to 12+ hours without crashing. Then it core will restart 10 times an hour. Then supervisor sometimes starts crashing, and eventually, the OS goes down. I'm at a loss.

### Anything in the Supervisor logs that might be useful for us?

```txt
NA
```

### Anything in the Host logs that might be useful for us?

```txt
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.656 INFO (MainThread) [homeassistant.setup] Setting up map
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.656 INFO (MainThread) [homeassistant.setup] Setup of domain map took 0.0 seconds
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.657 INFO (MainThread) [homeassistant.setup] Setting up application_credentials
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.658 INFO (MainThread) [homeassistant.setup] Setting up my
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.658 INFO (MainThread) [homeassistant.setup] Setup of domain my took 0.0 seconds
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.658 INFO (MainThread) [homeassistant.setup] Setting up energy
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.658 INFO (MainThread) [homeassistant.setup] Setup of domain energy took 0.0 seconds
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.659 INFO (MainThread) [homeassistant.setup] Setting up system_health
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.662 INFO (MainThread) [homeassistant.setup] Setting up media_source
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.665 INFO (SyncWorker_3) [homeassistant.loader] Loaded caldav from homeassistant.components.caldav
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.673 INFO (MainThread) [homeassistant.setup] Setting up logbook
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.674 INFO (MainThread) [homeassistant.components.media_player] Setting up media_player.itunes
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.682 INFO (MainThread) [homeassistant.setup] Setup of domain tag took 0.1 seconds
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.683 INFO (SyncWorker_1) [homeassistant.loader] Loaded binary_sensor from homeassistant.components.binary_sensor
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.693 INFO (SyncWorker_0) [homeassistant.loader] Loaded siren from homeassistant.components.siren
Sep 30 10:38:20 homeassistant homeassistant[394]: 2022-09-30 06:38:20.697 INFO (SyncWorker_2) [homeassistant.loader] Loaded light from homeassistant.components.light
Sep 30 10:38:20 homeassistant dockerd[394]: panic: runtime error: invalid memory address or nil pointer dereference
Sep 30 10:38:20 homeassistant dockerd[394]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x22c8a4d]
Sep 30 10:38:20 homeassistant dockerd[394]: goroutine 694 [running]:
Sep 30 10:38:20 homeassistant dockerd[394]: github.com/docker/docker/daemon/logger/journald.(*journald).Log(0xc0017d25b8, 0xc00021e000)
Sep 30 10:38:20 homeassistant dockerd[394]:         github.com/docker/docker/daemon/logger/journald/journald.go:106 +0x10d
Sep 30 10:38:20 homeassistant dockerd[394]: github.com/docker/docker/daemon/logger.(*Copier).copySrc(0xc000efc9c0, {0x2b9bf78, 0x6}, {0x7fe6f9611e90, 0xc000efc940})
Sep 30 10:38:20 homeassistant dockerd[394]:         github.com/docker/docker/daemon/logger/copier.go:128 +0xf0d
Sep 30 10:38:20 homeassistant dockerd[394]: created by github.com/docker/docker/daemon/logger.(*Copier).Run
[... 128 lines cut ...]
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/model.py", line 766 in <listcomp>
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/model.py", line 766 in _convert_bluetooth_le_service_uuids
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/model.py", line 62 in __post_init__
  File "<string>", line 9 in __init__
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/model.py", line 81 in from_pb
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 392 in on_msg
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 471 in _read_once
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 480 in _read_loop
  File "/usr/local/lib/python3.10/asyncio/events.py", line 80 in _run
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 1896 in _run_once
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 600 in run_forever
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 633 in run_until_complete
  File "/usr/src/homeassistant/homeassistant/runner.py", line 119 in run
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 191 in main
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 203 in <module>
  File "/usr/local/lib/python3.10/runpy.py", line 86 in _run_code
  File "/usr/local/lib/python3.10/runpy.py", line 196 in _run_module_as_main

Extension modules: _cffi_backend, multidict._multidict, yarl._quoting_c, ciso8601, aiohttp._helpers, aiohttp._http_writer, _brotli, aiohttp._http_parser, aiohttp._websocket, cchardet._cchardet, frozenlist._frozenlist, yaml._yaml, markupsafe._speedups, PIL._imaging, simplejson._speedups, sqlalchemy.cimmutabledict, greenlet._greenlet, sqlalchemy.cprocessors, sqlalchemy.cresultproxy, MySQLdb._mysql, psutil._psutil_linux, psutil._psutil_posix, pydantic.typing, pydantic.errors, pydantic.version, pydantic.utils, pydantic.class_validators, pydantic.config, pydantic.color, pydantic.datetime_parse, pydantic.validators, pydantic.networks, pydantic.types, pydantic.json, pydantic.error_wrappers, pydantic.fields, pydantic.parse, pydantic.schema, pydantic.main, pydantic.dataclasses, pydantic.annotated_types, pydantic.decorator, pydantic.env_settings, pydantic.tools, pydantic, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, bitarray._bitarray, lxml._elementpath, lxml.etree, av.logging, av._core, av.bytesource, av.buffer, av.audio.format, av.enum, av.error, av.utils, av.option, av.descriptor, av.container.pyio, av.dictionary, av.format, av.stream, av.container.streams, av.sidedata.motionvectors, av.sidedata.sidedata, av.packet, av.container.input, av.bitstream.filter, av.bitstream.context, av.container.output, av.container.core, av.codec.context, av.video.format, av.video.reformatter, av.plane, av.video.plane, av.video.frame, av.video.stream, av.codec.codec, av.frame, av.audio.layout, av.audio.plane, av.audio.frame, av.audio.stream, av.audio.fifo, av.filter.pad, av.filter.link, av.filter.context, av.filter.graph, av.filter.filter, av.audio.resampler, matplotlib._c_internal_utils, matplotlib._path, kiwisolver._cext, matplotlib._image, psycopg2._psycopg (total: 109)
```

### Additional information

Happy to provide any other logs.
I've been using journalctl via SSH to check the logs myself, but I'm not seeing anything that I can make sense of.

agners, 2022-10-03:
It seems like Docker crashes hard while writing the log:

```
Sep 30 10:38:20 homeassistant dockerd[394]: panic: runtime error: invalid memory address or nil pointer dereference
Sep 30 10:38:20 homeassistant dockerd[394]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x22c8a4d]
Sep 30 10:38:20 homeassistant dockerd[394]: goroutine 694 [running]:
Sep 30 10:38:20 homeassistant dockerd[394]: github.com/docker/docker/daemon/logger/journald.(*journald).Log(0xc0017d25b8, 0xc00021e000)
Sep 30 10:38:20 homeassistant dockerd[394]:         github.com/docker/docker/daemon/logger/journald/journald.go:106 +0x10d
Sep 30 10:38:20 homeassistant dockerd[394]: github.com/docker/docker/daemon/logger.(*Copier).copySrc(0xc000efc9c0, {0x2b9bf78, 0x6}, {0x7fe6f9611e90, 0xc000efc940})
Sep 30 10:38:20 homeassistant dockerd[394]:         github.com/docker/docker/daemon/logger/copier.go:128 +0xf0d
Sep 30 10:38:20 homeassistant dockerd[394]: created by github.com/docker/docker/daemon/logger.(*Copier).Run
Sep 30 10:38:20 homeassistant dockerd[394]:         github.com/docker/docker/daemon/logger/copier.go:48 +0x71
```
Are those messages always the same?

How much disk space does your system still has?

JonLaliberte, 2022-10-03:
I'm hesitant to type what I'm about to given that I could immediately jinx myself. But it _may be_ running smoothly again. It's been running now for about 20 hours.

Yesterday, I decided to try to start "fresh" (from a backup) - so I downloaded the latest ova image and started to set up a new VM. But, after initially starting up, I was in a boot loop. The VM started up, got to the HA CLI console (sometimes), then almost immediately something signaled it to reboot, and it did. It started booting to the emergency console, etc.

This confused me since it was as fresh as you can get. Eventually, somehow, I ended up being able to get to the Home Assistant setup screen. I selected my backup, and.. nothing. I let it sit for a while (20+ minutes), and I saw on the console (I regret I did not screenshot it) some logging that seemed to show something going wrong with hardware back when I had first started it - and nothing since, so the system seems to have locked up at this point.

I found this issue: https://github.com/home-assistant/operating-system/issues/1782
I had been on Virtualbox 6.1.32, but there was a 6.1.38 release out (latest discussed in the issue was 6.1.36), so I figured I'd try that release before trying the downgrade to 6.1.30 suggested in that issue.
I also figured since I was doing that, I'd try booting my old VM first. And.. What do you know, it's been running smoothly since.

If you want to keep this open for another day or two, I'll come back and close it if it remains stable.
Right now it's still running OS 8.5, but I'll try 9.0 after a few days if things stay up.

JonLaliberte, 2022-10-05:
Ok. It’s been running for 2.5 days. I’m going to call this one solved by the VirtualBox update.

closed 2022-10-05 by JonLaliberte