home-assistant/operating-system #2172: BT/Wifi is unavailable on Beelink BT3 PRO

Lumpusz, 2022-10-06:
### Describe the issue you are experiencing

After installing the OS by flashing it to the emmc storage, WIFI/BT does not show up.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install HA OS 9.0
2. Try to add Bluetooth integration
3. It complains about missing controller

### Anything in the Supervisor logs that might be useful for us?

```txt
22-10-06 13:29:46 INFO (MainThread) [supervisor.store] Loading add-ons from store: 68 all - 0 new - 0 remove
22-10-06 13:29:46 INFO (MainThread) [supervisor.store] Loading add-ons from store: 68 all - 0 new - 0 remove
22-10-06 13:29:59 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
22-10-06 13:29:59 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
22-10-06 13:29:59 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_vscode
22-10-06 13:30:01 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
22-10-06 13:30:01 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_appdaemon
22-10-06 13:30:01 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
22-10-06 13:30:03 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
22-10-06 13:32:57 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
22-10-06 13:33:04 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
22-10-06 13:33:04 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_vscode
22-10-06 13:33:04 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
22-10-06 13:34:02 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
22-10-06 13:34:02 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
22-10-06 13:34:02 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from a0d7b954_ssh
22-10-06 13:34:02 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from a0d7b954_ssh
22-10-06 13:41:27 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
22-10-06 13:41:32 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
22-10-06 13:41:32 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_vscode
22-10-06 13:41:32 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
22-10-06 13:44:09 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
22-10-06 13:46:05 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-10-06 13:51:06 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session None
22-10-06 13:51:10 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session a3740d5d0caeca9e3086f8a01b04c014f57478cce0bdea5fb403cab691f2ac69eba78da449b544701b48cd23a1053e211c879448b00885c90d636689818bdd85
22-10-06 13:51:10 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 4f23b0f8c46542eda2defae767f53d542405fa882e23defc8882eab46faa07daf271ddab57b30d5b1221a4e9058ba2db0392d1ca1cfd5e411c99e8e5e1440d38
22-10-06 14:00:58 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
22-10-06 14:04:12 ERROR (MainThread) [supervisor.api.ingress] Stream error with http://172.30.33.1:1337/stable-784b0177c56c607789f9638da7b6bf3230d47a8c/static/node_modules/vscode-textmate/release/main.js.map: Cannot write to closing transport
22-10-06 14:15:27 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-10-06 14:15:27 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
22-10-06 14:15:27 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.NO_CURRENT_BACKUP/ContextType.SYSTEM
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-10-06 14:15:29 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-10-06 14:16:10 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-10-06 14:16:33 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
22-10-06 14:16:33 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
22-10-06 14:29:50 WARNING (MainThread) [supervisor.api.addons] Changing protected flag for a0d7b954_ssh!
22-10-06 14:29:54 INFO (SyncWorker_5) [supervisor.docker.interface] Stopping addon_a0d7b954_ssh application
22-10-06 14:29:58 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning addon_a0d7b954_ssh application
22-10-06 14:29:58 WARNING (SyncWorker_0) [supervisor.docker.addon] SSH & Web Terminal running with disabled protected mode!
22-10-06 14:29:59 INFO (SyncWorker_0) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/ssh/amd64 with version 12.0.2
22-10-06 14:30:13 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
22-10-06 14:30:13 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
22-10-06 14:30:13 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from a0d7b954_ssh
22-10-06 14:30:13 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from a0d7b954_ssh
22-10-06 14:46:13 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-10-06 14:49:39 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session None
```

### Anything in the Host logs that might be useful for us?

```txt
[    0.335263] Advanced Linux Sound Architecture Driver Initialized.
[    0.335581] Bluetooth: Core ver 2.22
[    0.335616] NET: Registered PF_BLUETOOTH protocol family
[    0.335626] Bluetooth: HCI device and connection manager initialized
[    0.335639] Bluetooth: HCI socket layer initialized
[    0.335648] Bluetooth: L2CAP socket layer initialized
[    0.335666] Bluetooth: SCO socket layer initialized
...
[   35.350256] eth0: renamed from vetha8e657f
[   35.355509] IPv6: ADDRCONF(NETDEV_CHANGE): veth9a35eeb: link becomes ready
[   35.355594] hassio: port 5(veth9a35eeb) entered blocking state
[   35.355602] hassio: port 5(veth9a35eeb) entered forwarding state
[   37.503534] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   37.503547] Bluetooth: BNEP filters: protocol multicast
[   37.503558] Bluetooth: BNEP socket layer initialized
[   40.700610] hassio: port 6(vetha80db88) entered blocking state
[   40.700623] hassio: port 6(vetha80db88) entered disabled state
[   40.700791] device vetha80db88 entered promiscuous mode
[   40.700881] kauditd_printk_skb: 47 callbacks suppressed
```

### Additional information

It seem that the chip is a Broadcom BCM43455. Is that supported? If not can drivers be included in the OS build?
Note: I also posted this question on the forum: https://community.home-assistant.io/t/no-bluetooth-on-beelink-bt3-pro/470754

agners, 2022-10-11:
It seems that this driver is not enabled currently.

Lumpusz, 2022-10-11:
> It seems that this driver is not enabled currently.

Can it be enabled?

agners, 2022-10-11:
Yes. I will have a look before the next release.

Lumpusz, 2022-10-11:
> Yes. I will have a look before the next release.

Thanks.

agners, 2022-10-18:
Unfortunately, the documentation to that device is absolutely horrible. There is nowhere a schematic, block diagram or even an exact data sheet to be found :cry: The only information is this community wiki: https://deviwiki.com/wiki/Beelink_BT3_Pro

The WiFi part is supported by the upstream Broadcom BRCMFMAC driver. However, it seems that the upstream firmware repository does not have all the firmware needed, it needs at least one additional file according to this  thread: https://unix.stackexchange.com/questions/328001/trouble-activating-the-sdio-broadcom-bcm43430-wifi-chip-on-linux-4-8-0/411293#411293

I hesitate to add random files to the image (for licensing reasons).

Also, I assume this is mostly about Bluetooth? Unfortunately, the it seems that the Bluetooth driver is enabled, but not sure how detection of that part works on x86 as it is connected via UART. Again, since there is no documentation/reference software it is kinda hard to get this going.

Does Debian work on that platform? If so, I'd suggest to try a Supervised installation if you really want to use that platform.

Lumpusz, 2022-10-19:
> Unfortunately, the documentation to that device is absolutely horrible. There is nowhere a schematic, block diagram or even an exact data sheet to be found 😢 The only information is this community wiki: https://deviwiki.com/wiki/Beelink_BT3_Pro

Yeah, I wasn't able to find  anything else myself either.

> The WiFi part is supported by the upstream Broadcom BRCMFMAC driver. However, it seems that the upstream firmware repository does not have all the firmware needed, it needs at least one additional [...]

I understand.

> Also, I assume this is mostly about Bluetooth? Unfortunately, the it seems that the Bluetooth driver is enabled, but not sure how detection of that part works on x86 as it is connected via UART. Again, since there is no documentation/reference software it is kinda hard to get this going.

Correct, BT would have been the priority as ethernet works out of the box.

> Does Debian work on that platform? If so, I'd suggest to try a Supervised installation if you really want to use that platform.

I have my production system installed on this box, so I have no way to test it without re-flashing everything which would render my house offline for few hours at least. I will just buy a supported USB adapter or use esp32 proxies.
Also supervised install would mean that I have to keep updating the OS myself, which is something I wanted to avoid, hence went for the HA OS native install.

Thanks for spending the time on this.

closed 2022-10-19 by Lumpusz

tony77682, 2022-10-31:
Hi I also use this material, sad to see that there is no solution 😔