home-assistant/operating-system #2176: FFmpeg is missing in 2022.10.1 version

pShota, 2022-10-09:
### Describe the issue you are experiencing

ffmpeg command does not exist

## Reproduce step

1. flash install on raspberry pi 3
2. find ffmpeg

## Expected
ffmpeg command exist out of box like 2022.09 which I have in my synology docker

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

2022.10.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. flash install on raspberry pi 3
2. find ffmpeg

### Anything in the Supervisor logs that might be useful for us?

```txt
none
```

### Anything in the Host logs that might be useful for us?

```txt
none
```

### System Health information

cannot find the top right button in Settings -> About

frenck, 2022-10-09:
The screenshot provided is from the core SSH add-on (which runs in a separate isolated container). This is no the environment Home Assistant itself runs in.

I'm therefore not sure what the problem is you are trying to raise, as the issue description itself doesn't contain or describe the root issue you are trying to resolve.

../Frenck

pShota, 2022-10-09:
thank for quick response,
sorry I'm new to HA and its weird that my synology docker has it but my raspberry pi doesn't.
I can find both /config folder in the ssh session so I assume it is same environment?
background: I'm trying to add frigate add-on and only getting green screen