home-assistant/operating-system #2180: HAOS memory leak?

codemannh, 2022-10-11:
### Describe the issue you are experiencing

HAOS is using about 2.4GB of memory after a reboot. Immediately after the reboot, the memory usage climbs at a rate of 10 MB every 10 minutes. If Core is restarted, the memory usage doubles after the first restart, and goes up by 1 - 2 GB the next time. After that, HA will eventually hang and the VM stopped to recover. This is something I've noticed as HA Core restarts every day at 0100. Have not focused on that once it appeared that was causing the out of memory condition.

I have been rebooting proactively daily as that is the only way the memory used in the VM is returned. I am not sure how to troubleshoot this so I figured it might be better to open an issue. The graph below shows the memory used on the VM over the last week.
[image]

I've tried to do all I can to find the issue, but I've run completely out of ideas.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Reboot HAOS
2. Monitor memory usage

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing in the log.
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing
```

### System Health information

Home Assistant 2022.10.3
Supervisor 2022.10.0
Operating System 9.2
Frontend 20221010.0 - latest

### Additional information

HA is running on Proxmox version 7.2-3. System has one other Linux container on it that has Frigate running. Other than the disk space, that container uses very little resources. CPU < 5% and memory < 1GB
I have tried stopping all my addons one at a time and watched the system to see if the behavior changes, it doesn't.

agners, 2022-10-13:
What is the console showing when the system hangs?

I am not aware of an active memory leak issue in the Core, but it often depends on configuration etc.

Can you try a new vanilla VM on the side to see if its a system issue or if its configuration related?

codemannh, 2022-10-17:
It has been a while since the console got to a point where it hangs. I will try a new VM and see if there is still the same issue. May take me a little bit of time to get to that done this week.

LeSchurke, 2022-11-22:
I have the same issue. This is the error message from terminal. [image]

LarsNorgaard, 2022-11-24:
This also started happening to me. Proxmox host.

LarsNorgaard, 2022-11-25:
[image]

sahni2021, 2022-11-26:
I am experiencing the same problem, not sure since when exactly but it might be that it started after one of the recent HA updates. There has been no new integration/add-on added to my system. The attached screenshot shows a histogram, and it clearly indicates a systematic increase of RAM usage (memory leak?) which results in a self-reboot situation once the memory overflows.  Any advice is highly appreciated.

Home Assistant 2022.11.4
Supervisor 2022.11.2
Operating System 9.3
Frontend 20221108.0 - latest

[image]

memphisraynz, 2022-12-14:
I have the same memory issue using Proxmox and the HAOS OVA.
Proxmox is reporting aroun a 90% memory uasage on 4BG of ram.

Strangely I performed the OS upgrade to 9.4 this morning and memory dropped to 40%. Until I upgraded HA to 12.5, since then it's back up at 90% again.

pmfernandes, 2023-01-09:
Hi,
Same problem here with Proxmox.
[image]
The drops in the graphic were caused by VM restart.

RosemaryOrchard, 2023-01-17:
I have been experiencing similar issues with HAOS installed on a dedicated machine. I stopped a number of addons via SSH earlier as I was unable to even open the add-ons panel otherwise and now it's sitting at around 2.5GB of RAM used (of 8GB).

RosemaryOrchard, 2023-01-17:
Thanks to some useful guides on the forums, I've used py-spy to get some more data which may prove useful.
[image]
[image]
[dump_process_id.txt](https://github.com/home-assistant/operating-system/files/10439834/dump_process_id.txt)

Though I did leave a number of add-ons stopped still while doing this.

beber-msft, 2023-02-11:
Hello, I do have the same issue, is there any further guidance on how to debug this?

jlandgrave, 2023-02-13:
SAME HERE!!! Someone needs to look into this!

agners, 2023-02-13:
There are countless reasons why a system might run out-of-memory. Every now and then parts of our system encounter a bug which might explain that, often the next patch release already has a fix for it. Without exact information what you experience, annd which software versions you are running, it is impossible to say if you experience the same as the original poster.

Also since @codemannh did not report back since quite a while, maybe the problem the original poster had is actually resolved.

memphisraynz, 2023-02-13:
It's not just that the system is out of memory, It's that the memory starts of low and you can actually watch the usage increase until it hits 100%.

I am happy to provide as much information as you like. Unsure on what would be usefull in determining any issues.

System Health information
Home Assistant 2023.2.2
Supervisor 2023.01.1
Operating System 9.5
Frontend 20230202.0 - latest

Additional information
HA is running on Proxmox version 7.2-11 as a VM from the ova image

codemannh, 2023-02-13:
I apologize for not posting the current status of this issue for me at least. I crafted a post a couple weeks back and must have failed to hit the submit button. 
The issue was never resolved, but given the issues it was causing I had to find another solution. That solution was to dump HAOS and run HA as a container. I moved all the addons into their own containers and configured to work with HA. Since then I have had zero issues. Rock solid on the same hardware. The underlying OS is Ubuntu running Docker. 
Seems this is still an issue, but not sure if it requires me as the original poster to continue working to resolve.

agners, 2023-02-14:
@codemannh thanks for the update! Which Ubuntu version are you using? Ubuntu by default uses swap on the flash drive, which we avoided so far to help SD cards to survive longer :sweat_smile: However, this is not really a concern for virtual machines, so it probably would be worth changing the architecture here.

Do you have a memory usage graph of the same setup? My guess is that Ubuntu with swap enabled allows higher memory peaks, and therefor doesn't crash.

agners, 2023-02-14:
> It's not just that the system is out of memory, It's that the memory starts of low and you can actually watch the usage increase until it hits 100%.

Which memory usage statistics are you observing? What happens in your case? Does the system reboot by itself or does it freeze?

codemannh, 2023-02-14:
@agners the Ubuntu version I'm running is Ubuntu 22.04.1 LTS in a Linux Container running on the same instance of Proxmox that HAOS was running on previously. That is running Debian 11. My drive is NVMe so no issues with any swap that I've ever seen. Memory usage for the last month is captured in the graph. 
[image]
The slight dips are when I'm doing any container updates. As you can see, the workload is generally the same as when I provided the initial graph on this issue, but at no point am I using anywhere near the same amount of memory. In truth, I'm running a couple of additional containers on the same LXC, Prometheus Node exporter and Promtail from Grafana to capture logs to Loki. Both pretty lightweight.

closed 2023-05-22 by github-actions[bot]

chuxubank, 2023-11-29:
> That solution was to dump HAOS and run HA as a container. I moved all the addons into their own containers and configured to work with HA. Since then I have had zero issues.
@codemannh Could you help provide more information about how to dump HAOS and run HA (as well as their addons) as container and config them to run together?

BYW: I don't think this issue should be closed.

agners, 2023-11-29:
We have a total of 204k reported instances using Home Assistant OS. I run Home Assistant OS as well, and monitor memory usage. With quite high certainty, I can say there is no *general* problem with HAOS. However, depending on the exact configuration (add-on used etc.) there might be issue of course. To track this down we'd need the reports help.

> BYW: I don't think this issue should be closed.

Since @codemannh changed to use a container based installation, there is not much we can do here. It is hard to tell what caused the higher than expected memory usage in his configuration, changing from Home Assistant OS/Supervisor (potentially with add-ons) to Container is a significant change in how the system is setup.

This is not the place to ask for guidance for a container installation. Please refer to the documentation or check the community forums/discord.

If you experience memory related problem, and are willing to help track down the issue, please try to isolate the problem to the individual component of the system (e.g. by using the "Glances" add-on, it shows memory per add-on/container). If you think your case is Home Assistant OS related please feel free [to open a new issue](https://github.com/home-assistant/operating-system/issues) along with all information.

Note, some variation in overall memory usage is expected as noted in the [Home Assistant OS 10 blog post](https://www.home-assistant.io/blog/2023/04/18/home-assistant-os-release-10/). Home Assistant OS is tuned to not swap out as much, which leads to higher memory usage than other operating systems might use.