home-assistant/operating-system #2182: Cant download new supervisor when starting HAOS.

juppe1969, 2022-10-12:
### Describe the issue you are experiencing

I'm trying to install the newest version, haos_ova-9.2.vmdk.zip, of HAOS on ESXi. Following this guide: https://community.home-assistant.io/t/hass-io-on-vmware-esxi-6-7-step-by-step/151419.

Everything seems to work, but when it has been preparing for a while, there is an error that it cant download the new supervisor because of time out. And if it keeps going there are the same error over and over again.

I've had the same problem from version 8.5.

The install might work if you dont have to download a new version of the supervisor, but as I'm doing it a few day after the release I have to download a new version or my systems isn't healthy and wont start. And as I cant download a new version it wont start at all.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Create a new virtual machine on ESXI
2. Add the haos_ova-9.2.vmdk as harddrive
3. Start the virtual machine
4. 
 As I said above, I'm using this guide to install HAOS: 
https://community.home-assistant.io/t/hass-io-on-vmware-esxi-6-7-step-by-step/151419

### Anything in the Supervisor logs that might be useful for us?

```txt
Home Assistant logo
Preparing Home Assistant
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[16:03:10] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
s6-rc: info: service legacy-services successfully started
[16:03:10] INFO: Starting local supervisor watchdog...
22-10-12 16:03:13 INFO (MainThread) [__main__] Initializing Supervisor setup
22-10-12 16:03:13 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
22-10-12 16:03:13 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
22-10-12 16:03:13 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: qemux86-64
22-10-12 16:03:13 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/amd64-hassio-supervisor with version 2022.09.1
22-10-12 16:03:13 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
22-10-12 16:03:14 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
22-10-12 16:03:14 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-10-12 16:03:14 INFO (MainThread) [__main__] Setting up Supervisor
22-10-12 16:03:14 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
22-10-12 16:03:14 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
22-10-12 16:03:14 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
22-10-12 16:03:14 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface io.hass.os
22-10-12 16:03:14 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.systemd1
22-10-12 16:03:14 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.login1
22-10-12 16:03:14 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.hostname1
22-10-12 16:03:14 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.timedate1
22-10-12 16:03:14 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.NetworkManager
22-10-12 16:03:14 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface de.pengutronix.rauc
22-10-12 16:03:14 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.resolve1
[... 282 lines cut ...]
22-10-12 18:43:27 ERROR (SyncWorker_2) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2022.10.3: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.10.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Get "https://ghcr.io/v2/": context deadline exceeded")
22-10-12 18:43:27 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-10-12 18:43:57 INFO (SyncWorker_1) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:landingpage to ghcr.io/home-assistant/qemux86-64-homeassistant:2022.10.3
22-10-12 18:43:57 INFO (SyncWorker_1) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2022.10.3.
22-10-12 18:49:33 ERROR (SyncWorker_1) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2022.10.3: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.10.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
22-10-12 18:49:33 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-10-12 18:50:03 INFO (SyncWorker_0) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:landingpage to ghcr.io/home-assistant/qemux86-64-homeassistant:2022.10.3
22-10-12 18:50:03 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2022.10.3.
22-10-12 18:55:43 ERROR (SyncWorker_0) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2022.10.3: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.10.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Get "https://ghcr.io/v2/": context deadline exceeded")
22-10-12 18:55:43 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-10-12 18:56:13 INFO (SyncWorker_2) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:landingpage to ghcr.io/home-assistant/qemux86-64-homeassistant:2022.10.3
22-10-12 18:56:13 INFO (SyncWorker_2) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2022.10.3.
(this can take up to 20 minutes)
While waiting, some suggestions:
Read our founder's vision for the perfect home automation.
Install the official Home Assistant apps
Get it on the Apple App Store Get it on Google Play
Join our community on the forums or on the chat.
```

### Anything in the Host logs that might be useful for us?

```txt
Can't get the Host logs
```

juppe1969, 2022-10-13:
There is an error:

22-10-13 18:12:24 ERROR (SyncWorker_2) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2022.10.3: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.10.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")

And when I try to go to https://ghcr.io/v2/ I get this error:
{"errors":[{"code":"UNAUTHORIZED","message":"authentication required"}]}

If that might be some help.

agners, 2022-10-13:
This usually points to a DNS problem.

What is `ha supervisor info` saying?

juppe1969, 2022-10-14:
ha supervisor info:
```
addons: []
addons_repositories:
- name: Local add-ons
  slug: local
- name: Official add-ons
  slug: core
- name: Home Assistant Community Add-ons
  slug: a0d7b954
- name: ESPHome
  slug: 5c53de3b
arch: amd64
auto_update: true
channel: stable
debug: false
debug_block: false
diagnostics: null
healthy: true
ip_address: 172.30.32.2
logging: info
supported: true
timezone: Europe/Stockholm
update_available: true
version: 2022.09.1
version_latest: 2022.10.0
wait_boot: 5
´´´

And the ha network info:
´´´
docker:
  address: 172.30.32.0/32
  dns: 172.30.32.3
  gateway: 172.30.32.1
  interface: hassio
host_internet: true
interfaces:
- connected: true
  enabled: true
  interface: enp11s0
  ipv4:
    address:
    - 192.168.1.30/24
    gateway: 192.168.1.1
    method: auto
    nameservers:
    - 192.168.1.1
    ready: true
  ipv6:
    address:
    - fe80::a998:17a2:340c:9266/64
    gateway: null
    method: auto
    nameservers: []
    ready: false
  primary: true
  type: ethernet
  vlan: null
  wifi: null
supervisor_internet: true
´´´

Any suggestion if this is a bug or anything wrong on my side?

LukaszUrb, 2022-10-15:
The same here, what router do you have? My problems with HA started when I moved to Netgear.

juppe1969, 2022-10-15:
I have a Netgear R6260. It might be that. I've looked at the router settings and I can't find anything to change to get it to work. Was looking if there was any port that was blocked, but can't find any port to open. So dont know what to do.

Did you get it to work?

LukaszUrb, 2022-10-15:
Not yet, do you have another router to try and confirm that this is a problem with the router? I don't have this possibility.

juppe1969, 2022-10-15:
I've tested i little now and it seems like it is the Netgear router that is blocking the request to the site for download.
I think it is there armor thing that is blocking some sites that are needed for downloading.
I think that armor uses bitdefender for blocking sites that not are secure.
So the problem here seems to be that some of the sites needed for download not are secure according to bitdefender.

I'm not sure that this is the problem, but after my short investigation it's the most probably cause.
So it would be good if some one in the HA-team could look in to this and check with bitdefender if they are blocking any of the needed sites.

LukaszUrb, 2022-10-15:
Is this possible to turn off the netgear armor?

juppe1969, 2022-10-15:
I don't know. I looked at it, but couldn't find a way to do that.

juppe1969, 2022-10-15:
I looked into this a bit more and i seems like a was wrong: the armor thing seems to be something you have to buy to enable, so after a lite research it doesn't seems to be the problem.

LukaszUrb, 2022-10-16:
There is a solution, netgear supports openWRT so there is a possibility of changing router's software, I will try it this month.

juppe1969, 2022-10-16:
It seems like this is not a bug in HA, but rather a bug in Netgear routers. After a change or router everything is working.
So this can be removed or be left here if someone else having the same problem.

Thanks LukaszUrb for the idea that it could be the router.

LukaszUrb, 2022-10-16:
The issue should stay because other people may have the same problem :) Solved.

agners, 2022-10-17:
@juppe1969 so you replaced your router and things are working fine now?

@LukaszUrb how did you solve the problem? By installing OpenWrt? What model are you using?

I won't delete the issue, but I intend to close it as the problem is solved for @juppe1969.

LukaszUrb, 2022-10-17:
@agners I'm going to install open RTW in the coming days, let you know in this topic if this helped.

BTW I also found very similar topic: 
https://github.com/home-assistant/operating-system/issues/1108

juppe1969, 2022-10-18:
@agners yes I replaced my router and now everything is working just fine.
I think that someone in the HA-team should look in to this and ask Netgear what they are doing to block this.

LukaszUrb, 2022-10-19:
Guys, perfect news, installing openWRT on Netgear solved this problem :) 

[image]

cramertj, 2022-11-27:
I'm also encountering this issue when trying to install add ons using a CenturyLink router + modem combo.

agners, 2022-12-01:
@LukaszUrb what Netgear model do you have exactly?

LukaszUrb, 2022-12-01:
Netgear R6220

kasekesy, 2023-02-10:
i have this same problem any fix besides openwrt my nighthawk is a ac2600 and isnt in the list of supported that i can see

LukaszUrb, 2023-02-10:
Use another router while updating HA

LukaszUrb, 2023-02-10:
Every update. Why are saying your router doesn't support openWRT?
As I can see it supports it: 
https://openwrt.org/toh/netgear/r7800

agners, 2023-02-14:
I've tried to reproduce this issue using a NETGEAR R6120. Initially, HAOS had problems connecting to the internet, but I realized that also my laptop had since I didn't setup the router yet :see_no_evil: . So I clicked through their registration process (which is absolutely horrible, lots of terms accepting as well as local and forced cloud account creation :vomiting_face:. I literally feel shame for the whole tech scene that we throw such products at users). Not to mention that their web interface is filled with advertisements and looks really bad! Please, use OpenWrt, it is so much better. Enough ranting. 

After setting up the router, HAOS had no problems going through initial setup as well as downloading a different Core version.

I did update the firmware of that particular device (from 1.0.0.70 to 1.0.0.84). Not sure if that changed something.

closed 2023-05-22 by github-actions[bot]