home-assistant/operating-system #2183: OS 9.2 on Raspberry Pi 4 and Coral

AFumi39, 2022-10-12:
### Describe the issue you are experiencing

Today I updated Home Assistant OS on my Raspberry Pi 4 8GB to 9.2
After the reboot Frigate didn't detected the Coral USB anymore.
I tried rebooting, removing and reconnecting the Coral, etc. but nothing works.
Fortunately the USB SSD have no issue.
I think there's a problem with this release.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Have the 9.1 OS and Frigate with Coral USB
2. Upgrade to 9.2

### Anything in the Supervisor logs that might be useful for us?

```txt
22-10-12 20:59:43 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Frigate is failed, restarting...
```

### Anything in the Host logs that might be useful for us?

```txt
Process detector:coral:
[2022-10-12 21:00:30] frigate.edgetpu                ERROR   : No EdgeTPU was detected. If you do not have a Coral device yet, you must configure CPU detectors.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/frigate/frigate/edgetpu.py", line 135, in run_detector
    object_detector = LocalObjectDetector(
  File "/opt/frigate/frigate/edgetpu.py", line 43, in __init__
```

agners, 2022-10-12:
Can you share the host logs?

Also pleas check if a downgrade indeed helps:

```
ha os upgrade --version 9.2
```

AFumi39, 2022-10-12:
> Can you share the host logs?
> 
> Also pleas check if a downgrade indeed helps:
> 
> ```
> ha os upgrade --version 9.2
> ```

> Can you share the host logs?
> 
> Also pleas check if a downgrade indeed helps:
> 
> ```
> ha os upgrade --version 9.2
> ```

I added the host log as attachment.

Just downgraded to 9.1 and IT WORKS.
Confirmed that the bug is related to 9.2

agners, 2022-10-12:
Nothing changed between 9.1 and 9.2 for Raspberry Pi. So maybe the reboot helped? Can you upgrade to 9.2 and reboot and check a couple of times?

AFumi39, 2022-10-12:
I've already rebooted lot of times with 9.2, at least 6-7 times (disconnecting and reconnecting the Coral) with and without Coral, but nothing changed.
When downgraded to 9.1 worked immediately.

AFumi39, 2022-10-14:
I've tried upgrading again to 9.2 (because I had some problems with Local Tuya HACS integration). Now it works flawlessly, even after 3-4 reboots...
I don't know what happened last time, but I think that something went wrong with the upgrade with 9.2. Downgrading and upgrading again solved the problem.

agners, 2022-10-17:
> I don't know what happened last time, but I think that something went wrong with the upgrade with 9.2. Downgrading and upgrading again solved the problem.

It sounds to me like a power/reset type of problem: Sometimes the USB stick doesn't reset right, and it seems rather reproducible because its dependent on temperature or some other external factor. It could be in-rush power issue which makes the Coral not start properly sometimes. But that is just speculation.

What I try if it doesn't work next time is to connect the stick through an external separately powered USB hub.

I'll close the issue for now, thanks for the update!

closed 2022-10-17 by agners