home-assistant/operating-system #2186: OS 9.2 appears to have broken the Z-Wave JS integration

Chef-de-IT, 2022-10-13:
### Describe the issue you are experiencing

After upgrade to OS 9.2, Z-Wave JS no longer finds the USB stick.

Related issue with specifics on the Z-Wave JS github:
https://github.com/home-assistant/core/issues/80146

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade OS to 9.2
2. Cry

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

agners, 2022-10-14:
Did you try to rollback to HAOS 9.0 as suggested in https://github.com/home-assistant/core/issues/80146#issuecomment-1276665516?

Chef-de-IT, 2022-10-14:
> Did you try to rollback to HAOS 9.0 as suggested in [home-assistant/core#80146 (comment)](https://github.com/home-assistant/core/issues/80146#issuecomment-1276665516)?

Yes, just did it. The issue does go away under HAOS 9.0

Chef-de-IT, 2022-10-14:
> Also, can you share the Z-Wave JS Add-on configuration and logs?

Some logs as of 9.2 & config are on my original report on the https://github.com/home-assistant/core/issues/80146#issuecomment-1276665516
 
This is after the downgrade to 9.0; some strings replaced with [...]:

 Logger: homeassistant.config_entries
Source: config_entries.py:1049
First occurred: 3:15:22 PM (1 occurrences)
Last logged: 3:15:22 PM
Config entry 'Z-Wave JS' for zwave_js integration not ready yet: Failed to connect: Cannot connect to host core-zwave-js:3000 ssl:default [Connect call failed ('172.30.33.0', 3000)]; Retrying in background

========================

[    0.000000] Linux version 5.15.67 (builder@3c891a422ff0) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g5468d36a2) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Tue Sep 13 13:19:08 UTC 2022
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt2)/bzImage root=PARTUUID=8[...]9-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=47[...]43 fsck.repair=yes console=tty1 rauc.slot=A
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 2032
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000b843efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b843f000-0x00000000b843ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000b8440000-0x00000000b8440fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b8441000-0x00000000c9793fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c9794000-0x00000000cadfcfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000cadfd000-0x00000000cae42fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000cae43000-0x00000000cb776fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cb777000-0x00000000cbd17fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000cbd18000-0x00000000cbdfefff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000cbdff000-0x00000000cbdfffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cbe00000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000082dffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.60 by American Megatrends
[    0.000000] efi: ACPI=0xcae0b000 ACPI 2.0=0xcae0b000 SMBIOS=0xf0000 SMBIOS 3.0=0xf0020 TPMFinalLog=0xcb71a000 ESRT=0xcbb6a698 MEMATTR=0xc71f0298 
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: Dell Inc. OptiPlex 7050/0D24M8, BIOS 1.19.0 12/02/2021
[    0.000000] tsc: Detected 2900.000 MHz processor
[    0.000000] tsc: Detected 2899.886 MHz TSC
[    0.001120] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001123] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001131] last_pfn = 0x82e000 max_arch_pfn = 0x400000000
[    0.001224] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002077] last_pfn = 0xcbe00 max_arch_pfn = 0x400000000
[    0.009306] esrt: Reserving ESRT space from 0x00000000cbb6a698 to 0x00000000cbb6a6d0.
[    0.009312] Using GB pages for direct mapping
[    0.009543] Secure boot could not be determined
[    0.009545] ACPI: Early table checksum verification disabled
[    0.009547] ACPI: RSDP 0x00000000CAE0B000 000024 (v02 DELL  )
[    0.009551] ACPI: XSDT 0x00000000CAE0B0B8 0000F4 (v01 DELL   CBX3     01072009 AMI  00010013)
[    0.009556] ACPI: FACP 0x00000000CAE36210 00010C (v05 DELL   CBX3     01072009 AMI  00010013)
[    0.009561] ACPI: DSDT 0x00000000CAE0B240 02AFCA (v02 DELL   CBX3     01072009 INTL 20160422)
[    0.009564] ACPI: FACS 0x00000000CB771F00 000040
[    0.009566] ACPI: APIC 0x00000000CAE36320 0000BC (v03 DELL   CBX3     01072009 AMI  00010013)
[... 901 lines cut ...]
[   58.593926] audit: type=1334 audit(1665773595.782:206): prog-id=0 op=UNLOAD
[   58.593932] audit: type=1334 audit(1665773595.782:207): prog-id=0 op=UNLOAD
[   58.593935] audit: type=1334 audit(1665773595.782:208): prog-id=0 op=UNLOAD
[ 1389.290568] audit: type=1334 audit(1665774926.448:209): prog-id=0 op=UNLOAD
[ 1389.360015] hassio: port 6(vethf8807c6) entered disabled state
[... cut ...]
[ 1389.632723] audit: type=1300 audit(1665774926.789:214): arch=c000003e syscall=44 success=yes exit=40 a0=f a1=c000f951a0 a2=28 a3=0 items=0 ppid=1 pid=451 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj=unconfined key=(null)
[ 1389.632731] audit: type=1327 audit(1665774926.789:214): proctitle=2F7573[...]43A2F2F
[ 1389.632786] hassio: port 6(veth287173f) entered blocking state
[ 1389.632789] hassio: port 6(veth287173f) entered forwarding state
[ 1389.827059] eth0: renamed from veth04e78dc
[ 1389.839145] IPv6: ADDRCONF(NETDEV_CHANGE): veth287173f: link becomes ready

StonyManager, 2022-10-26:
Z-Wave JS flaked out on me about a two months ago after an upgrade. I didn't see any related changes in the release notes, but after tinkering with the version and backups, I found a combo that worked again. I'm not upgrading again until I see this issue resolved. Glad to test and help if that is useful.

agners, 2022-10-27:
@Chef-de-IT it seems that those logs are the host logs & logs of the Core about the Z-Wave JS integration. The Add-on logs are missing, and the host logs in the non-working (HAOS 9.2) are not present in the link.

Ideally, I need the following information so I can diff the before/after:
* Host logs of HAOS 9.0 (working) and HAOS 9.2 (non-working):  Settings -> System -> Logs -> Select "Host" in the top right menu
* Add-on logs of HAOS 9.0 (working) and HAOS 9.2 (non-working):  Settings -> System -> Logs -> Select "Z-Wave JS" in the top right menu

@StonyManager what hardware are you using?

The problem must be quite hardware specific, as we have more than 40k users on OS 9.2, 15.4% use Z-Wave, that means approximately 6000 people are running this combination. I am not saying that your problem does not exist, I am just saying its not a generic problem, it must be some combination of circumstances.

There are also not that much changes between 9.0 and 9.2. The most likely cause is the updated LTS Linux kernel. I usually pick the latest kernel from kernel.org to get the latest bugfixes. But sometimes these "bugfixes" break certain setups, unfortunately.

@Chef-de-IT @StonyManager maybe before testing too much, can you try and see if the latest HAOS 9.3 release fixed this problem?

Chef-de-IT, 2022-10-28:
@agners Stefan,
> There are also not that much changes between 9.0 and 9.2. 
Your commit comment on 9.2 mentions "improve virtual hw support" - I'm not sure if that's even relevant but it sounds so to someone with my level of HAOS code familiarity which is none

> @Chef-de-IT @StonyManager maybe before testing too much, can you try and see if the latest HAOS 9.3 release fixed this problem?
I don't have a dev HA instance - that'll be changing soon, as I've rolled out several pilot installations and it's a winner (THANK YOU!)

The pilot instances all had the same issue (expected as they're all the same HW as well - over-spec'd Dell OptiPlex HW and 700series Zooz stick on up-to-date FW).

> The problem must be quite hardware specific, as we have more than 40k users on OS 9.2, 15.4% use Z-Wave, that means approximately 6000 people are running this combination.

It's possible, although one must take into account what % of those 6K people have consistently opened github tix in cases of past issues that affected all? It stands to reason, 100% of them, and looking at git history, I suspect not even 1%. Which isn't a bad thing :-) so long as we don't conflate it to the issue affecting fewer people than it in fact might.

Again, thank you for all your work; unfortunately I had to roll back all the 9.2 installations of mine in double-quick time before collecting logs, and I promise to have a dev/lab instance up and running shortly.

Chef-de-IT, 2022-10-29:
@agners Stefan,

> There are also not that much changes between 9.0 and 9.2. 

Your commit comment on 9.2 mentions "improve virtual hw support" - I'm not sure if that's even relevant but it sounds so to someone with my level of HAOS code familiarity which is none

> @Chef-de-IT @StonyManager maybe before testing too much, can you try and see if the latest HAOS 9.3 release fixed this problem?

I don't have a dev HA instance where I can test 9.3 - that'll be changing soon, as I've rolled out several pilot HA installations to clients (that have to operate stably / I can't risk just "trying something") and HA overall is a winner when stable (THANK YOU!)

The pilot instances all had the same issue (expected as they're all the same HW as well - over-spec'd Dell OptiPlex HW and 700series Zooz stick on up-to-date FW).

> The problem must be quite hardware specific, as we have more than 40k users on OS 9.2, 15.4% use Z-Wave, that means approximately 6000 people are running this combination.

It's possible, although one must take into account what % of those 6K people have consistently opened github tix in cases of past issues that affected all? It stands to reason, NOT 100% of them, and, looking at git history, I suspect not even 1% of end-users open tix every time they have an issue. Which isn't a bad thing :-) so long as we don't conflate it to the issue affecting fewer people than it in fact might.

Again, thank you for all your work; unfortunately I had to roll back all the 9.2 installations of mine in double-quick time before collecting logs, and I promise to have a dev/lab instance up and running shortly.

agners, 2022-10-31:
> Your commit comment on 9.2 mentions "improve virtual hw support" - I'm not sure if that's even relevant but it sounds so to someone with my level of HAOS code familiarity which is none

This change (https://github.com/home-assistant/operating-system/pull/2138) only affected the image optimized for virtualization (`ova`). The image you are using `generic-x86-64` did not get that change.

> Again, thank you for all your work; unfortunately I had to roll back all the 9.2 installations of mine in double-quick time before collecting logs, and I promise to have a dev/lab instance up and running shortly.

:hugs:  :+1:

StonyManager, 2022-10-31:
My Z-Wave hardware is Z-Wave HUSBZB-1, hosting HA on an ODroid.

StonyManager, 2022-11-04:
I upgraded to latest Z-Wave JS UI, HA 2022.11.1, Supervisor 2022.10.2, OS 9.3, Frontend 20221102.1  Everything appears to be working correctly although most of my automations don't run until sunset (in a few hours). Thanks.

closed 2022-11-25 by github-actions[bot]