home-assistant/operating-system #2187: Proxmox - OVA - Operating System 9.2 - wrong RAM

euDominic, 2022-10-14:
### Describe the issue you are experiencing

Hello.

I am running a Home Assistant 2022.10.3 instant with new Operating System 9.2 on a Proxmox server. 
PVE-Manager-Version: pve-manager/7.2-11/b76d3178
Kernel version: Linux 5.15.60-1-pve #1 SMP PVE 5.15.60-1 (Mon, 19 Sep 2022 17:53:17 +0200

I have assigned 4GB RAM to the instance and 1 CPU with 3 cores.

However, only 1GB RAM is displayed in Home Assistant.
Shouldn't the update to Operating System 9.2 solve this problem?
Or was this assumption wrong?

https://github.com/home-assistant/operating-system/pull/2138
 
Regardless, is there a fix in the works and/or does this bug affect the performance of the system when more than 1GB is used in Home Assistant?

[image]
[image]
[image]
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Operating System 9.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

-

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

agners, 2022-10-14:
What memory usage is reported by Proxmox? Is maybe memory ballooning active? If that is the case, the VM will only allocate as much memory as necessary, and grow memory if needed.

euDominic, 2022-10-14:
[image]
[image]
[image]

agners, 2022-10-14:
Yeah, so when memory ballooning is enabled, this is expected behavior. Shutdown the machine, disable memory balloning and the VM should get a fixed allocation of 4GB.

closed 2022-10-14 by agners

euDominic, 2022-10-14:
Does memory ballooning have any effect on the stability of the system?
Should memory ballooning always be deactivated with home assistant?

LucidityCrash, 2022-10-29:
I've got ballooning enabled on my Proxmox VM but the full 4G gets seen ... I believe the issue is actually the hotplug setting 
[image]
Disable the memory hotplug (not sure about CPU but I did that anyway) and the VM should see all the memory.
I too thought this was fixed in 9 but I guess not in all cases.

michaelw, 2022-12-06:
I don't see the udev support for hotplug, this might explain it:

can we get this added?
```
/lib/udev/rules.d/80-hotplug-cpu-mem.rules

SUBSYSTEM=="cpu", ACTION=="add", TEST=="online", ATTR{online}=="0", ATTR{online}="1"
SUBSYSTEM=="memory", ACTION=="add", TEST=="state", ATTR{state}=="offline", ATTR{state}="online"
```