home-assistant/operating-system #2188: After upgrade to OS 9.2, integrations requiring pymodbus.client.sync failed to start

Claudio1L, 2022-10-14:
### Describe the issue you are experiencing

as per the title, 2x integration failed to start after upgrade to OS 9.2 due to the following errors:

Logger: homeassistant.setup
Source: setup.py:192
First occurred: 10:02:37 (2 occurrences)
Last logged: 10:02:44

    Setup failed for custom integration solax_modbus: Unable to import component: No module named 'pymodbus.client.sync'
    Setup failed for custom integration abb_powerone_pvi_sunspec: Unable to import component: Exception importing custom_components.abb_powerone_pvi_sunspec

Logger: homeassistant.loader
Source: custom_components/abb_powerone_pvi_sunspec/__init__.py:9
Integration: ABB Power-One PVI SunSpec ([documentation](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec), [issues](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues))
First occurred: 10:02:44 (1 occurrences)
Last logged: 10:02:44
Unexpected exception importing component custom_components.abb_powerone_pvi_sunspec

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 703, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/abb_powerone_pvi_sunspec/__init__.py", line 9, in <module>
    from pymodbus.client.sync import ModbusTcpClient
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 166, in <module>
    class ModbusTcpClient(BaseModbusClient):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 170, in ModbusTcpClient
    def __init__(self, host='127.0.0.1', port=Defaults.Port,
AttributeError: type object 'Defaults' has no attribute 'Port'

reverting back to a 9.0 snapshot started working again

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.boot or reboot the HAOS
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Logger: homeassistant.setup
Source: setup.py:192
First occurred: 10:02:37 (2 occurrences)
Last logged: 10:02:44

    Setup failed for custom integration solax_modbus: Unable to import component: No module named 'pymodbus.client.sync'
    Setup failed for custom integration abb_powerone_pvi_sunspec: Unable to import component: Exception importing custom_components.abb_powerone_pvi_sunspec

Logger: homeassistant.loader
Source: custom_components/abb_powerone_pvi_sunspec/__init__.py:9
Integration: ABB Power-One PVI SunSpec ([documentation](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec), [issues](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues))
First occurred: 10:02:44 (1 occurrences)
Last logged: 10:02:44
Unexpected exception importing component custom_components.abb_powerone_pvi_sunspec

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 703, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/abb_powerone_pvi_sunspec/__init__.py", line 9, in <module>
    from pymodbus.client.sync import ModbusTcpClient
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 166, in <module>
    class ModbusTcpClient(BaseModbusClient):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 170, in ModbusTcpClient
    def __init__(self, host='127.0.0.1', port=Defaults.Port,
AttributeError: type object 'Defaults' has no attribute 'Port'
```

### Anything in the Host logs that might be useful for us?

```txt
not apparently
```

agners, 2022-10-14:
This is unrelated to Home Assistant OS. This is likely caused by a Core update, and the successive download of dependencies of custom components. 

I see you already reported this to the custom component: https://github.com/wills106/homeassistant-solax-modbus/issues/154

agners, 2022-10-14:
@alexdelprete the problem is not in your custom component, it's in the  solax modbus custom component linked above. There can only one version of a module be installed, and it seems that component causes a newer one to be installed.

@Claudio1L stop creating issues all over the place! You are using custom components, they can make things break. Only report issues to core repositories when all custom

agners, 2022-10-14:
@alexdelprete its not yours it's the other custom compnent he is using.

@Claudio1L stoo creating isssues all over the place! You are using custom components, there is no support for those in upstream repositories.

closed 2022-10-14 by agners

Claudio1L, 2022-10-15:
My apologise Stefan, in a first phase was convinced the issue started after the upgrade to 9.2. Instead it was a conflict between the Bluetooth Key and the integration I recently added. Removing them, the issue disappeared. 
The issue was arising after the reboot , so was working fine as soon as installed the BT key and only after the reboot for the 9.2 upgrade the issue started. I realized was not due to upgrade after going back to 9.0 using a previous VM snapshot.

Thanks for your support.

Claudio