home-assistant/operating-system #2190: Linux Kernel Security

Delta1977, 2022-10-16:
### Describe the issue you are experiencing

CVE-Nummer | Art | Kurzbeschreibung
-- | -- | --
CVE-2022-41674 | RCE | fix u8 overflow in cfg80211_update_notlisted_nontrans -> max 256 byte overwrite
CVE-2022-42719 | RCE | wifi: mac80211: fix MBSSID parsing use-after-free -> use after free condition
CVE-2022-42720 | RCE | wifi: cfg80211: fix BSS refcounting bugs ref counting -> use-after-free possibilities
CVE-2022-42721 | DOS | wifi: cfg80211: avoid nontransmitted BSS list corruption list corruption -> according to Johannes will however just make it endless loop
CVE-2022-42722 | DOS | wifi: mac80211: fix crash in beacon protection for P2P-device -> NULL ptr dereference crash

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

No

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

No

### Additional information

I think  kernel 5.15.72 is affected

agners, 2022-10-17:
This is addressed with #2191. Please help us test the upcoming nightly.

agners, 2022-10-20:
Addressed with HAOS 9.3.

closed 2022-10-20 by agners