home-assistant/operating-system #2201: Severe Lag on HA Yellow

ecto1a2003, 2022-10-23:
### Describe the issue you are experiencing

I’ve been running the yellow for about a week now and I’m not impressed.

The yellow on the other hand ranges for the same performance speed wise to up to 10 seconds latency in performing automations, browsing the interface or just responding. I’m about to sell this and go back. can anyone help?

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

9.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

use it at all. 

### Anything in the Supervisor logs that might be useful for us?

```txt
not currently
```

### Anything in the Host logs that might be useful for us?

```txt
hard to tell raw log is too large to upload here
```

### System information

## System Information

version | core-2022.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.61-v8
arch | aarch64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.28.2
Stage | running
Available Repositories | 1199
Downloaded Repositories | 63

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | July 22, 2023 at 5:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | us-west-2-2.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.0
agent_version | 1.4.1
docker_version | 20.10.17
disk_total | 916.9 GB
disk_used | 12.7 GB
healthy | true
supported | true
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (2022.10.0), TasmoAdmin (0.20.0), Samba share (10.0.0), Mosquitto broker (6.1.3), Zigbee2MQTT (1.28.0-1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 28
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 14, 2022 at 9:03 PM
-- | --
current_recorder_run | October 22, 2022 at 1:16 AM
estimated_db_size | 4888.06 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Additional information

im trying a conbee2 coordinator to see if that improves

ecto1a2003, 2022-10-23:
HOST LOGS
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.61-v8 (builder@95b6295275c8) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g5468d36a2) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Fri Oct 7 12:06:17 UTC 2022
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003dffffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0, zone DMA32: 8192 pages in unavailable ranges
[    0.000000] percpu: Embedded 28 pages/cpu s77400 r8192 d29096 u114688
[    0.000000] pcpu-alloc: s77400 r8192 d29096 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 507904
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=5cba6efe01c040fa8d409e232c2a0fe2 cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=E4:5F:01:A2:BB:98 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA2,115200n8 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] cgroup: Enabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x000000003a000000-0x000000003e000000] (64MB)
[    0.000000] Memory: 1867672K/2064384K available (12672K kernel code, 2002K rwdata, 4376K rodata, 3968K init, 996K bss, 131176K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 40866 entries in 160 pages
[    0.000000] ftrace: allocated 160 pages with 2 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000305] Console: colour dummy device 80x25
[    0.000890] printk: console [tty1] enabled
[    0.000962] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.001003] pid_max: default: 32768 minimum: 301
[    0.001136] LSM: Security Framework initializing
[    0.001271] AppArmor: AppArmor initialized
[    0.001468] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.001521] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.004982] rcu: Hierarchical SRCU implementation.
[    0.006112] EFI services will not be available.
[    0.006642] smp: Bringing up secondary CPUs ...
[    0.007679] Detected PIPT I-cache on CPU1
[    0.007753] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.008922] Detected PIPT I-cache on CPU2
[    0.008972] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[... 1024 lines cut ...]
[142860.974414] audit: type=1300 audit(1666566284.429:758): arch=c00000b7 syscall=208 success=yes exit=0 a0=4 a1=0 a2=40 a3=55916b12d0 items=0 ppid=409 pid=38720 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses [... cut ...]
[143205.845695] usb 1-1.2: USB disconnect, device number 3
[143218.455963] usb 1-1.2: new full-speed USB device number 6 using dwc2
[143218.661975] usb 1-1.2: New USB device found, idVendor=04b4, idProduct=0003, bcdDevice= 0.00
[143218.661995] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[143218.662002] usb 1-1.2: Product: RFA-Z105-2 HW2.7.3 EMU-2
[143218.662008] usb 1-1.2: Manufacturer: Rainforest Automation, Inc.
[143218.663315] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device

ecto1a2003, 2022-10-23:
SUPERVISOR LOGS
22-10-23 16:04:52 WARNING (MainThread) [supervisor.addons.options] Unknown option 'keepalive' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:52 WARNING (MainThread) [supervisor.addons.options] Unknown option 'reject_unauthorized' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:52 WARNING (MainThread) [supervisor.addons.options] Unknown option 'version' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:52 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'keepalive' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'reject_unauthorized' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'version' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'keepalive' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'reject_unauthorized' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'version' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'keepalive' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'reject_unauthorized' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Unknown option 'version' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:04:57 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:09:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:09:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'keepalive' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:09:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'reject_unauthorized' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:09:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'version' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:09:54 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:10:29 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Rainforest_Automation__Inc._RFA-Z105-2_HW2.7.3_EMU-2-if00
22-10-23 16:10:29 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/003 - None
22-10-23 16:10:44 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/006 - None
22-10-23 16:10:44 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Rainforest_Automation__Inc._RFA-Z105-2_HW2.7.3_EMU-2-if00
22-10-23 16:14:16 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:14:16 WARNING (MainThread) [supervisor.addons.options] Unknown option 'keepalive' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:14:16 WARNING (MainThread) [supervisor.addons.options] Unknown option 'reject_unauthorized' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:14:16 WARNING (MainThread) [supervisor.addons.options] Unknown option 'version' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:14:16 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:14:16 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:14:16 WARNING (MainThread) [supervisor.addons.options] Unknown option 'keepalive' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:14:16 WARNING (MainThread) [supervisor.addons.options] Unknown option 'reject_unauthorized' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:14:16 WARNING (MainThread) [supervisor.addons.options] Unknown option 'version' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:14:16 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
[... cut ...]
22-10-23 16:30:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'keepalive' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:30:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'reject_unauthorized' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:30:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'version' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-10-23 16:30:14 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)

ecto1a2003, 2022-10-23:
CORE LOGS
Resource connection failed to plex.tv: Chrome
4:31:43 PM – (ERROR) Plex Media Server - message first occurred at October 22, 2022 at 4:16:23 PM and shows up 784 times
Payload is not supported (e.g. open, closed, opening, closing, stopped): None
4:31:42 PM – (WARNING) MQTT - message first occurred at October 22, 2022 at 4:09:22 PM and shows up 503 times
issued command while not connected
4:31:32 PM – (ERROR) Rainforest EMU-2 (custom integration) - message first occurred at 4:10:58 PM and shows up 40 times
issued command while not connected
4:31:32 PM – (ERROR) Rainforest EMU-2 (custom integration) - message first occurred at 4:10:58 PM and shows up 40 times
Updating samsungtv_smart media_player took longer than the scheduled update interval 0:00:15
4:31:32 PM – (WARNING) Media player - message first occurred at October 22, 2022 at 4:11:39 PM and shows up 450 times
Failed to to call /supervisor/stats -
4:31:32 PM – (ERROR) Home Assistant Supervisor - message first occurred at October 22, 2022 at 4:10:06 PM and shows up 17 times
Timeout on /supervisor/stats request
4:31:32 PM – (ERROR) Home Assistant Supervisor - message first occurred at October 22, 2022 at 4:10:06 PM and shows up 17 times
Update of sensor.sonarr_shows is taking over 10 seconds
4:30:45 PM – (WARNING) helpers/entity.py - message first occurred at October 22, 2022 at 4:09:19 PM and shows up 934 times
Error looking up Roomba error message: 121
4:27:23 PM – (WARNING) /usr/local/lib/python3.10/site-packages/roombapy/roomba.py - message first occurred at 3:18:49 PM and shows up 18 times
Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'game' when rendering '{{ states.sensor.steam_76561199062976938.attributes.game }}'
4:17:18 PM – (WARNING) helpers/template.py - message first occurred at October 22, 2022 at 5:49:18 PM and shows up 51 times
light.front_door_exterior: Could not calculate power. no brightness set
4:15:53 PM – (ERROR) Powercalc (custom integration) - message first occurred at October 22, 2022 at 6:16:12 PM and shows up 3 times
Error doing job: Unclosed connection
4:15:29 PM – (ERROR) runner.py - message first occurred at October 22, 2022 at 4:40:41 PM and shows up 17 times
device reports readiness to read but returned no data (device disconnected or multiple access on port?)
4:10:30 PM – (ERROR) Rainforest EMU-2 (custom integration)
Timeout fetching OpenSprinkler resource status data
4:10:11 PM – (ERROR) OpenSprinkler (custom integration) - message first occurred at October 22, 2022 at 5:21:17 PM and shows up 35 times
websocket connected
4:09:05 PM – (WARNING) SamsungTV Smart (custom integration) - message first occurred at October 22, 2022 at 4:17:22 PM and shows up 54 times
websocket connected
4:09:05 PM – (WARNING) SamsungTV Smart (custom integration) - message first occurred at October 22, 2022 at 4:17:22 PM and shows up 54 times
websocket connected
4:09:05 PM – (WARNING) SamsungTV Smart (custom integration) - message first occurred at October 22, 2022 at 4:17:22 PM and shows up 55 times
AIOHTTP websocket error
4:03:47 PM – (ERROR) /usr/local/lib/python3.10/site-packages/plexwebsocket.py - message first occurred at October 22, 2022 at 8:33:10 PM and shows up 13 times
alert that washing machine is done: Already running
3:38:31 PM – (WARNING) Automation - message first occurred at 3:32:24 PM and shows up 3 times
Update not successful, no data received from <UsgsEarthquakeHazardsProgramFeed(home=(34.1064, -117.5931), url=https://earthquake.usgs.gov/earthquakes/feed/v1.0/summary/all_month.geojson, radius=100.0, magnitude=3.0)>
2:32:37 PM – (WARNING) components/usgs_earthquakes_feed/geo_location.py - message first occurred at October 22, 2022 at 8:26:54 PM and shows up 7 times
Requesting data from https://earthquake.usgs.gov/earthquakes/feed/v1.0/summary/all_month.geojson failed with timeout error
2:32:37 PM – (WARNING) /usr/local/lib/python3.10/site-packages/aio_geojson_client/feed.py - message first occurred at October 22, 2022 at 8:26:54 PM and shows up 7 times
Error in 'if[0]' evaluation: In 'state': In 'state' condition: unknown entity light.rumpus_room_bulbs
2:23:33 PM – (WARNING) helpers/script.py - message first occurred at October 22, 2022 at 9:52:58 PM and shows up 2 times
Could not connect to Plex server: LaserGrid (('Connection aborted.', RemoteDisconnected('Remote end closed connection without response')))
1:52:51 PM – (ERROR) Plex Media Server
Error fetching tautulli data: Request timeout for 'http://192.168.50.198:49159/api/v2?apikey=[REDACTED_API_TOKEN]&cmd=get_activity'
1:44:40 PM – (ERROR) Tautulli - message first occurred at October 22, 2022 at 5:54:15 PM and shows up 11 times
Could not fetch stats for 5c53de3b_esphome:
12:45:42 PM – (WARNING) Home Assistant Supervisor - message first occurred at 12:45:42 PM and shows up 5 times
Timeout on /addons/5c53de3b_esphome/stats request
12:45:42 PM – (ERROR) Home Assistant Supervisor - message first occurred at 12:45:42 PM and shows up 5 times
Error fetching co2signal data: request to https://api.electricitymap.org/v1/co2?lon=-117.59097352623941&lat=34.136381034597775 failed, reason: getaddrinfo ENOTFOUND api.electricitymap.org
12:22:08 PM – (ERROR) CO2 Signal - message first occurred at October 22, 2022 at 6:34:19 PM and shows up 6 times
Error fetching vpn data: None
11:52:23 AM – (ERROR) ASUS Router (custom integration) - message first occurred at 11:49:36 AM and shows up 5 times
[... cut ...]
Error getting initial data for 192.168.50.224: Error while reading data: [Errno 104] Connection reset by peer
October 22, 2022 at 6:05:40 PM – (WARNING) ESPHome
Login attempt or request with invalid authentication from localhost (127.0.0.1). Requested URL: '/api/websocket'. (Mozilla/5.0 (Linux; Android 12; SM-F711U Build/SP2A.220305.013; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/106.0.5249.126 Mobile Safari/537.36 Home Assistant/2022.10.2-2770 (Android 12; SM-F711U))
October 22, 2022 at 5:12:00 PM – (WARNING) HTTP

agners, 2022-10-24:
@ecto1a2003 I am sorry to hear that.

> The yellow on the other hand ranges for the same performance speed wise to up to 10 seconds latency in performing automations, browsing the interface or just responding. I’m about to sell this and go back. can anyone help?

What setup have you been using before?

What is the typical CPU load of your unit? (under Settings -> System -> Hardware, wait for a few seconds to get some points on the graph).

agners, 2022-10-24:
> CPU load has peaks up to 24 to 26 percent

That looks like one CPU is fully loaded (it is  a quad-core system, so at 25% one of the cores is probably fully loaded).

Can you check the installed add-ons, on the info page on the right side you'll see "Add-on CPU Usage". Is one of those standing out?

ecto1a2003, 2022-10-24:
not really , they dont seem to update unless i spam f5. ill keep an eye out

agners, 2022-10-24:
> not really , they dont seem to update unless i spam f5. ill keep an eye out

Yes that is true, but if the add-on is already running, and it is using all the CPU, it would be at 25% on first load.

Is there something in the logs which causes lots of messages maybe?

If no add-on is the culprit, it is most likely the Core. Probably the best approach is to disable integrations to try to find/isolate the offending one.

> I was running it in docker on my qnap nas

What type/architecture was that? Wondering if it had a much more powerful CPU than the CM4.

ecto1a2003, 2022-10-24:
Hi, highest I saw is 2 percent. the qnap is a hex core with 3.3ghz. Is there no way to make the addons multithreaded?

in the logs i see a lot of messages about timeouts as i posted above. even with all addons turned off i still see spikes.

ecto1a2003, 2022-10-24:
i disable a bunch of integrations, still see the spikes on the same-ish schedule but it only peaks to about 12 percent

ecto1a2003, 2022-10-24:
I disabled a few more. now its only 3 percent spikes, seems to do with the ha mobile app....

ill turn everything else back on and see what happens

ecto1a2003, 2022-10-24:
hmm stil seeing the spikes the exact same now. ill go device by device i guess, still odd that it all runs on one core

ecto1a2003, 2022-10-24:
and the sonarr integration. ill monitor

agners, 2022-10-24:
Sorry, not familiar with either of these two integrations. If it turns out to be them, probably worth checking the Core issue tracker: https://github.com/home-assistant/core

bdraco, 2022-10-24:
Related https://community.home-assistant.io/t/home-assistant-yellow-performance-issues/479274/7?u=bdraco

agners, 2022-10-26:
Thanks for the cross-reference. Seems there is nothing wrong with OS here, I am closing this for now.

closed 2022-10-26 by agners