home-assistant/operating-system #2207: HAOS Install Fails on Odroid N2

CodePhase, 2022-10-31:
### Describe the issue you are experiencing

I have been trying to install HAOS on my Odroid N2 (not the plus) 4 gig SBC for some time now without success.  I have tried multiple versions (9.3, 9.2, 8.5, 6.0, 5.13) all with the same issue.  I have followed the install documentation using Etcher on [https://www.home-assistant.io/installation/odroid](https://www.home-assistant.io/installation/odroid) for every attempt.  I have also tried using 'dd' to copy the image file after extracting it from xz format.

I have tried ruling out my SBC as the issue by:
- trying two different Odroid brand power supplies
- trying different electrical outlets
- trying different network cables as suggested in another issue here
- trying different network switches (one uses STP so delays the link; the other does not)
- trying the official odroid ubuntu OS's (both the mate and minimal latest releases work fine)
- trying both installing to eMMC and SD card; same result with each

The full log for the 9.3 attempt is in the host log section below

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

9.3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Download the latest HAOS image from [https://github.com/home-assistant/operating-system/releases/download/9.3/haos_odroid-n2-9.3.img.xz](https://github.com/home-assistant/operating-system/releases/download/9.3/haos_odroid-n2-9.3.img.xz)
2. Extract the image with xz (xz -d <path to haos.img.xz)
3. Write image to eMMC (dd if=<path to haos.img> of=/dev/sdb bs=2M status=progress)
4. Or, use Etcher as described in odroid install page
5. Install eMMC into Odroid N2
6. Attach network cable
7. Attach power and attempt to boot

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
G12B:BL:6e7c85:7898ac;FEAT:E0F83180:402000;POC:F;RCY:0;EMMC:0;READ:0;0.
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

L0:00000000
L1:00000703
L2:0000c067
L3:14000020
B2:00402000
B1:e0f83180

TE: 303086

BL2 Built : 06:17:13, Jun 28 2019. g12b gf0505d7-dirty - qi.duan@droid13

Board ID = 4
Set A53 clk to 24M
Set A73 clk to 24M
Set clk81 to 24M
A53 clk: 1200 MHz
A73 clk: 1200 MHz
CLK81: 166.6M
[... 965 lines cut ...]
[   49.477648]  touch_atime+0x1b0/0x250
[   49.481179]  ovl_file_accessed.part.0+0x4c/0x90
[   49.485676]  ovl_read_iter+0x178/0x250
[   49.489367]  new_sync_read+0xe8/0x190
[   49.492997]  vfs_read+0x148/0x1e0
[   49.496266]  ksys_read+0x6c/0x100
[   49.499532]  __arm64_sys_read+0x1c/0x30
[   49.503330]  invoke_syscall+0x44/0x110
[   49.507034]  el0_svc_common.constprop.0+0xcc/0xf0
[   49.511695]  do_el0_svc+0x24/0x90
[   49.514923]  el0_svc+0x20/0x60
[   49.517896]  el0t_64_sync_handler+0xb0/0xc0
[   49.521999]  el0t_64_sync+0x1a0/0x1a4
[   49.525574] Code: f9402a82 f100003f d1004023 9a811061 (f9402821) 
[   49.531596] ---[ end trace d3ac3d64f6ae0e0a ]---
[   49.536141] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[   49.542965] SMP: stopping secondary CPUs
[   49.546834] Kernel Offset: disabled
[   49.550253] CPU features: 0x000060d1,00000846
[   49.554594] Memory Limit: none
[   49.557601] Rebooting in 5 seconds..
bl31 reboot reason: 0xd
bl31 reboot reason: 0x0
system cmd  1.
```

agners, 2022-10-31:
Are the errors on all OS versions similar looking?

The error above looks like some type of hardware error, possible memory issue.

> trying the official odroid ubuntu OS's (both the mate and minimal latest releases work fine)

Did you also try to stress those systems a bit? HAOS creates quite some stress as it downloads the latest version of Home Assistant Core on first boot (which leads to 1GB+ downloads, extraction and lots of writes to the eMMC).

agners, 2022-10-31:
How old is that device? What is the board revision?

CodePhase, 2022-10-31:
I tried the stress-ng suite for CPU and memory and didn't see anything unusual.  If there's some output that would be particularly helpful let me know and I can run and post the results.  As far as errors on the other OS versions, I'd say they are similar but not identical.  For example, here is an excerpt from an attempt with 8.5:

```
[    3.827019] ------------[ cut here ]------------
[    3.832381] meson8b-dwmac ff3f0000.ethernet: Ring mode enabled
[    3.839351] WARNING: CPU: 2 PID: 270 at kernel/rcu/tree_plugin.h:725 rcu_sched_clock_irq+0x664/0xd80
[    3.839368] Modules linked in: crct10dif_ce panfrost gpu_sched meson_dw_hdmi(+) dw_hdmi rc_odroid cec dwmac_meson8b(+) meson_drm meson_ir rtc_meson_vrtc stmmac_platform rc_core stmmac pcs_xpcs snd_soc_simple_amplifier fuse
[    3.839398] CPU: 2 PID: 270 Comm: systemd-udevd Not tainted 5.15.60 #1
[    3.839404] Hardware name: Hardkernel ODROID-N2 (DT)
[    3.839407] pstate: 400000c5 (nZcv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    3.845415] meson8b-dwmac ff3f0000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    3.850817] pc : rcu_sched_clock_irq+0x664/0xd80
[    3.850827] lr : rcu_sched_clock_irq+0x464/0xd80
[    3.850830] sp : ffff80000966bce0
[    3.850832] x29: ffff80000966bce0 x28: ffff800008129ad0 x27: ffff0000e07b52c0
[    3.850839] x26: 0000000000000000 x25: 0000000000000001 x24: 00000000000000c0
[    3.850844] x23: 00000000e3edac47 x22: ffff0000e07b5320 x21: ffff0000078d3a00
[    3.850848] x20: 0000000000000001 x19: ffff80000929b2c0 x18: 0000000000000000
[    3.850852] x17: ffff8000d751f000 x16: ffff80000966c000 x15: 0000000000004000
[    3.850856] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[0;32m  OK  ] Found device /dev/ttyAML0.
[    3.850860] x11: 0000000000000000 x10: 0000000000000000 x9 : 0000000000000000
[    3.976751] x8 : 0000000000000000 x7 : 0000000000000000 x6 : ffff800009495170
[    3.976755] x5 : ffff0000078d3a00 x4 : ffff80000966bce0 x3 : ffff0000078d3a00
[    3.976760] x2 : ffff0000078d3a00 x1 : 0000000000000000 x0 : 00000000ffffffff
[    3.976764] Call trace:
[    3.976768]  rcu_sched_clock_irq+0x664/0xd80
[    3.976774]  update_process_times+0x98/0xf0
[    3.976779]  tick_sched_handle.isra.0+0x34/0x60
[    3.976784]  tick_sched_timer+0x4c/0xb0
[    3.976787]  __hrtimer_run_queues+0x114/0x1c0
[    3.976790]  hrtimer_interrupt+0xe8/0x250
[    3.976793]  arch_timer_handler_phys+0x34/0x50
[    3.976799]  handle_percpu_devid_irq+0x84/0x140
[    3.976804]  handle_domain_irq+0x60/0x90
[    3.976807]  gic_handle_irq+0xa4/0xe0
[    3.976812]  call_on_irq_stack+0x28/0x58
[    3.976817]  do_interrupt_handler+0x54/0x60
[    3.976821]  el0_interrupt+0x34/0xc0
[    3.976825]  __el0_irq_handler_common+0x18/0x30
[    3.976829]  el0t_64_irq_handler+0x10/0x20
[    3.976831]  el0t_64_irq+0x1a0/0x1a4
[    3.976835] ---[ end trace 5afd489c0731d9d8 ]---
```

and

```
[    5.298724] Unable to handle kernel execute from non-executable memory at virtual address ffff0000e07d2f48
[    5.303941] Mem abort info:
[    5.303945]   ESR = 0x8600000f
[[    5.303948]   EC = 0x21: IABT (current EL), IL = 32 bits
[    5.314993]   SET = 0, FnV = 0
[    5.317971]   EA = 0, S1PTW = 0
[    5.321040]   FSC = 0x0f: level 3 permission fault
[    5.325795] swapper pgtable: 4k pages, 48-bit VAs, pgdp=00000000093dd000
[    5.332437] [ffff0000e07d2f48] pgd=18000000f4e52003, p4d=18000000f4e52003, pud=18000000e4864003, pmd=18000000e4760003, pte=00680000e07d2707
[    5.346016] Internal error: Oops: 8600000f [#1] PREEMPT SMP
```

The board is a couple of years old, 2019 I think.  The revision is 0.4 20190327.  Thank you for having a look!

agners, 2022-10-31:
Hm, that error looks quite a bit different again. The problem reminds me of  #1471.

These type of errors are usually caused by memory problems. Now since it works on a different OS, it seems to suggest that we use different memory settings than they are using. The memory timings are set by the bootloader, and last time I checked the official ODROID image and ours was using the exact same firmware there.

What official release did you test? Can you share the image link?

Did you boot the offical image and HAOS image from the same boot media?

CodePhase, 2022-10-31:
Just for good measure, here's some stress-ng results:

```
stress-ng --brk 2 --stack 2 --bigheap 2 -t 2m

stress-ng: info:  [3304] setting to a 120 second (2 mins, 0.00 secs) run per stressor
stress-ng: info:  [3304] dispatching hogs: 2 brk, 2 stack, 2 bigheap
stress-ng: info:  [3304] successful run completed in 120.20s (2 mins, 0.20 secs)
```

```
stress-ng --matrix 0 -t 2m --times

stress-ng: info:  [3347] setting to a 120 second (2 mins, 0.00 secs) run per stressor
stress-ng: info:  [3347] dispatching hogs: 6 matrix
stress-ng: info:  [3347] successful run completed in 120.00s (2 mins, 0.00 secs)
stress-ng: info:  [3347] for a 120.00s run time:
stress-ng: info:  [3347]     720.02s available CPU time
stress-ng: info:  [3347]     719.40s user time   ( 99.91%)
stress-ng: info:  [3347]       0.05s system time (  0.01%)
stress-ng: info:  [3347]     719.45s total time  ( 99.92%)
stress-ng: info:  [3347] load average: 5.85 3.39 1.51
```

```
stress-ng --class memory --all 1 -t 2m

stress-ng: info:  [3356] setting to a 120 second (2 mins, 0.00 secs) run per stressor
stress-ng: info:  [3356] dispatching hogs: 1 atomic, 1 bad-altstack, 1 bsearch, 1 context, 1 full, 1 heapsort, 1 hsearch, 1 judy, 1 list, 1 lockbus, 1 lsearch, 1 malloc, 1 matrix, 1 matrix-3d, 1 mcontend, 1 membarrier, 1 memcpy, 1 memfd, 1 memrate, 1 memthrash, 1 mergesort, 1 mincore, 1 misaligned, 1 null, 1 numa, 1 pipe, 1 pipeherd, 1 prefetch, 1 qsort, 1 radixsort, 1 randlist, 1 remap, 1 resources, 1 rmap, 1 shellsort, 1 skiplist, 1 sparsematrix, 1 stack, 1 stackmmap, 1 str, 1 stream, 1 tlb-shootdown, 1 tmpfs, 1 tree, 1 tsearch, 1 vm, 1 vm-addr, 1 vm-rw, 1 vm-segv, 1 wcs, 1 zero, 1 zlib
stress-ng: info:  [3378] stress-ng-memthrash: starting 6 threads on each of the 1 stressors on a 6 CPU system
stress-ng: info:  [3381] stress-ng-misaligned: skipping method int16atomic, misaligned operations tripped signal 7 'SIGBUS'
stress-ng: info:  [3387] stress-ng-prefetch: using built-in defaults as no suitable cache found
stress-ng: info:  [3381] stress-ng-misaligned: skipping method int32atomic, misaligned operations tripped signal 7 'SIGBUS'
stress-ng: info:  [3383] stress-ng-numa: system has 1 of a maximum 4 memory NUMA nodes
stress-ng: info:  [3381] stress-ng-misaligned: skipping method int64atomic, misaligned operations tripped signal 7 'SIGBUS'
stress-ng: info:  [3411] stress-ng-sparsematrix: 5000 items in 500 x 500 sparse matrix (2.00% full)
stress-ng: info:  [3424] stress-ng-stream: using built-in defaults as no suitable cache found
stress-ng: info:  [3424] stress-ng-stream: stressor loosely based on a variant of the STREAM benchmark code
stress-ng: info:  [3424] stress-ng-stream: do NOT submit any of these results to the STREAM benchmark results
stress-ng: info:  [3424] stress-ng-stream: Using CPU cache size of 2048K
stress-ng: info:  [3381] stress-ng-misaligned: skipping method int128tomic, misaligned operations tripped signal 7 'SIGBUS'
stress-ng: info:  [3381] stress-ng-misaligned: exercised int16rd int16wr int16inc int32rd int32wr int32inc int64rd int64wr int64inc int128rd int128wr int128inc
stress-ng: info:  [3411] stress-ng-sparsematrix: hash     161.4K       280708.79 Get/s       134882.43 Put/s
stress-ng: info:  [3411] stress-ng-sparsematrix: judy      85.6K       210287.73 Get/s        70618.53 Put/s
stress-ng: info:  [3411] stress-ng-sparsematrix: list     174.6K        11763.04 Get/s         8481.36 Put/s
stress-ng: info:  [3411] stress-ng-sparsematrix: mmap       1.9M       161663.47 Get/s        39406.55 Put/s
stress-ng: info:  [3411] stress-ng-sparsematrix: qhash    162.4K       462007.66 Get/s       232900.91 Put/s
stress-ng: info:  [3411] stress-ng-sparsematrix: rb       232.7K       109434.05 Get/s        50993.22 Put/s
stress-ng: info:  [3424] stress-ng-stream: memory rate: 95.28 MB/sec, 38.11 Mflop/sec (instance 0)
stress-ng: info:  [3377] stress-ng-memrate:  write1024:        88.10 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:   write512:        80.99 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:   write256:        81.33 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:   write128:        79.87 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:    write64:        91.02 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:    write32:        84.02 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:    write16:        67.07 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:     write8:        42.75 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:   read64pf:       123.98 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:   read1024:       101.65 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:    read512:       138.15 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:    read256:       138.58 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:    read128:       130.77 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:     read64:       107.59 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:     read32:       109.55 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:     read16:       103.49 MB/sec
stress-ng: info:  [3377] stress-ng-memrate:      read8:        66.54 MB/sec
stress-ng: info:  [3480] stress-ng-zlib: instance 0: compression ratio: 60.92% (0.03 MB/sec)
stress-ng: info:  [3387] stress-ng-prefetch: using a 4096 KB L3 cache, 1408 benchmark rounds
stress-ng: info:  [3387] stress-ng-prefetch: non-prefetch read rate @ 0.26 GB/s
[... cut ...]
stress-ng: info:  [23948] successful run completed in 120.01s (2 mins, 0.01 secs)
stress-ng: info:  [23948] Cannot read perf counters, do not have CAP_SYS_ADMIN capability or /proc/sys/kernel/perf_event_paranoid is set too high (3)
stress-ng: info:  [23948] for a 120.01s run time:
stress-ng: info:  [23948]     720.04s available CPU time
stress-ng: info:  [23948]     135.47s user time   ( 18.81%)
stress-ng: info:  [23948]     548.77s system time ( 76.21%)
stress-ng: info:  [23948]     684.24s total time  ( 95.03%)
stress-ng: info:  [23948] load average: 30.89 49.90 24.32
```

CodePhase, 2022-11-01:
> Hm, that error looks quite a bit different again. The problem reminds me of #1471.
>
> These type of errors are usually caused by memory problems. Now since it works on a different OS, it seems to [...]

Yeah, I saw that issue and tried different network cables and even different switches just in case.  I tried two Odroid images, [minimal](https://east.us.odroid.in/ubuntu_22.04lts/N2/ubuntu-22.04-4.9-minimal-odroid-n2-20220622.img.xz) version 20220228 and [mate](https://east.us.odroid.in/ubuntu_22.04lts/N2/ubuntu-22.04-4.9-mate-odroid-n2-20220629.img.xz) version 20220228

I can check the checksums of the firmware you mentioned in both official Odroid images and the HAOS if you think that would help.  Do you know the locations of the files?

agners, 2022-11-02:
> I can check the checksums of the firmware you mentioned in both official Odroid images and the HAOS if you think that would help. Do you know the locations of the files?

Unfortunately it is not that easy: These firmware files are bundled at build time and part of the U-Boot boot loader. Since we build our own version of that, these things won't match up. You basically have to figure out how the images are being built. Unfortunately these things are not well documented (or I am unable to find the documentation).

Things are better documented/clearer for Armbian, they get the firmware files from https://github.com/LibreELEC/amlogic-boot-fip. I was considering switching to those anyways.

Can you try Armbian on that particular board (along with stressing a bit)? https://www.armbian.com/odroid-n2/

agners, 2022-11-02:
From the explanations in [this slide deck](https://elinux.org/images/e/ef/ELC-E_2020_U-Boot_porting_and_maintaining_a_bootloader_for_a_multimedia_SoC_family.pdf), it seems that `acs.bin` is the responsible file for memory/DDR timings.

At least Armbian seems to use the very same timings as we are using, so if that is the problem, Armbian should fail on your platform as well.

CodePhase, 2022-11-03:
Interesting.  I do get similar errors from Armbian version 22.08:

```
[    6.356893] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000019                                                     
[    6.365645] Mem abort info:                                                                                                                                       
[    6.365650]   ESR = 0x0000000096000044                                                                                                                            
[    6.365652]   EC = 0x25: DABT (current EL), IL = 32 bits                                                                                                          
[    6.365654]   SET = 0, FnV = 0                                                                                                                                    
[    6.365656]   EA = 0, S1PTW = 0                                                                                                                                   
[    6.365657]   FSC = 0x04: level 0 translation fault                                                                                                               
[    6.365659] Data abort info:                                                                                                                                      
[    6.365660]   ISV = 0, ISS = 0x00000044                                                                                                                           
[    6.365661]   CM = 0, WnR = 1                                                                                                                                     
[    6.365663] user pgtable: 4k pages, 48-bit VAs, pgdp=00000000081ec000                                                                                             
[    6.365666] [0000000000000019] pgd=0000000000000000, p4d=0000000000000000                                                                                         
[    6.365671] Internal error: Oops: 96000044 [#1] PREEMPT SMP                                                                                                       
[    6.365675] Modules linked in: sch_fq_codel ramoops reed_solomon pstore_blk pstore_zone efi_pstore ip_tables x_tables autofs4 meson_gxl panfrost gpu_sched rtc_pcb
[    6.365704] CPU: 3 PID: 868 Comm: loadkeys Tainted: G        W         5.19.17-meson64 #22.08.7                                                                   
[    6.365708] Hardware name: Hardkernel ODROID-N2 (DT)                                                                                                              
[    6.365710] pstate: 40000005 (nZcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)                                                                                       
[    6.365714] pc : _raw_spin_unlock_irqrestore+0x10/0x50                                                                                                            
[    6.365725] lr : __stack_depot_save+0x300/0x4e0                                                                                                                   
[    6.424588] systemd[1]: Finished Coldplug All udev Devices.                                                                                                       
[    6.443074] sp : ffff80000acebbc0                                                                                                                                 
[    6.443075] x29: ffff80000acebbc0 x28: ffff00003f35d880 x27: 0000000000000000                                                                                     
[    6.443081] x26: 0000000000000001 x25: ffff00000000c400 x24: ffff00000000c400                                                                                     
[    6.443085] x23: 0000000000000001 x22: 0000000000004b47 x21: 0000000000000000                                                                                     
[    6.496035] x20: ffff80000945ecb8 x19: 0000ffffe5750328 x18: 0000000000000000                                                                                     
[    6.496040] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000                                                                                     
[    6.496044] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000                                                                                     
[    6.517252] x11: 0000000000000000 x10: 0000000000000000 x9 : 0000000000000000                                                                                     
[    6.520367] systemd[1]: Starting Helper to synchronize boot up for ifupdown...                                                                                    
[    6.524320]                                                                                                                                                       
[    6.524322] x8 : 0000000000000000 x7 : 0000000000000000 x6 : 0000000000000000                                                                                     
[    6.524326] x5 : 0000000000400100 x4 : ffff00003f35d880 x3 : 0000000000000000                                                                                     
[... cut ...]
[   67.508651] task:systemd-udevd   state:R  running task     stack:    0 pid:  960 ppid:   892 flags:0x00000802                                                     
[   67.518483] Call trace:                                                                                                                                           
[   67.520898]  __switch_to+0xfc/0x1b8                                                                                                                               
[   67.524348]  0xffff0000087d0245
```

I can't get far enough to run the stress test because the system crashes at startup.

agners, 2022-11-03:
Hm interesting indeed, so probably something with the new kernel.

Can you try a 5.10.x based build [from their archive](https://imola.armbian.com/archive/odroidn2/archive/), e.g.
https://imola.armbian.com/archive/odroidn2/archive/Armbian_21.05.4_Odroidn2_buster_current_5.10.43.img.xz

We did had issues at one point with frequency scaling, but currently we HAOS uses the performance scaling governor, which keeps the CPU clock on a high frequency.

rpardini, 2022-11-03:
I've not heard similar on the N2+ yet...
I'm currently running 6.0.5 on the N2+.
`sbc-bench` results: http://ix.io/4eQP
Today's testing images: https://github.com/rpardini/armbian-release/releases/download/20221103a/Armbian_20221103a-rpardini_Odroidn2_kinetic_edge_6.0.6.img.xz

CodePhase, 2022-11-03:
using the Armbian 21.05.4 image I still get errors at startup and a failed boot. I attached the log.

[odroidN2-armbian21_05-emmc-2.log](https://github.com/home-assistant/operating-system/files/9933671/odroidN2-armbian21_05-emmc-2.log)

Seems to be a similar error

CodePhase, 2022-11-03:
trying the 6.0.6 Armbian image also gives me virtual memory errors.

[odroidN2-armbian6_0_6-emmc-2.log](https://github.com/home-assistant/operating-system/files/9933716/odroidN2-armbian6_0_6-emmc-2.log)

I think I'll try the Odroid image again and see if there's some other way to do another memory test.  I'll report back afterwards.

CodePhase, 2022-11-04:
I ran the stress-ng tests again as well as the sbc-bench (thanks @rpardini I didn't know about this).  I can't see any issues with memory in any of the test results.  Here's the link to the sbc-bench results: [http://ix.io/4eSN](http://ix.io/4eSN).  Here's a complete log with the bootup, testing, dmesg, and complete journal in case anyone is interested:

[odroidN2-ubuntu20220622_minimal-1-emmc.log](https://github.com/home-assistant/operating-system/files/9934379/odroidN2-ubuntu20220622_minimal-1-emmc.log)

rpardini, 2022-11-04:
Yeah, I'm stumped. I've not the original N2, and my N2+ works with the 6.0.6 image.
Looking around, there's a lot of similar stuff, https://github.com/home-assistant/operating-system/issues/1096 and https://github.com/home-assistant/operating-system/issues/1471 at least...

The fact vendor kernel 4.9 works doesn't help much; did you ever get a 5.10+ kernel working?

rpardini, 2022-11-04:
That `21.05.4`'s logs indicate a vendor u-boot trying and failing to boot a kernel... it's an unrelated error. That image might have been messed up 🤦  or maybe you've used an uboot in SPI?

@CodePhase Maybe try [Armbian_22.08.1_Odroidn2_jammy_current_5.10.139](https://imola.armbian.com/archive/odroidn2/archive/Armbian_22.08.1_Odroidn2_jammy_current_5.10.139.img.xz)? (make sure switch is in eMMC position)

CodePhase, 2022-11-06:
Thanks @rpardini but that image doesn't work either.  I can confirm that I have the boot selector switch to the 'MMC' option on the board.  Looks like the uboot on the image has the same issue where it can't boot the kernel.  To answer your previous question, no I have never gotten a 5.10+ kernel to boot.  I usually only use the Odroid Ubuntu images.  Maybe I'll try a simple one like CoreElec or something.

[odroidN2-armbian22_08_1-k5_10_139-emmc.log](https://github.com/home-assistant/operating-system/files/9946208/odroidN2-armbian22_08_1-k5_10_139-emmc.log)

CodePhase, 2022-11-06:
The latest build of CoreElec works ([https://github.com/CoreELEC/CoreELEC/releases/download/20.0-Nexus_alpha3/CoreELEC-Amlogic-ng.arm-20.0-Nexus_alpha3-Odroid_N2.img.gz](https://github.com/CoreELEC/CoreELEC/releases/download/20.0-Nexus_alpha3/CoreELEC-Amlogic-ng.arm-20.0-Nexus_alpha3-Odroid_N2.img.gz)) however it is running kernel 4.9.269:

> uname -r
> 4.9.269

I suppose there is some issue with kernels in the 5.x and above series for N2 (non-plus) boards?

[odroidN2-coreelec-20_0Nexus_alpha3-emmc.log](https://github.com/home-assistant/operating-system/files/9946223/odroidN2-coreelec-20_0Nexus_alpha3-emmc.log)

CodePhase, 2022-11-06:
I wonder if it's just time to give up on getting this to work and just install HomeAssistant on the Odroid Ubuntu Minimal image.  I am also stumped that I seem to be the only person having this issue.  Surely others have attempted to install this on a non-plus N2 board.  Do either of you (@agners  or @rpardini) see any value in continuing to troubleshoot this or is it time to throw in the towel?

rpardini, 2022-11-07:
Thanks for testing @CodePhase. Maybe some Manjaro kernels? I'm out of ideas, sorry.

agners, 2022-11-07:
There are definitely other non-N2+ installations out there, I think @pvizeli is running one. But it seems that your particular device is very special 😰. Without having a failing HW here, it's hard to debug. If using their Odroid image works for you, probably the best way forward, I am sorry.

pvizeli, 2022-11-08:
Yes, my Parents have an N2. I think it should be one of the latest HW revisions because it was the Prototype for the Blue and Hardkernel switch 2 month later over to N2+

What I can say, is it seems to work stable, but I can take a look after my parents are back

closed 2023-02-13 by github-actions[bot]