home-assistant/operating-system #2212: Amcrest Integration Error - Mapping Values Not Allowed in Configuration.yaml

tarheelheath, 2022-11-02:
### Describe the issue you are experiencing

Trying to add amcrest cameras using the native integration steps at https://www.home-assistant.io/integrations/amcrest/.
When adding a camera to the configuration.yaml, error thrown: "Error loading /config/configuration.yaml: mapping values are not allowed here"

Here's what I've added to configuration.yaml using format shown on instruction page:
amcrest:
  -host: 192.168.1.98
   username: admin
   password: password
   name: FrontYard
   port: 80

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 9.3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. add amcrest code to configuration.yaml
2. amcrest:
  -host: 192.168.1.98
   username: admin
   password: password
   name: FrontYard
   port: 80
3. developer tools / check configuration
4. error message appears
...

### Anything in the Supervisor logs that might be useful for us?

```txt
None.
```

### Anything in the Host logs that might be useful for us?

```txt
Logger: homeassistant.util.yaml.loader
Source: util/yaml/loader.py:155
First occurred: 10:29:45 AM (1 occurrences)
Last logged: 10:31:28 AM

mapping values are not allowed here in "/config/configuration.yaml", line 31, column 12
```

### System information

## System Information

version | core-2022.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.74
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4844
Installed Version | 1.28.3
Stage | running
Available Repositories | 1133
Downloaded Repositories | 24

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | October 18, 2023 at 8:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-5.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.0
agent_version | 1.4.1
docker_version | 20.10.18
disk_total | 56.0 GB
disk_used | 4.5 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.6.1), File editor (5.4.1), Samba share (10.0.0), Z-Wave JS (0.1.74), Z-Wave JS UI (1.2.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 14
views | 17
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 23, 2022 at 7:51 PM
-- | --
current_recorder_run | November 2, 2022 at 10:27 AM
estimated_db_size | 378.66 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

unclehack, 2022-11-04:
This seems to be a problem related to the configuration set up for Home Asssistant. I don't see how it is related to the HA OS.

Perhaps you can ask for help on the forums or on Discord? This repository is for issues related to the OS.

tarheelheath, 2022-11-04:
Thanks unclehack - when we had Core we had posted integration issues in that version of the github forum, but we moved to OS. Is Discord the best place to report integration problems on OS then? Thanks!

unclehack, 2022-11-04:
> Thanks unclehack - when we had Core we had posted integration issues in that version of the github forum, but we moved to OS. Is Discord the best place to report integration problems on OS then? Thanks!

Discord should be a good choice. I'm just a user as well. I've made that suggestion because the error is something which comes from HA itself. The OS shouldn't have anything to do with that error.

closed 2022-11-21 by github-actions[bot]

agners, 2022-11-21:
> Is Discord the best place to report integration problems on OS then?

This is a formatting issue of your YAML from what I can tell. Maybe not the proper dash `-`?