home-assistant/operating-system #2215: Core Update failing - Conflict. The container name "/homeassistant" is already in use

Paxy, 2022-11-04:
### Describe the issue you are experiencing

For some time (few HA Core versions) I can't update Core just using UX. Update do backup, then HA instance stops and never comes online. Reboot just return the previous version.

Log of update process shows following:
`22-11-04 10:23:39 INFO (MainThread) [supervisor.backups.manager] Creating new partial backup with slug 514f59d0
22-11-04 10:23:39 INFO (MainThread) [supervisor.homeassistant.module] Backing up Home Assistant Core config folder
22-11-04 10:25:02 INFO (MainThread) [supervisor.homeassistant.module] Backup Home Assistant Core config folder done
22-11-04 10:25:03 INFO (MainThread) [supervisor.backups.manager] Creating partial backup with slug 514f59d0 completed
22-11-04 10:25:03 INFO (MainThread) [supervisor.homeassistant.core] Updating Home Assistant to version 2022.11.1
22-11-04 10:25:03 INFO (SyncWorker_6) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:2022.11.0 to ghcr.io/home-assistant/qemux86-64-homeassistant:2022.11.1
22-11-04 10:25:03 INFO (SyncWorker_6) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2022.11.1.
22-11-04 10:26:35 INFO (SyncWorker_6) [supervisor.docker.interface] Stopping homeassistant application
22-11-04 10:27:00 INFO (SyncWorker_6) [supervisor.docker.interface] Cleaning homeassistant application
22-11-04 10:27:05 INFO (MainThread) [supervisor.homeassistant.module] Update pulse/client.config: /data/tmp/homeassistant_pulse
22-11-04 10:27:05 INFO (SyncWorker_4) [supervisor.docker.interface] Cleaning homeassistant application
22-11-04 10:27:10 ERROR (SyncWorker_4) [supervisor.docker.manager] Can't create container from homeassistant: 409 Client Error for http+docker://localhost/v1.41/containers/create?name=homeassistant: Conflict ("Conflict. The container name "/homeassistant" is already in use by container "afd73194b8e03529d016b51d4b173cbd2d3cb07e8b86a2be5996a2c38b114b5f". You have to remove (or rename) that container to be able to reuse that name.")
22-11-04 10:27:10 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/config: Cannot connect to host 172.30.32.1:8123 ssl:False [Connect call failed ('172.30.32.1', 8123)]
`

I can fix the issue manually by doing following:
- reboot
- docker rm -f homeassistant (while system still boots up) 

How can I fix this issue and make UX upgrade working again?

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

It started a few Supervisor and Core Updates ago, and I have had to manually remove the home assistant docker to get it updated.

### Anything in the Supervisor logs that might be useful for us?

```txt
22-11-04 10:23:39 INFO (MainThread) [supervisor.backups.manager] Creating new partial backup with slug 514f59d0
22-11-04 10:23:39 INFO (MainThread) [supervisor.homeassistant.module] Backing up Home Assistant Core config folder
22-11-04 10:25:02 INFO (MainThread) [supervisor.homeassistant.module] Backup Home Assistant Core config folder done
22-11-04 10:25:03 INFO (MainThread) [supervisor.backups.manager] Creating partial backup with slug 514f59d0 completed
22-11-04 10:25:03 INFO (MainThread) [supervisor.homeassistant.core] Updating Home Assistant to version 2022.11.1
22-11-04 10:25:03 INFO (SyncWorker_6) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:2022.11.0 to ghcr.io/home-assistant/qemux86-64-homeassistant:2022.11.1
22-11-04 10:25:03 INFO (SyncWorker_6) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2022.11.1.
22-11-04 10:26:35 INFO (SyncWorker_6) [supervisor.docker.interface] Stopping homeassistant application
22-11-04 10:27:00 INFO (SyncWorker_6) [supervisor.docker.interface] Cleaning homeassistant application
22-11-04 10:27:05 INFO (MainThread) [supervisor.homeassistant.module] Update pulse/client.config: /data/tmp/homeassistant_pulse
22-11-04 10:27:05 INFO (SyncWorker_4) [supervisor.docker.interface] Cleaning homeassistant application
22-11-04 10:27:10 ERROR (SyncWorker_4) [supervisor.docker.manager] Can't create container from homeassistant: 409 Client Error for http+docker://localhost/v1.41/containers/create?name=homeassistant: Conflict ("Conflict. The container name "/homeassistant" is already in use by container "afd73194b8e03529d016b51d4b173cbd2d3cb07e8b86a2be5996a2c38b114b5f". You have to remove (or rename) that container to be able to reuse that name.")
22-11-04 10:27:10 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/config: Cannot connect to host 172.30.32.1:8123 ssl:False [Connect call failed ('172.30.32.1', 8123)]
```

### Anything in the Host logs that might be useful for us?

```txt
When i try to remove docker before reboot, i am getting "device or resource busy" message:
# docker rm -f homeassistant
Error response from daemon: container edca448772d104c827aba7f5b985f7f3760c1566978977907cf7aaf63ced60bf: driver "overlay2" failed to remove root filesystem: unlinkat /mnt/data/docker/overlay2/4803224d2ff08657113bbad40c6399bb847744493cd51bbc06cb8e6f33bf9188/merged: device or resource busy
```

### System information

## System Information

version | core-2022.11.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.74
arch | x86_64
timezone | Europe/Belgrade
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4891
Installed Version | 1.28.3
Stage | running
Available Repositories | 1140
Downloaded Repositories | 7

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.2
agent_version | 1.4.1
docker_version | 20.10.18
disk_total | 30.8 GB
disk_used | 12.0 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.4.1), Samba share (10.0.0), Mosquitto broker (6.0.1), Run On Startup.d (0.11a)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 14
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 28, 2022 at 08:33
-- | --
current_recorder_run | November 4, 2022 at 10:30
estimated_db_size | 809.02 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

agners, 2022-11-04:
Hm, that is weird, It seems as if something is still using the Core container, for some reason.

Can you try:
```
docker stop homeassistant 
docker rm -f homeassistant 
```

Do you have other add-ons running? Can you make sure to stop all of them and then try to update the Core?

Paxy, 2022-11-04:
Yeah, i have tried with stopping and removing docker manually when update fails. I was getting "resource or device busy" message.
I do have some add-ons running. Will try to stop them before next UX update attempt (when we get next core update).

Paxy, 2022-11-09:
Stopping Addons before starting UX Update did not resolve the issue.
After the update attempt, docker refuses to remove homeassistant container.
```
# docker stop homeassistant
homeassistant
# docker kill homeassistant
Error response from daemon: Cannot kill container: homeassistant: Container 90d8f3c8c6360c82923589434066ab841dbeb518795fcecbb68038f57ee1cce1 is not running
# docker stop homeassistant
homeassistant
# docker rm -f homeassistant
Error response from daemon: container 90d8f3c8c6360c82923589434066ab841dbeb518795fcecbb68038f57ee1cce1: driver "overlay2" failed to remove root filesystem: unlinkat /mnt/data/docker/overlay2/aed157905f56a60eb40e7e489be89f59684577a2a9e32d6cb3893c420ed91ac2/merged: device or resource busy
# 
```

Only after a reboot can I issue rm of the homeassistant container, and the supervisor deploys the most recent version.

agners, 2022-11-10:
It could be some file system corruption or similar which prevents the daemon to unlink that particular directory. Is there maybe an error reported in the kernel logs?

Paxy, 2022-11-10:
Ok, will check dmesg log on the next update.

Paxy, 2022-11-18:
Hmm, the latest HA Core update passed without a glitch. No issues with stopping and deleting containers this time. Everything was done by UX update.
The only thing its different on my system is that I have deleted (not just stopped) Run On Startup.d (0.11a)  custom addon.

agners, 2022-11-21:
Interesting find, that seems to be the culprit: The "Run on Startup.d" add-on is using `docker exec` to execute something in the `homeassistant` (Core) container. This causes the container to be still in use when the Supervisor attempts to update the container.

https://github.com/adamoutler/HassOSConfigurator/blob/main/Run/run.sh#L60

closed 2022-11-21 by agners