home-assistant/operating-system #2218: Problem with Filesystem on VM

enwi, 2022-11-05:
### Describe the issue you are experiencing

Home assistant does not function correctly due to errors with the filesystem.

[image]
[image]

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Let HA run for a few weeks

### Anything in the Supervisor logs that might be useful for us?

```txt
22-11-05 17:29:51 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-mariadb with version 2.5.1
22-11-05 17:29:51 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.4.1
22-11-05 17:29:51 INFO (MainThread) [supervisor.backups.manager] Found 9 backup files
22-11-05 17:29:51 INFO (MainThread) [supervisor.discovery] Loaded 1 messages
22-11-05 17:29:51 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
22-11-05 17:29:51 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
22-11-05 17:29:51 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-11-05 17:29:51 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
22-11-05 17:29:51 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-11-05 17:29:51 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - CoreState.SETUP
22-11-05 17:29:51 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
22-11-05 17:29:51 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-11-05 17:29:51 INFO (MainThread) [__main__] Running Supervisor
22-11-05 17:29:52 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good
22-11-05 17:29:52 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
22-11-05 17:29:52 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' starting 2 add-ons
22-11-05 17:29:52 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_core_mariadb application
22-11-05 17:29:59 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mariadb with version 2.5.1
22-11-05 17:29:59 INFO (SyncWorker_2) [supervisor.docker.interface] Cleaning addon_core_mosquitto application
22-11-05 17:30:09 INFO (SyncWorker_2) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mosquitto with version 6.1.3
22-11-05 17:30:14 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' starting 0 add-ons
22-11-05 17:30:14 INFO (MainThread) [supervisor.core] Skiping start of Home Assistant
22-11-05 17:30:14 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.APPLICATION' starting 2 add-ons
22-11-05 17:30:14 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning addon_a0d7b954_nodered application
22-11-05 17:30:18 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2224950-if00
22-11-05 17:30:18 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/002/002 - None
22-11-05 17:30:24 INFO (SyncWorker_5) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/node-red/amd64 with version 13.5.0
22-11-05 17:30:25 INFO (SyncWorker_2) [supervisor.docker.interface] Cleaning addon_9074a9fa_cloudflared application
22-11-05 17:30:29 ERROR (MainThread) [supervisor.homeassistant.api] Error on call http://172.30.32.1:8123/api/config: 
22-11-05 17:30:31 INFO (MainThread) [supervisor.services.modules.mysql] Set core_mariadb as service provider for MySQL
22-11-05 17:30:35 INFO (SyncWorker_2) [supervisor.docker.addon] Starting Docker add-on ghcr.io/brenner-tobias/cloudflared/amd64 with version 4.0.3
22-11-05 17:30:38 INFO (MainThread) [supervisor.services.modules.mqtt] Set core_mosquitto as service provider for mqtt
22-11-05 17:30:39 ERROR (MainThread) [supervisor.homeassistant.api] Error on call http://172.30.32.1:8123/api/config: 
22-11-05 17:30:41 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
22-11-05 17:30:41 INFO (MainThread) [supervisor.core] Supervisor is up and running
22-11-05 17:30:41 INFO (MainThread) [supervisor.host.info] Updating local host information
22-11-05 17:30:41 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-11-05 17:30:41 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-11-05 17:30:41 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-11-05 17:30:41 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-11-05 17:30:42 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-11-05 17:30:43 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-11-05 17:30:43 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-11-05 17:30:43 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
[... 193 lines cut ...]

</details>

<details><summary>Home Assistant Supervisor</summary>

error | failed to load: unknown
-- | --

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 9
views | 8
mode | storage

</details>

### Additional information

[image]
[image]
[image]

enwi, 2022-11-05:
Might be related with #2216

agners, 2022-11-10:
What Hypervisor is this? IDE is a rather outdated and not well supported interface for hard drives these days, can you choose something else (e.g. SCSI)?

enwi, 2022-11-10:
When I select `SCSI` HAOS does not start at all, `SATA` on the other hand works

enwi, 2022-11-14:
I will update this issue if I see the error again, but this might take a few weeks

closed 2022-11-21 by github-actions[bot]

enwi, 2022-12-01:
The system is more stable than before, but now I get the follwoing errror from time to time, but it does not crash HA completely.
[image]

enwi, 2022-12-01:
Update after a restart of the VM HA does not start anymore, filesystem seems to be broken