home-assistant/operating-system #2220: Raspberry Pi Zero 2 W / HAOS 9.x

TiceRex, 2022-11-05:
### Describe the issue you are experiencing

The Raspberry Pi 2 W was able to run the image for RPi3+ without any problem up to version 8.5.
Unfortunately the network is broken in 9.x, the OS finds the wlan0 device, it can connect to the AP, but cannot get IP address.

More details with dmesg and journalctl logs from version 8.5 and 9.3:
https://community.home-assistant.io/t/raspberry-pi-zero-2-w-haos-8-5-9-x/481461

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

9.x

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Create a fresh boot media for RPi3+ (it doesn't matter if 32 or 64-bit version).
2. Start RPi Zero 2 W with the SD card.
3. No network connectivity: the system can connect to the AP, but cannot get IP address.
4. The error message on the OS console is: ieee80211 phy0: brcmf_escan_timeout: timer expired

### Anything in the Supervisor logs that might be useful for us?

```txt
Network is not connected on OS level, Supervisor logs are not relevant.
```

### Anything in the Host logs that might be useful for us?

```txt
journalctl -b 0 -u NetworkManager

Jun 02 18:09:22 homeassistant NetworkManager[349]: <info>  [1654193362.4639] device (wlan0): driver supports Access Point (AP) mode
Jun 02 18:09:22 homeassistant NetworkManager[349]: <info>  [1654193362.4687] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/2)
Jun 02 18:09:22 homeassistant NetworkManager[349]: <info>  [1654193362.4930] device (wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Jun 02 18:09:22 homeassistant NetworkManager[349]: <info>  [1654193362.7462] device (wlan0): supplicant interface state: internal-starting -> disconnected
Jun 02 18:09:22 homeassistant NetworkManager[349]: <info>  [1654193362.7468] device (wlan0): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
(...)
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6635] policy: auto-activating connection 'IOT-01011223' (5f3f190e-6ff2-4b69-b959-ffe69cb8f77b)
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6687] device (wlan0): Activation: starting connection 'IOT-01011223' (5f3f190e-6ff2-4b69-b959-ffe69cb8f77b)
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6702] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6730] manager: NetworkManager state is now CONNECTING
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6760] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6786] device (wlan0): Activation: (wifi) access point 'IOT-01011223' has security, but secrets are required.
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6789] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6877] device (wlan0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6909] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6940] device (wlan0): Activation: (wifi) connection 'IOT-01011223' has security, and secrets exist.  No new secrets needed.
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6943] Config: added 'ssid' value 'IOT-01011223'
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6945] Config: added 'scan_ssid' value '1'
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6945] Config: added 'bgscan' value 'simple:30:-70:86400'
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6946] Config: added 'key_mgmt' value 'WPA-PSK'
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6947] Config: added 'auth_alg' value 'OPEN'
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.6949] Config: added 'psk' value '<hidden>'
Jun 02 19:58:41 homeassistant NetworkManager[349]: <info>  [1654199921.7819] device (wlan0): supplicant interface state: disconnected -> associating
Jun 02 19:58:42 homeassistant NetworkManager[349]: <info>  [1654199922.4073] device (wlan0): supplicant interface state: associating -> completed
Jun 02 19:58:42 homeassistant NetworkManager[349]: <info>  [1654199922.4075] device (wlan0): Activation: (wifi) Stage 2 of 5 (Device Configure) successful. Connected to wireless network "IOT-01011223"
Jun 02 19:58:42 homeassistant NetworkManager[349]: <info>  [1654199922.4085] device (wlan0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Jun 02 19:58:42 homeassistant NetworkManager[349]: <info>  [1654199922.4207] dhcp4 (wlan0): activation: beginning transaction (timeout in 45 seconds)
Jun 02 19:59:27 homeassistant NetworkManager[349]: <warn>  [1654199967.8466] dhcp4 (wlan0): request timed out
Jun 02 19:59:27 homeassistant NetworkManager[349]: <info>  [1654199967.8468] dhcp4 (wlan0): state changed unknown -> timeout
Jun 02 19:59:27 homeassistant NetworkManager[349]: <info>  [1654199967.8470] device (wlan0): state change: ip-config -> failed (reason 'ip-config-unavailable', sys-iface-state: 'managed')
Jun 02 19:59:27 homeassistant NetworkManager[349]: <info>  [1654199967.8486] manager: NetworkManager state is now DISCONNECTED
```

sgouedj, 2022-11-09:
hello, I have the same problem as you, after updating to HA OS 9.3, core 2022.11.1.  
I am using a pi zero 2w.  
CLI loads fine but I keep getting the following message: 
ha>****** ieee80211 phy0: brcmf_escan_timeout: timer expired.  
Can we use the keyboard keys to interrupt this process?  
thank you

TiceRex, 2022-11-10:
@sgouedj The reconnect process is running in the background, you can stop it by deleting the connection parameters. I don't know the exact command by heart, but it should be something like "nmcli conn del <connection name>".

sgouedj, 2022-11-11:
@TiceRex thx.
I've reinitialized the network config by "# rm -r /mnt/overlay/etc/NetworkManager/system-connections
# reboot "
HAOS start but i dont have any wifi connection . i continue to search.

agners, 2022-12-20:
I can reproduce the problem on my end as well. Buildroot is using the firmware from the LibreELEC repository (https://github.com/LibreELEC/brcmfmac_sdio-firmware-rpi). However, those should be the same as Raspberry Pi OS is distributing. I also tried updating to their latest version, unfortunately the same error continues to show up `ieee80211 phy0: brcmf_escan_timeout: timer expired`.

TiceRex, 2022-12-21:
Thank you for the confirmation.

I tend to think it's kind of router (dhcp) misconfiguration problem (TP-Link Archer C7 / OpenWrt 21.02.2). But it raises three questions:
- Why does it work with HAOS 8.5 flawlessly and not at all with 9.x?
- Why only the RPi0 2W fails, but not any other client on the same router?
- Why there's no such problem with any other OS (I've tried the official RPi OS Lite, Kodi, and MotionEyeOS) on this very same RPi0 2W?

I'm clueless... :-(

agners, 2022-12-21:
This is very curious: The firmware is the exact same as Raspberry Pi OS is using:

```
[    9.082082] brcmfmac: F1 signature read @0x18000000=0x1542a9a6                                                                                                                                                                                                                                                             
[    9.122411] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2                                                                                                                                                                                                                            
[    9.123930] usbcore: registered new interface driver brcmfmac                                                                                                                                                                                                                                                              
[    9.335125] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2                                                                                                                                                                                                                            
[    9.366040] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Mar 31 2022 17:24:51 version 9.88.4.77 (g58bc5cc) FWID 01-3b307371
```

Also we use the Raspberry Pi Linux kernel source tree based on 5.15.76.

The only difference now is user space.

It also seems that on a WiFi level, the device is able to connect: I see the MAC appear in the list of associated devices on my router, and `wpa_supplicant` prints the very same messages between HAOS and Raspberry Pi OS. But it seems not traffic is going through (even DHCP fails). I do not see any `ieee80211 phy0: brcmf_escan_timeout: timer expired` messages in Raspberry Pi OS, so I assume it must be related to those.

agners, 2022-12-22:
It seems that with the firmware update WiFi stopped working. When I use the old firmware binary in the latest release, WiFi seems to be able to connect. However, after a while I do see SDIO errors in the console. I am not sure if those have been there in HAOS 8.5 as well. It seems to me that the older firmware does not work well with the latest kernel.

I am a bit out of ideas here. Moving back to an old firmware seems not really possible, moving back the kernel is not a solution either. I created an issue in the Raspberry Pi firmware repository, I hope to get some useful information to resolve the issue: https://github.com/RPi-Distro/firmware-nonfree/issues/32.

For the time being I think sticking with HAOS 8.5 is the way to go.

TiceRex, 2022-12-22:
Thank you for raising the issue at RPi-Distro project!

Actually, I lost track about firmware versions, but there seems to be two versions of RPi0 2W with different wifi chips, hence need to load different firmwares. There were other reported issues about it, see the open and also the closed issues at RPi-Distro.

agners, 2022-12-22:
Yeah I've noticed the different firmware versions. In my case the `brcmfmac43436-sdio` firmware seems to be used (checking with `dmesg | grep brcmfmac`).

TiceRex, 2023-01-30:
Unfortunately the new HAOS version (9.5) fails the very same way. :-(
`ieee80211 phy0: brcmf_escan_timeout: timer expired`

agners, 2023-04-11:
I've tested with OS 10.0.rc3, the 32-bit image still doesn't seem to work reliable. The 64-bit works though, thanks to zswap I was able to go through full onboarding :tada: .

Eismasch1ne, 2023-05-06:
hi, the latest dev build works flawlessly with wifi, give it a try. Had the same issue 
https://os-builds.home-assistant.io/11.0.dev20230501/

closed 2023-08-11 by github-actions[bot]

xxgmxx, 2024-08-17:
> hi, the latest dev build works flawlessly with wifi, give it a try. Had the same issue https://os-builds.home-assistant.io/11.0.dev20230501/

Is this stil working for you? Which image did you install, pi3?