home-assistant/operating-system #2221: No update notification in webUI for HAOS since v7.5

jant90, 2022-11-07:
### Describe the issue you are experiencing

I've not been getting update notifications inside Home Assistant for HAOS for quite some time now, I believe it started with HAOS 7.4 or 7.5. Initially I stayed on v7.5 for some months but then I upgraded to v8.x sometime by running `os update` from the cli. Later I repeated this to upgrade to v9.0, the version that I'm on now.

I tried joining and leaving the beta channel (and force searching for updates in between) several times to no avail.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Do an online update check from within HA: Settings > System > Updates.

### Anything in the Supervisor logs that might be useful for us?

```txt
Supervisor logs show it's doing an online check:
22-11-07 14:17:49 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json

After switching to the beta channel again:
22-11-07 14:31:53 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/beta.json
```

### Anything in the Host logs that might be useful for us?

```txt
Not that I can see.
```

### System information

## System Information

version | core-2022.11.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.67
arch | x86_64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4668
Installed Version | 1.28.3
Stage | running
Available Repositories | 1205
Downloaded Repositories | 12

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.0
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.2
agent_version | 1.3.0
docker_version | 20.10.17
disk_total | 30.8 GB
disk_used | 14.9 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.0), WireGuard (0.7.0), Glances (0.16.0), File editor (5.4.1), Home Assistant Google Drive Backup (0.108.4), MariaDB (2.5.1), Node-RED (13.5.1), AdGuard Home (4.7.4), ESPHome (2022.10.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 5
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 9, 2022 at 08:46
-- | --
current_recorder_run | November 7, 2022 at 14:13
estimated_db_size | 926.64 MiB
database_engine | mysql
database_version | 10.6.8

</details>

### Additional information

Running `os info` from the cli returns this:
```
board: ova
boot: A
data_disk: /dev/sda8
update_available: true
version: "9.0"
version_latest: "9.3"
```

agners, 2022-11-21:
Can you run the following commands?

```
ha os info
ha supervisor info
```

jant90, 2022-11-21:
Sure!
`ha os info`
```
board: ova
boot: B
data_disk: /dev/sda8
update_available: true
version: "9.2"
version_latest: "9.3"
```
`ha supervisor info`
```
addons:
- icon: true
  name: Samba share
  repository: core
  slug: core_samba
  state: started
  update_available: false
  version: 10.0.0
  version_latest: 10.0.0
- icon: true
  name: WireGuard
  repository: a0d7b954
  slug: a0d7b954_wireguard
  state: started
  update_available: false
  version: 0.7.0
  version_latest: 0.7.0
- icon: true
  name: Glances
  repository: a0d7b954
  slug: a0d7b954_glances
  state: unknown
  update_available: false
  version: 0.17.1
  version_latest: 0.17.1
- icon: true
  name: File editor
  repository: core
  slug: core_configurator
  state: unknown
  update_available: false
  version: 5.4.2
  version_latest: 5.4.2
- icon: true
  name: Home Assistant Google Drive Backup
  repository: cebe7a76
  slug: cebe7a76_hassio_google_drive_backup
  state: started
  update_available: false
  version: 0.109.2
  version_latest: 0.109.2
- icon: true
  name: MariaDB
  repository: core
  slug: core_mariadb
  state: started
  update_available: false
  version: 2.5.1
  version_latest: 2.5.1
- icon: true
  name: Node-RED
  repository: a0d7b954
  slug: a0d7b954_nodered
  state: unknown
  update_available: false
  version: 13.5.3
  version_latest: 13.5.3
- icon: true
  name: Vaultwarden (Bitwarden)
  repository: a0d7b954
  slug: a0d7b954_bitwarden
  state: unknown
  update_available: false
  version: 0.18.0
  version_latest: 0.18.0
- icon: true
  name: Caddy 2
  repository: c80c7555
  slug: c80c7555_caddy-2
  state: started
  update_available: false
  version: 1.2.6
  version_latest: 1.2.6
- icon: true
  name: AdGuard Home
  repository: a0d7b954
  slug: a0d7b954_adguard
  state: unknown
  update_available: false
  version: 4.7.5
  version_latest: 4.7.5
- icon: true
  name: ESPHome
  repository: 5c53de3b
  slug: 5c53de3b_esphome
  state: started
  update_available: false
  version: 2022.11.1
  version_latest: 2022.11.1
- icon: true
  name: SSH & Web Terminal
  repository: a0d7b954
  slug: a0d7b954_ssh
  state: started
  update_available: false
  version: 12.1.3
  version_latest: 12.1.3
addons_repositories:
- name: Local add-ons
  slug: local
- name: Hass.io Addons by einschmidt
  slug: c80c7555
- name: Official add-ons
  slug: core
- name: ESPHome
  slug: 5c53de3b
- name: Home Assistant Google Drive Backup Repository
  slug: cebe7a76
- name: Home Assistant Community Add-ons
  slug: a0d7b954
arch: amd64
auto_update: true
channel: stable
debug: false
debug_block: false
diagnostics: false
healthy: true
ip_address: 172.30.32.2
logging: info
supported: true
timezone: Europe/Amsterdam
update_available: false
version: 2022.10.2
version_latest: 2022.10.2
wait_boot: 5
```

agners, 2022-11-22:
Hm, so the update is being picked up by the system, and the Supervisor knows about it. Seems this is more a Core or frontend issue not picking it up.

jant90, 2022-11-23:
I'm using Firefox and already tried forcefully refreshing the web UI, I also tested in Chrome and in incognito/private tabs as well. I also don't see the notification in the Android app on my phone. So I guess this is a core issue then. I'll create a new issue over there.

agners, 2022-11-25:
Ok, if you did so, please link it here for reference.

closed 2022-11-25 by github-actions[bot]