home-assistant/operating-system #2222: Firmware files are missing for broadcom adapters

bdraco, 2022-11-09:
### Describe the issue you are experiencing

related issue https://github.com/home-assistant/core/issues/78012

related update https://forums.raspberrypi.com/viewtopic.php?t=292113

files are available here https://github.com/winterheart/broadcom-bt-firmware

documented here https://github.com/home-assistant/home-assistant.io/pull/24865

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

9.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Install broadcom adapter, notice error in journal about missing firmware

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
see above
```

bdraco, 2022-11-10:
This PR fixes the discovery https://github.com/home-assistant/core/pull/81926

`CONFIG_FW_LOADER_USER_HELPER_FALLBACK=y` is only enabled on the Odroid n2+ platform so this issue is most relevant there.

nagi121, 2023-01-20:
> https://github.com/home-assistant/operating-system/issues/2222#issuecomment-1308959053

I tried the workaround to make my broadcom adapter work again with no success. The mount does not work:

```
[core-ssh ~]$ mount -o bind /mnt/data/firmware /lib/firmware

mount: mounting /mnt/data/firmware on /lib/firmware failed: Permission denied
```
I have a SSD attached to ODROID-N2+ and moved all data to SSD using Home Assistant [http://homeassistant.local:8123/config/storage](url)

When will there be a permanent solution to this problem?

bdraco, 2023-01-20:
> When will there be a permanent solution to this problem?

I would pick up one of the CSR adapters from the High performance list at https://www.home-assistant.io/integrations/bluetooth/ since they don't need any patch files. 

I'm using the Feasycom FSC-BP119 (CSR8510A10) US $19.99 | 100m Long range Bluetooth USB Dongle Wireless Bluetooth 4.0 CSR Adapter for Windows 10, 8, 7 XP vista
https://a.aliexpress.com/_mM1DlzG

nagi121, 2023-01-20:
The BCM firmware patch files fix a security issue.
Buying other hardware is only an intermediate solution, because in the future the new hardware may also need firmware updates to work. 

I think, a solution is needed to provide the Homeassistant OS with firmware patch files from other sources. I would like to have a folder, where I can place the downloaded firmware patch files and they will be picked up at startup. Maybe a script like the workaround mentioned above.

dbuezas, 2023-04-15:
@bdraco I can't get the workaround to work. I tried:
* copying the file directly to /lib/firmeware/bcrm (I assume that was a typo in your post but tried anyway)
* copying the file directly to /lib/firmeware/brcm
* copying the file directly to /lib/firmeware/
* rebooting and retrying
* removing and reinserting the dongle after each step
* using mnt/data and your exact bash commands (I had to `mkdir /mnt/data` too)
* adding the single file `dmesg | grep -i bluetooth` asks for
* adding all files from the drivers repository
* renaming the file to the second name dmesg suggests 

And unfortunately nothing seems to work :( 
I have a Home Assistant Blue running 2023.4.1 and
```bash
 ➜  dmesg | grep -i bluetooth
```
always gives back
```bash
[ 2341.885901] Bluetooth: hci2: BCM20702A
[ 2341.885916] Bluetooth: hci2: BCM20702A1 (001.002.014) build 0000
[ 2341.886931] Bluetooth: hci2: BCM: firmware Patch file not found, tried:
[ 2341.888364] Bluetooth: hci2: BCM: 'brcm/BCM20702A1-0a5c-21e8.hcd'
[ 2341.894405] Bluetooth: hci2: BCM: 'brcm/BCM-0a5c-21e8.hcd'
```

Anything else I can try? I have other two dongles working:
* `Feasycom FSC-BP119 (CSR8510A10) 📶`
* `ZEXMTE BT-505 (RTL8761BU) 📶`
but I'd really to get my `LM Technologies LM1010 (BCM20702A0) 📶` working again

bdraco, 2023-04-15:
You can use https://github.com/home-assistant/operating-system/pull/2225 now if you are running the 10.x RCs

dbuezas, 2023-04-15:
Am I doing something wrong?

### I'm now on 10.x RC

```bash
➜  ~ ha os info
board: odroid-n2
boot: A
data_disk: G1J39E-0x116d3956
update_available: false
version: 10.0.rc4     
version_latest: 10.0.rc4
```

---

### I created the folder in the right place and copied the driver it is asking for there

```bash
➜  ~ ls -la /mnt/data/supervisor/firmware/brcm/BCM20702A1-0a5c-21e8.hcd
-rw-r--r--    1 root     root         35934 Apr 15 22:00 /mnt/data/supervisor/firmware/brcm/BCM20702A1-0a5c-21e8.hcd
```

---

### But it still doesn't find it 

```bash
➜  ~ dmesg | grep -i bluetooth
...
[  355.836186] Bluetooth: hci1: BCM20702A1 (001.002.014) build 0000
[  355.837213] Bluetooth: hci1: BCM: firmware Patch file not found, tried:
[  355.838197] Bluetooth: hci1: BCM: 'brcm/BCM20702A1-0a5c-21e8.hardkernel,odroid-n2-plus.hcd'
[  355.846497] Bluetooth: hci1: BCM: 'brcm/BCM20702A1-0a5c-21e8.hcd'
[  355.852515] Bluetooth: hci1: BCM: 'brcm/BCM-0a5c-21e8.hardkernel,odroid-n2-plus.hcd'
[  355.860210] Bluetooth: hci1: BCM: 'brcm/BCM-0a5c-21e8.hcd'
[  355.918502] Bluetooth: MGMT ver 1.22
```

bdraco, 2023-04-15:
It looks right to me. Looks like #2225 might need some more work

emontnemery, 2023-05-09:
This works for me on a dev build where the feature was just introduced.

@dbuezas Based on what you've pasted above it looks like you've created a `/mnt/data/supervisor/firmware` from within the supervisor docker container, which won't work.

Output from *host ssh*:

```
# ha os info
board: yellow
boot: B
data_disk: AJTD4R-0x3b154ae2
update_available: true
version: 10.0.dev20230302
version_latest: 11.0.dev20230501
```

```
# readlink -f /usr/lib/firmware/updates/
/mnt/data/supervisor/firmware
```

```
# ls -la /mnt/data/supervisor/firmware/brcm
total 44
drwxr-xr-x    2 root     root          4096 Mar  6 18:23 .
drwxr-xr-x    3 root     root          4096 Mar  6 18:20 ..
-rw-r--r--    1 root     root         35000 Mar  6 18:23 BCM20702A1-0b05-17cb.hcd
```

```
# dmesg | grep -i bluetooth
[2929343.052516] Bluetooth: Core ver 2.22
[2929343.052678] NET: Registered PF_BLUETOOTH protocol family
[2929343.052686] Bluetooth: HCI device and connection manager initialized
[2929343.052710] Bluetooth: HCI socket layer initialized
[2929343.052722] Bluetooth: L2CAP socket layer initialized
[2929343.052752] Bluetooth: SCO socket layer initialized
[2929343.180874] Bluetooth: hci0: BCM: chip id 63
[2929343.181873] Bluetooth: hci0: BCM: features 0x07
[2929343.197876] Bluetooth: hci0: BCM20702A
[2929343.197902] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[2929343.201251] Bluetooth: hci0: BCM20702A1 'brcm/BCM20702A1-0b05-17cb.hcd' Patch
[2929344.093846] Bluetooth: hci0: BCM: features 0x07
[2929344.109865] Bluetooth: hci0: Broadcom Bluetooth Device
[2929344.109886] Bluetooth: hci0: BCM20702A1 (001.002.014) build 1467
[2929344.590181] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[2929344.590203] Bluetooth: BNEP filters: protocol multicast
[2929344.590221] Bluetooth: BNEP socket layer initialized
[2929344.594810] Bluetooth: MGMT ver 1.22
```

dbuezas, 2023-05-09:
I can confirm this works and I had missed the fact that this needed to be done from the host, not via "normal" ssh with the AddOn.

SUCCESS!
[image]

I ran this script from the host's keyboard
```
#!/bin/bash
mkdir -p /mnt/data/supervisor/firmware/brcm
cd /mnt/data/supervisor/firmware/brcm
curl -O https://github.com/winterheart/broadcom-bt-firmware/raw/master/brcm/BCM20702A1-0a5c-21e8.hcd
```

Thank you both @bdraco and @emontnemery!
I hope there will be a more straightforward way at some point, but thanks for making this possible at all :)

slingshot74, 2024-03-05:
I get the following error in HA.
firmware patch not found, tried brcm/BCM20702A1-0a5c-21f1.hcd brcm/BCM-05ac-21f1.hcd

Is there a way I can solve this myself?