home-assistant/operating-system #2226: systemd journal logs are piling up

auanasgheps, 2022-11-11:
### Describe the issue you are experiencing

Supervisor and Add-On logs take over 30s to load. Sometime an NginX Gateway timeout error is seen. After reloading multiple times, logs can be finally loaded.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Try to load Supervisor or Add-On logs in a installation that is not fresh.
Loading will fail and will throw an error as described above.

### Anything in the Supervisor logs that might be useful for us?

```txt
lol they don't load
```

### Anything in the Host logs that might be useful for us?

```txt
not much
```

### System information

## System Information

version | core-2022.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.61-v8
arch | aarch64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4385
Installed Version | 1.28.3
Stage | running
Available Repositories | 1142
Downloaded Repositories | 28

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.2
agent_version | 1.4.1
docker_version | 20.10.18
disk_total | 109.3 GB
disk_used | 9.5 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.109.1), SSH & Web Terminal (12.1.2), Studio Code Server (5.4.1), File editor (5.4.2), Let's Encrypt (4.12.7), NGINX Home Assistant SSL proxy (3.1.5), Vaultwarden (Bitwarden) (0.18.0), Mosquitto broker (6.1.3), Zigbee2MQTT (1.28.2-1), Network UPS Tools (0.11.0), WireGuard (0.7.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 13
views | 9
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 29 October 2022 at 12:10
-- | --
current_recorder_run | 11 November 2022 at 20:43
estimated_db_size | 453.57 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.7.1
-- | --
can_reach_server | failed to load: timeout
can_reach_spec | ok
logged_accounts | 0
total_devices | 0

</details>

### Additional information

I connected to the host and run
`journalctl --disk-usage`

It showed a whopping 3.3GB of logs. My installation is more than a year old if I recall correctly, so I'm sure it had the time to collect them

I ran `journalctl --rotate` to clear them and set a clear time with `journalctl --vacuum-time=4weeks` although I don't know if it will persist.

Now the usage only shows 180 Mb of used space and logs load instantly.

I am opening this issue to ask if a vacuum time is already set. If there's no value, would you consider adding it?

HASSOS is installed on a SSD connected via USB2 to my RP4.

auanasgheps, 2022-11-14:
> Any reason we should deviate from that?

My previous logs were 3.3 GB and were pretty heavy to load as I described.

auanasgheps, 2022-11-15:
Also note I'm not the only one who has this issue.
[Here's the discussion](https://community.home-assistant.io/t/home-assistant-logs-timing-out/321582) i found on HA forums describing this problem and the resolution too. 
I only took the extra step to open the issue here :)

bschatzow, 2022-11-15:
I ran the journalctl --disk-usage and get 3.9 G.  When I tried the command journalctl --rotate it does not clear anything for me.  Still at 3.9 G.

agners, 2022-11-16:
@bschatzow rotate does not delete old journals, it creates a new journal database file from what I undertand. `journalctl --vacuum-time=2weeks` or similar would probably help in your case.

In any case, 3.9GB disk usage is what is the expected ceiling currently. That said, I start to agree that it probably make sense to deviate from the systemd default and set it to e.g. 1GB since HAOS is commonly used on lower end hardware.

unclehack, 2022-11-17:
This configuration should be reasonable for HA OS:
```
#file /etc/systemd/journald.conf
[Journal]
....
SystemMaxUse=1G 
SystemKeepFree=5G
```

auanasgheps, 2022-11-17:
> @bschatzow rotate does not delete old journals, it creates a new journal database file from what I undertand. `journalctl --vacuum-time=2weeks` or similar would probably help in your case.
>
> In any [...]

I agree. Or use these custom values just for our poor Raspberries and leave the full fat log to MiniPCs and VMs :)

unclehack, 2022-12-01:
The following patch works:
```
diff --git a/buildroot-external/rootfs-overlay/etc/systemd/journald.conf b/buildroot-external/rootfs-overlay/etc/systemd/journald.conf
index 979f7e0f..b78fc295 100644
--- a/buildroot-external/rootfs-overlay/etc/systemd/journald.conf
+++ b/buildroot-external/rootfs-overlay/etc/systemd/journald.conf
@@ -1,3 +1,5 @@
 [Journal]
+SystemMaxUse=1G
+SystemKeepFree=5G
 Storage=auto
 Compress=yes
```

unclehack, 2022-12-05:
> A specific reason to set `SystemKeepFree`? Default is 4GB or 15%, which seems quite OK.
> 
> Actually one small worry: What if the file system size is 8GB or smaller? E.g. the OVA image uesd to be 6GB, and we likely have installations out there still running on that 6GB disk. The default will cap the journal size to 10% (600MB). So I think this change would effectively increase the journal in those cases.

SystemMaxUse=500M should be better in that case. This would make all platforms store up to 500MB of commpressed logs. SystemKeepFree wouldn't be necessary in that case.

auanasgheps, 2023-02-15:
I've cleared the logs in November before opening this issue, and now there are 1.5 GB of logs.

Logs do still load, but the struggle is evident, as it takes about 15/20 seconds to do so.

Have you taken a decision about the log limit? 1.5GB is already too much for the RPi4, we should probably not cross the 1GB mark at least on this device.

proffalken, 2023-03-01:
I'm also seeing this issue - I'm running HAOS on an old Wyse terminal with 8GB NVMe storage, and the thing that is taking up the most space is journald logs, everything else runs perfectly well.

At the moment, I'm SSH'ing into the physical box on a weekly basis to clear them out (freeing up at least 2G each time I run the command) and that's a real pain.

Whatever the "or 15%" thing is in the defaults, that's clearly not being respected so I'd like to be able to at least set a flag somewhere, or a cronjob, or something that runs `journalctl --vacuum-files=1` or similar to avoid having to constantly log in.

Is this something the team would consider including?

auanasgheps, 2023-03-02:
Indeed. 
I had to run the command recommended by @proffalken to clear out the logs.
I had 1.7 GB of journald logs, which were timing out constantly

Alfagek, 2023-03-26:
@agners 

How can i change the Journald.conf file inside HASS OS? 
Or how can i make a "automation" inside HA that cleans the journalctl logs at the moment that i want?

I now always do it manual  using the SSH therminal in "proxmox" use then "login" and then the rotate/vacuum command.
System wise, its not a problem that its growing to 4GB or more, but i like to have a clean system. And it also saves me a bit space with making backups in proxmox.

agners, 2023-03-28:
This will be part of the next release candidate 10.0.rc3.

closed 2023-03-28 by agners