home-assistant/operating-system #2237: HOAS trying to contact sentry.io

ovenslighted08, 2022-11-17:
### Describe the issue you are experiencing

HOAS attempts to contact sentry.io event monitoring. Integration is NOT installed.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

5.15.61-v8

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1.none specific -  instance reaching out to Sentry when the integration has never been installed
2. Observed attempted connections in logs of local dns service
3. See hoas logs below 
...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-11-18 07:27:09 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7faaf4b010>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
22-11-18 07:32:09 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fab7151e0>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
22-11-18 07:32:09 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7faafec8b0>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
22-11-18 07:32:09 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7faae45b70>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
22-11-18 07:32:21 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-11-18 07:37:09 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7faaf9fe20>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
22-11-18 07:37:09 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7faae46b60>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
22-11-18 07:37:09 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7faae458a0>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
22-11-18 07:42:10 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7faae459f0>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
22-11-18 07:42:10 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7faa1ca380>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
22-11-18 07:42:10 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7faa1c8c10>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
22-11-18 07:47:10 WARNING (raven-sentry.BackgroundWorker) [urllib3.connectionpool] Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7faaf20850>: Failed to establish a new connection: [Errno 111] Connection refused')': /api/5370612/envelope/
```

### Anything in the Host logs that might be useful for us?

```txt
INFO] 127.0.0.1:51208 - 27587 "NS IN . udp 17 false 512" NOERROR - 0 30.000635213s
[ERROR] plugin/errors: 2 . NS: dial tcp 1.0.0.1:853: i/o timeout
[INFO] 127.0.0.1:54899 - 24539 "NS IN . udp 17 false 512" NOERROR - 0 30.000687194s
[ERROR] plugin/errors: 2 . NS: dial tcp 1.0.0.1:853: i/o timeout
[INFO] 127.0.0.1:52737 - 21417 "NS IN . udp 17 false 512" NOERROR - 0 30.000665546s
[ERROR] plugin/errors: 2 . NS: dial tcp 1.1.1.1:853: i/o timeout
[INFO] 127.0.0.1:47287 - 56992 "NS IN . udp 17 false 512" NOERROR - 0 30.000569899s
[ERROR] plugin/errors: 2 . NS: dial tcp 1.0.0.1:853: i/o timeout
[INFO] 127.0.0.1:49606 - 3994 "NS IN . udp 17 false 512" NOERROR - 0 30.00071836s
```

### Additional information

[image]

ovenslighted08, 2022-11-17:
Follow-up why hoas is attempting to query cloud flare dns directly? IE, ignoring internal dns provided by dhcp.

agners, 2022-11-21:
This is likely a connection attempt made by Supervisor.

Can you provide more information about the versions of  your system?