home-assistant/operating-system #2238: h264 hardware acceleration in broken after 5.15.61 kernel update

hackneyb, 2022-11-20:
### Describe the issue you are experiencing

The Frigate add-on makes use of hardware acceleration for h264 camera streams.  Today, with a new install of HAOS on RPi4, the Frigate add-on fails to use ffmpeg 5.1 in its container to decode the h264 stream when this hw acceleration setting is enabled.  I'm experiencing this issue, and at least a couple HAOS users posting in other projects - links below.  

There are issues already open, describing this issue in the Frigate and the RPi linux projects:
Frigate https://github.com/blakeblackshear/frigate/issues/3780
 and 
RPi Linux https://github.com/raspberrypi/linux/issues/5150

Based on above threads, the issue does not only impact Frigate - it impacts using hardware acceleration for h264 in any scenario.  The fix appears to be updating ffmpeg in HAOS but I could be wrong in interpretation.  Need someone on the HAOS team to weigh in if there is work to be done in your project.  If not, should this issue be open to update the dependency in HAOS, to the new working ffmpeg release?

Specifically for bare metal RPi HAOS installs, there is a manual workaround now documented, that fixes the issue for at least a couple users, but it involves updating the container running Frigate and it doesn't survive the add-on restart - HAOS - specific instructions from https://github.com/jherby2k - https://github.com/blakeblackshear/frigate/issues/3780#issuecomment-1310468432

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install Frigate 0.11.1 add-on on RPi4 HAOS 9.3, fully updated
2. Configure an h264 camera stream in Frigate (but not hardware acceleration) - this works fine
3. Enable h264_v4l2m2m hardware acceleration OR drm hardware acceleration options - this causes ffmpeg crashes and "green screen" when viewing the camera stream
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No, there are no relevant messages in the Supervisor log.
```

### Anything in the Host logs that might be useful for us?

```txt
No, there are no relevant messages in the Host log.
```

### System information

There are currently no repairs available

### Additional information

My system:

- arm64 RPi4
- Home Assistant OS 9.3
- Home Assistance Core 2022.11.3
- Linux kernel 5.15.61-v8 aarch64
- Frigate Home Assistant Add-on version 0.11.1
- config.txt options:
`[pi4] 
dtoverlay=vc4-fkms-v3d,cma-512 
dtoverlay=cma,cma-size=536870912 
max_framebuffers=2 
gpu_mem=76 
dtoverlay=rpivid-v4l2 `

Frigate log (sanitized) when reproducing the issue by using Frigate "hwaccel_args: -c:v h264_v4l2m2m" or "hwaccel_args: -hwaccel drm":

`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[2022-11-19 17:30:43] frigate.app                    INFO    : Starting Frigate (0.11.1-2eada21)
[2022-11-19 17:30:43] frigate.app                    INFO    : Creating directory: /tmp/cache
Starting migrations
[2022-11-19 17:30:43] peewee_migrate                 INFO    : Starting migrations
There is nothing to migrate
[2022-11-19 17:30:43] peewee_migrate                 INFO    : There is nothing to migrate
[2022-11-19 17:30:43] detector.cpu2                  INFO    : Starting detection process: 216
[2022-11-19 17:30:43] frigate.edgetpu                WARNING : CPU detectors are not recommended and should only be used for testing or for trial purposes.
[2022-11-19 17:30:43] frigate.app                    INFO    : Output process started: 219
[2022-11-19 17:30:43] detector.cpu3                  INFO    : Starting detection process: 218
[2022-11-19 17:30:43] frigate.edgetpu                WARNING : CPU detectors are not recommended and should only be used for testing or for trial purposes.
[2022-11-19 17:30:43] frigate.app                    INFO    : Camera processor started for Camera01: 222
[2022-11-19 17:30:43] ws4py                          INFO    : Using epoll
[2022-11-19 17:30:43] frigate.app                    INFO    : Capture process started for Camera01: 225
[2022-11-19 17:30:43] ws4py                          INFO    : Using epoll
[2022-11-19 17:30:45] frigate.video                  ERROR   : Camera01: Unable to read frames from ffmpeg process.
[2022-11-19 17:30:45] frigate.video                  ERROR   : Camera01: ffmpeg process is not running. exiting `

and later:

`[2022-11-19 18:22:13] watchdog.Camera01     ERROR   : Ffmpeg process crashed unexpectedly for Camera01.
[2022-11-19 18:22:13] watchdog.Camera01     ERROR   : The following ffmpeg logs include the last 100 lines prior to exit.
[2022-11-19 18:22:13] ffmpeg.Camera01.detect ERROR   : Device creation failed: -14.
[2022-11-19 18:22:13] ffmpeg.Camera01.detect ERROR   : [h264 @ 0x55bcf77340] No device available for decoder: device type drm needed for codec h264.
[2022-11-19 18:22:13] ffmpeg.Camera01.detect ERROR   : Device setup failed for decoder on input stream #0:0 : Bad address`

hackneyb, 2022-11-20:
Based on comments in another thread, if HAOS updates its dependency to ffmpeg 5.1.2 the issue may be solved.

agners, 2022-11-21:
Hm, 5.15.61 came in with HAOS 9.0.rc2, so since quite a while.

> Based on above threads, the issue does not only impact Frigate - it impacts using hardware acceleration for h264 in any scenario. The fix appears to be updating ffmpeg in HAOS but I could be wrong in interpretation.

HAOS itself does not ship ffmpeg. The add-ons using ffmpeg/hardware acceleration (such as Frigate) need to provide (and do) their own version of ffmpeg. Hence the add-on requires to update ffmpeg.

closed 2022-11-21 by agners