home-assistant/operating-system #2254: Google Coral device disappears after a while

FS1961, 2022-12-05:


agners, 2022-12-09:
Can you check the host logs when it happens (or `dmesg` in the shell).

Also check the logs of your host system, if the USB device disappears there too. I have seen previous reports of such behavior, and using just a different (root) port of the host device helped.

FS1961, 2023-02-07:
So I decided to move the VM to a Win10 host running VirtualBox, not the best situation but it's working there. I think the MacPro is just too old running High Sierra so I might just install Debian/Ubuntu and see what happens there.

closed 2023-02-07 by FS1961

apadua, 2023-07-22:
I have an ESX server running a VM of Ubuntu 22.04 LTS. I have passed through a PCI USB controller. I have plugged in my Coral to that. It also disappears after a while - and it's not running Home Assistant - just frigate. Exactly the same as you - it's on lsusb, then it isn't.

I'm assuming this isn't HA specific, and thanks to your thread i now think it isn't ESX issue either (of course, these could be two separate issues altogether). But figured i'd share since this feels odd.

Cheers!

rayzorben, 2023-09-15:
I came to this thread because it was happening on my rPI4 as well. So I actually moved over to x86_64 completely where I run HA in a virtual machine in unRAID, so completely different setup. The Coral is also disappearing on there at the host level. I had a spare Coral that I know was working and I put that in its place and it STILL loses it. Makes me think it is a kernel issue.

edibledog, 2023-10-13:
This is happening to me too on proxmox - I don't think its related to HA

root@pve1:~# lsusb
Bus 001 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 009: ID 18d1:9302 Google Inc. 
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 005: ID 12d1:1001 Huawei Technologies Co., Ltd. E161/E169/E620/E800 HSDPA Modem
Bus 002 Device 003: ID 0bda:a725 Realtek Semiconductor Corp. Bluetooth 5.1 Radio
Bus 002 Device 002: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@pve1:~# lsusb
Bus 001 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 005: ID 12d1:1001 Huawei Technologies Co., Ltd. E161/E169/E620/E800 HSDPA Modem
Bus 002 Device 003: ID 0bda:a725 Realtek Semiconductor Corp. Bluetooth 5.1 Radio
Bus 002 Device 002: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub