home-assistant/operating-system #2256: aarch image converted from img to vhdx not working on Hyper-V ARM64 (stuck at waiting for root device)

ms300, 2022-12-07:
### Describe the issue you are experiencing

When i convert aarch64 images from img to vhdx and run it from Hyper-V,
the screen stucks at **waiting for root device**
[image]

i've tried 8.4 9.4 and 9.3, none of them were working. They're all stucked at **waiting for root device**

### What operating system image do you use?

generic-aarch64 (Generic UEFI capable aarch64 systems)

### What version of Home Assistant Operating System is installed?

8.4-9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.convert vmdk, img or qcow2 to vhdx  using qemu-img
2.set the vm following https://www.home-assistant.io/installation/windows/
3.press run

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### Additional information

Device: Windows Dev Kit 2023(https://www.microsoft.com/en-us/d/windows-dev-kit-2023/94k0p67w7581)

agners, 2022-12-09:
It seems the Linux kernel cannot access the disk drive. What type of disk (controller) emulation are you using?

dro159, 2022-12-09:
@ms300 Did you find a work around yet by chance? I have the exact same issue on the exact same device but didn't use qemu to convert. I used StarWind V2V Converter.

ms300, 2022-12-09:
> It seems the Linux kernel cannot access the disk drive. What type of disk (controller) emulation are you using?

SCSI controller.
IDE controller is not allowed in generation 2 VM which is the only choice for windows 11 arm64.

agners, 2022-12-10:
Do you happen to know what type of SCSI controller Hyper-V is emulating?

agners, 2022-12-10:
It seems that Hyper-V is supporting a couple of SCSI controllers, I wonder which one it's using by default:
https://learn.microsoft.com/en-us/powershell/module/virtualmachinemanager/new-scvirtualscsiadapter?view=systemcenter-ps-2022#-scsicontrollertype

ms300, 2022-12-10:
@agners I've tried to figure out the type of scsi controller but `New-SCVirtualScsiAdapter` is a PS command for System Center 2022. This command is not working on Windows 11 Pro. 
I tried to extract VM configs from vmcx file and the keyword relavant to SCSI controller is `name: Synthetic SCSI Controller` , `storage_allow_full_scsi_command_set ` , `disk_merge_pending `  and `Version`.

agners, 2022-12-10:
Hm, I see, so its using the paravirtualization drivers. Somehow I thought they don't compile for aarch64, but it seems they do. Will enable them.

dro159, 2022-12-12:
> Hm, I see, so its using the paravirtualization drivers. Somehow I thought they don't compile for aarch64, but it seems they do. Will enable them.

Tested the 10.0 development build and can confirm its working!

ms300, 2022-12-13:
With 10.0 dev20221210 it works on my Hyper-V ARM64 too. Great job!

closed 2022-12-13 by ms300