home-assistant/operating-system #2264: 9.4 introduced intermittend device state changes to unavailable

DarkNinja77, 2022-12-13:
### Describe the issue you are experiencing

Here: philips hue integrated devices randomly become unavailable and then available again within ~30 minutes after booting 9.4. 
After downgrade to 9.3, the problem disappeared.
Hardware : RPi 3B, hue bridge. 

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Just wait
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A after downgrade
```

### Anything in the Host logs that might be useful for us?

```txt
N/A after downgrade
```

agners, 2022-12-13:
Hm, kinda hard to tell what is wrong without logs. Can you run an upgrade again and upload logs?

DarkNinja77, 2022-12-14:
I updated to HAOS 9.4 again and could not reproduce the problem. Running 4 hours without the bug now.

Overnight, my HA core updated to 2022.12.5. Maybe that did it or something else. I have no idea, I got nothing.

DarkNinja77, 2022-12-15:
Cannot reproduce after HA core update.

closed 2022-12-15 by DarkNinja77

agners, 2022-12-15:
Ok, so might have been coincidence. Thanks for the update!