home-assistant/operating-system #2265: Bluetooth USB Adapter no longer recognized by HAOS after 9.4 update

Dvalin21, 2022-12-14:
### Describe the issue you are experiencing

After the latest updates, HAOS does not see my Bluetooth USB Adapter. The host system still sees it so its not a driver issue on the host behalf. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Boot into HAOS

### Anything in the Supervisor logs that might be useful for us?

```txt
22-12-13 23:14:24 INFO (MainThread) [supervisor.backups.manager] Creating new full backup with slug efd6cb00
22-12-13 23:14:24 INFO (MainThread) [supervisor.backups.manager] Backing up efd6cb00 store Add-ons
22-12-13 23:14:25 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on a0d7b954_vscode
22-12-13 23:14:25 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon a0d7b954_vscode
22-12-13 23:14:25 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on core_ssh
22-12-13 23:14:25 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon core_ssh
22-12-13 23:14:26 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on 15d21743_samba_backup
22-12-13 23:14:26 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon 15d21743_samba_backup
22-12-13 23:14:26 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on cebe7a76_hassio_google_drive_backup
22-12-13 23:14:26 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon cebe7a76_hassio_google_drive_backup
22-12-13 23:14:28 INFO (MainThread) [supervisor.homeassistant.module] Backing up Home Assistant Core config folder
22-12-13 23:14:54 INFO (MainThread) [supervisor.homeassistant.module] Backup Home Assistant Core config folder done
22-12-13 23:14:54 INFO (MainThread) [supervisor.backups.manager] Backing up efd6cb00 store folders
22-12-13 23:14:54 INFO (SyncWorker_4) [supervisor.backups.backup] Backing up folder share
22-12-13 23:14:54 INFO (SyncWorker_4) [supervisor.backups.backup] Backup folder share done
22-12-13 23:14:54 INFO (SyncWorker_0) [supervisor.backups.backup] Backing up folder addons/local
22-12-13 23:14:54 INFO (SyncWorker_0) [supervisor.backups.backup] Backup folder addons/local done
22-12-13 23:14:54 INFO (SyncWorker_7) [supervisor.backups.backup] Backing up folder ssl
22-12-13 23:14:54 INFO (SyncWorker_7) [supervisor.backups.backup] Backup folder ssl done
22-12-13 23:14:54 INFO (SyncWorker_3) [supervisor.backups.backup] Backing up folder media
22-12-13 23:14:54 INFO (SyncWorker_3) [supervisor.backups.backup] Backup folder media done
22-12-13 23:14:55 INFO (MainThread) [supervisor.backups.manager] Creating full backup with slug efd6cb00 completed
22-12-13 23:14:55 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
22-12-13 23:14:55 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
22-12-13 23:14:55 INFO (MainThread) [supervisor.api.middleware.security] /backups/efd6cb00/info access from cebe7a76_hassio_google_drive_backup
22-12-13 23:14:57 INFO (MainThread) [supervisor.api.middleware.security] /backups/27b1f162 access from cebe7a76_hassio_google_drive_backup
22-12-13 23:14:57 INFO (MainThread) [supervisor.backups.manager] Removed backup file 27b1f162
22-12-13 23:14:57 INFO (MainThread) [supervisor.api.middleware.security] /backups/efd6cb00/download access from cebe7a76_hassio_google_drive_backup
22-12-13 23:14:57 INFO (MainThread) [supervisor.api.backups] Downloading backup efd6cb00
22-12-13 23:15:17 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
22-12-13 23:15:17 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
22-12-13 23:15:18 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
22-12-13 23:15:18 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.NO_CURRENT_BACKUP/ContextType.SYSTEM
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-12-13 23:35:26 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
[... 135 lines cut ...]
version_api | ok
installed_addons | Studio Code Server (5.4.1), Terminal & SSH (9.6.1), Samba Backup (5.2.0), Home Assistant Google Drive Backup (0.109.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 4
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | December 5, 2022 at 4:13 AM
-- | --
current_recorder_run | December 13, 2022 at 3:18 PM
estimated_db_size | 122.57 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

Dvalin21, 2022-12-16:
> It seems you are using the `ova` image (not `generic-x86-64`), I changed your report.
> 
> What Hypervisor are you using? Can you remove/readd the pass through and see if that helps? Can you check the Host logs (the logs you've provided under Host logs are not from the Host section).

Im using the VirtualMachine app in Synology. 
Here is the host log
Dec 16 02:44:09 homeassistant systemd[1]: Starting Time & Date Service...
Dec 16 02:44:09 homeassistant systemd[1]: Started Time & Date Service.
Dec 16 02:44:39 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Dec 16 02:44:39 homeassistant kernel: audit: type=1334 audit(1671158679.170:601): prog-id=0 op=UNLOAD
Dec 16 02:44:39 homeassistant kernel: audit: type=1334 audit(1671158679.170:602): prog-id=0 op=UNLOAD
Dec 16 02:44:39 homeassistant kernel: audit: type=1334 audit(1671158679.170:603): prog-id=0 op=UNLOAD
Dec 16 02:44:39 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
Dec 16 02:44:39 homeassistant kernel: audit: type=1334 audit(1671158679.314:604): prog-id=0 op=UNLOAD
Dec 16 02:44:39 homeassistant kernel: audit: type=1334 audit(1671158679.314:605): prog-id=0 op=UNLOAD
Dec 16 02:44:39 homeassistant kernel: audit: type=1334 audit(1671158679.314:606): prog-id=0 op=UNLOAD
Dec 16 02:46:54 homeassistant systemd[1]: docker-eed433c6d6f1f67f126024dbe3633ae5b1a040be5172179e2dbbfcce77f5892b.scope: Deactivated successfully.
Dec 16 02:46:54 homeassistant systemd[1]: docker-eed433c6d6f1f67f126024dbe3633ae5b1a040be5172179e2dbbfcce77f5892b.scope: Consumed 12min 55.075s CPU time.
Dec 16 02:46:54 homeassistant kernel: audit: type=1334 audit(1671158814.053:607): prog-id=0 op=UNLOAD
Dec 16 02:46:54 homeassistant dockerd[410]: time="2022-12-16T02:46:54.057548882Z" level=info msg="ignoring event" container=eed433c6d6f1f67f126024dbe3633ae5b1a040be5172179e2dbbfcce77f5892b module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
Dec 16 02:46:54 homeassistant dockerd[418]: time="2022-12-16T02:46:54.058006703Z" level=info msg="shim disconnected" id=eed433c6d6f1f67f126024dbe3633ae5b1a040be5172179e2dbbfcce77f5892b
Dec 16 02:46:54 homeassistant dockerd[418]: time="2022-12-16T02:46:54.058063710Z" level=warning msg="cleaning up after shim disconnected" id=eed433c6d6f1f67f126024dbe3633ae5b1a040be5172179e2dbbfcce77f5892b namespace=moby
Dec 16 02:46:54 homeassistant dockerd[418]: time="2022-12-16T02:46:54.058083460Z" level=info msg="cleaning up dead shim"
Dec 16 02:46:54 homeassistant dockerd[418]: time="2022-12-16T02:46:54.067718587Z" level=warning msg="cleanup warnings time=\"2022-12-16T02:46:54Z\" level=info msg=\"starting signal loop\" namespace=moby pid=20305 runtime=io.containerd.runc.v2\n"
Dec 16 02:47:01 homeassistant dockerd[418]: time="2022-12-16T02:47:01.670318352Z" level=info msg="loading plugin \"io.containerd.event.v1.publisher\"..." runtime=io.containerd.runc.v2 type=io.containerd.event.v1
Dec 16 02:47:01 homeassistant dockerd[418]: time="2022-12-16T02:47:01.670422414Z" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
Dec 16 02:47:01 homeassistant dockerd[418]: time="2022-12-16T02:47:01.670483441Z" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Dec 16 02:47:01 homeassistant dockerd[418]: time="2022-12-16T02:47:01.670732681Z" level=info msg="starting signal loop" namespace=moby path=/run/docker/containerd/daemon/io.containerd.runtime.v2.task/moby/eed433c6d6f1f67f126024dbe3633ae5b1a040be5172179e2dbbfcce77f5892b pid=20323 runtime=io.containerd.runc.v2
Dec 16 02:47:01 homeassistant systemd[1]: Started libcontainer container eed433c6d6f1f67f126024dbe3633ae5b1a040be5172179e2dbbfcce77f5892b.
Dec 16 02:47:01 homeassistant kernel: audit: type=1334 audit(1671158821.718:608): prog-id=248 op=LOAD
Dec 16 02:47:01 homeassistant kernel: audit: type=1300 audit(1671158821.718:608): arch=c000003e syscall=321 success=yes exit=14 a0=5 a1=c000195a30 a2=78 a3=c00026cf98 items=0 ppid=20323 pid=20334 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
Dec 16 02:47:01 homeassistant kernel: audit: type=1327 audit(1671158821.718:608): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F7661722F72756E2F646F636B65722F636F6E7461696E6572642F6461656D6F6E2F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F65656434333363
Dec 16 02:47:01 homeassistant kernel: audit: type=1334 audit(1671158821.718:609): prog-id=249 op=LOAD
Dec 16 02:47:01 homeassistant kernel: audit: type=1300 audit(1671158821.718:609): arch=c000003e syscall=321 success=yes exit=16 a0=5 a1=c0001957c8 a2=78 a3=c00026cfe8 items=0 ppid=20323 pid=20334 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
Dec 16 02:47:01 homeassistant kernel: audit: type=1327 audit(1671158821.718:609): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F7661722F72756E2F646F636B65722F636F6E7461696E6572642F6461656D6F6E2F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F65656434333363
[... cut ...]
Dec 16 03:15:19 homeassistant kernel: IPv6: ADDRCONF(NETDEV_CHANGE): vethe406882: link becomes ready
Dec 16 03:15:19 homeassistant kernel: hassio: port 6(vethe406882) entered blocking state
Dec 16 03:15:19 homeassistant kernel: hassio: port 6(vethe406882) entered forwarding state
Dec 16 03:15:19 homeassistant NetworkManager[353]: <info>  [1671160519.5658] device (vethe406882): carrier: link connected

I just installed the new core update (not really sure if its associated or not) and then I disconnected and reconnected. Now everything is working. Closing

Dvalin21, 2022-12-17:
So after installing all updates to the HAOS, Bluetooth integration will kill the service and then fail to restart the service with the adapter. 

Update of light.keith_s_lamp is taking over 10 seconds
7:35:31 PM – (WARNING) helpers/entity.py - message first occurred at December 15, 2022 at 11:30:45 PM and shows up 1099 times
hci0 (08:BE:AC:30:D6:60): Failed to restart Bluetooth scanner: hci0 (08:BE:AC:30:D6:60): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
7:35:28 PM – (ERROR) Bluetooth - message first occurred at 11:28:03 AM and shows up 976 times
Bluetooth adapter hci0 [08:BE:AC:30:D6:60] could not be reset due to timeout
7:35:26 PM – (WARNING) components/bluetooth/util.py - message first occurred at 11:28:00 AM and shows up 1951 times
Connection: BleakError: F8:24:41:E6:5B:1C - F8:24:41:E6:5B:1C: Failed to connect:
7:35:07 PM – (ERROR) Yeelight bluetooth (custom integration) - message first occurred at December 15, 2022 at 11:31:54 PM and shows up 1548 times
Updating yeelight_bt light took longer than the scheduled update interval 0:00:30
7:34:51 PM – (WARNING) Light - message first occurred at December 15, 2022 at 11:32:35 PM and shows up 1130 times

agners, 2022-12-20:
And this started with HAOS 9.3? Can you try downgrading to see if the problem indeed disappears?

```
ha os update --version 9.3
```

Dvalin21, 2022-12-21:
> And this started with HAOS 9.3? Can you try downgrading to see if the problem indeed disappears?
> 
> ```
> ha os update --version 9.3
> ```

Yes this it did, I didn't see the message before I updated to 9.4, but here is what I'm getting now
Updating yeelight_bt light took longer than the scheduled update interval 0:00:30
2:14:00 PM – (WARNING) Light - message first occurred at 1:21:30 PM and shows up 16 times
Update of light.keith_s_lamp is taking over 10 seconds
2:13:10 PM – (WARNING) helpers/entity.py - message first occurred at 12:33:40 PM and shows up 17 times
Connection: BleakError: F8:24:41:E6:5B:1B - F8:24:41:E6:5B:1B: Failed to connect: [org.bluez.Error.Failed] le-connection-abort-by-local: Interference/range; External Bluetooth adapter w/extension may help; Extension cables reduce USB 3 port interference
2:09:05 PM – (ERROR) Yeelight bluetooth (custom integration) - message first occurred at 12:34:11 PM and shows up 6 times
Failed to cancel connection (/org/bluez/hci0/dev_F8_24_41_E6_5B_1C):
1:44:03 PM – (WARNING) components/bluetooth/wrappers.py
AP-91054500ACE0: Device unexpectedly disconnected; RSSI: -46
12:33:57 PM – (WARNING) runner.py - message first occurred at 12:33:48 PM and shows up 3 times
Uncaught thread exception
12:33:45 PM – (ERROR) /usr/src/homeassistant/homeassistant/bootstrap.py
error while get mqtt config
12:33:45 PM – (ERROR) /usr/local/lib/python3.10/site-packages/tuya_iot/openmq.py

Dvalin21, 2022-12-21:
> And this started with HAOS 9.3? Can you try downgrading to see if the problem indeed disappears?
> 
> ```
> ha os update --version 9.3
> ```

Downgrading now, will let you know.

Dvalin21, 2022-12-21:
> And this started with HAOS 9.3? Can you try downgrading to see if the problem indeed disappears?
> 
> ```
> ha os update --version 9.3
> ```

Also if it helps at all, here is the adapter that I'm using attached to my synology nas https://www.amazon.com/Edimax-BT-8500-Bluetooth-Supports-Controllers/dp/B08M1VJHVD/ref=sr_1_3?crid=1118BD8KO2U1Y&keywords=edimax+usb+bluetooth+adapter&qid=1671654933&s=electronics&sprefix=Edimax+usb+blu%2Celectronics%2C141&sr=1-3

Dvalin21, 2022-12-22:
This is what I got from the core logs

Update of light.keith_s_lamp is taking over 10 seconds
4:08:18 PM – (WARNING) helpers/entity.py - message first occurred at 2:31:18 PM and shows up 24 times
Connection: BleakError: F8:24:41:E6:5B:1C - F8:24:41:E6:5B:1C: Failed to connect: [org.bluez.Error.Failed] le-connection-abort-by-local: Interference/range; External Bluetooth adapter w/extension may help; Extension cables reduce USB 3 port interference
4:07:43 PM – (ERROR) Yeelight bluetooth (custom integration) - message first occurred at 2:32:02 PM and shows up 17 times
Updating yeelight_bt light took longer than the scheduled update interval 0:00:30
4:07:08 PM – (WARNING) Light - message first occurred at 2:32:38 PM and shows up 31 times

Dvalin21, 2022-12-22:
So last night all the lights that were bluetooth were connected and still able to control. Today none of them are even though the bluetooth adapter is recognized. (Still on 9.3) Was there an update to bleak or bluez? Is it possible the issue is there?

Connection: BleakError: F8:24:41:E6:5B:1C - F8:24:41:E6:5B:1C: Failed to connect:
1:12:46 PM – (ERROR) Yeelight bluetooth (custom integration) - message first occurred at December 21, 2022 at 2:32:02 PM and shows up 1323 times
Updating yeelight_bt light took longer than the scheduled update interval 0:00:30
1:12:40 PM – (WARNING) Light - message first occurred at December 21, 2022 at 2:32:38 PM and shows up 1441 times
hci0 (08:BE:AC:30:D6:60): Failed to restart Bluetooth scanner: hci0 (08:BE:AC:30:D6:60): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
1:12:32 PM – (ERROR) Bluetooth - message first occurred at 8:33:07 AM and shows up 560 times
Bluetooth adapter hci0 [08:BE:AC:30:D6:60] could not be reset due to timeout
1:12:30 PM – (WARNING) components/bluetooth/util.py - message first occurred at 8:33:05 AM and shows up 1119 times
Update of light.keith_s_lamp is taking over 10 seconds
1:12:20 PM – (WARNING) helpers/entity.py - message first occurred at December 21, 2022 at 2:31:18 PM and shows up 1065 times
Attempting to login again.
12:56:12 PM – (ERROR) /usr/local/lib/python3.10/site-packages/somecomfort/client.py - message first occurred at December 21, 2022 at 2:46:11 PM and shows up 10 times
Connection Error occurred.
12:56:12 PM – (ERROR) /usr/local/lib/python3.10/site-packages/somecomfort/client.py - message first occurred at December 21, 2022 at 2:46:11 PM and shows up 10 times
Error fetching AP-91054500ACE0 data: AP-91054500ACE0 - 91:05:45:00:AC:E0: Failed to connect: [org.bluez.Error.Failed] le-connection-abort-by-local: Interference/range; External Bluetooth adapter w/extension may help; Extension cables reduce USB 3 port interference
8:33:49 AM – (ERROR) LED BLE - message first occurred at December 21, 2022 at 3:03:11 PM and shows up 26 times
Unexpected error while attempting USB reset of hci0: [Errno 19] No such device
8:33:06 AM – (ERROR) /usr/local/lib/python3.10/site-packages/bluetooth_auto_recovery/recover.py
AP-91054500ACE0: Device unexpectedly disconnected; RSSI: -48
8:33:00 AM – (WARNING) runner.py - message first occurred at December 21, 2022 at 2:31:09 PM and shows up 1080 times
Detected integration that calls media_source.async_resolve_media without passing an entity_id. Please report issue for media_source using this method at homeassistant/components/media_source/__init__.py, line 147: report("calls media_source.async_resolve_media without passing an entity_id")
8:22:14 AM – (WARNING) helpers/frame.py
Failed to cancel connection (/org/bluez/hci0/dev_F8_24_41_E6_5B_1B):
8:02:16 AM – (WARNING) components/bluetooth/wrappers.py - message first occurred at December 21, 2022 at 2:31:51 PM and shows up 198 times
Error doing job: Task was destroyed but it is pending!
7:58:46 AM – (ERROR) runner.py - message first occurred at 6:29:56 AM and shows up 3 times
Error fetching roku data: Invalid response from API: Timeout occurred while connecting to device
7:33:19 AM – (ERROR) Roku - message first occurred at December 21, 2022 at 3:57:31 PM and shows up 19 times

Saentist, 2023-04-04:
2 virtual machines 2 different Bluetooth adapters same situation.
Some kernel problem with cannot restart adapter.
```
Home Assistant 2023.3.6
Supervisor 2023.03.3
Operating System 9.5
```

Saentist, 2023-04-13:
@pvizeli @agners @rbray89 @frenck and other contributor's and members of https://github.com/orgs/home-assistant/people
When this bug will be fixed?

Saentist, 2023-04-14:
@agners 
If not added all this people no one will see this issue for long time.

Bluetooth work in virtualization on any other OS with I used so this is not a issue and work, even on kernel 6+

Log and access to HAOS Virtual images is not so simple as o HA installed in regular OS
/HAOS need reporting tool not all users are experienced Linux users/

there is no `update-pciids` tool
so guess with is USB Bluetooth
```shell
➜  ~ lsusb
Bus 002 Device 002: ID 0e0f:0003
Bus 001 Device 001: ID 1d6b:0002
Bus 002 Device 004: ID 1a86:55d4
Bus 002 Device 003: ID 0e0f:0002
Bus 002 Device 001: ID 1d6b:0001
Bus 002 Device 007: ID 0a12:0001
```
[image]

[image]

----

p.s. in beta channel work

closed 2023-07-14 by Dvalin21