home-assistant/operating-system #2267: [rpi4] bluetooth is off after reboot

regevbr, 2022-12-14:
### Describe the issue you are experiencing

After updating to 9.4, bluetooth is powered off after every reboot. The only way to start it again is to manually run `bluetoothctl power on`

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. reboot
2. bluetoothctl scan on

### Anything in the Supervisor logs that might be useful for us?

```txt
nope
```

### Anything in the Host logs that might be useful for us?

```txt
nope
```

### System information

## System Information

version | core-2022.12.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.76-v8
arch | aarch64
timezone | Asia/Jerusalem
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4442
Installed Version | 1.28.4
Stage | running
Available Repositories | 1160
Downloaded Repositories | 44

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.11.2
agent_version | 1.4.1
docker_version | 20.10.19
disk_total | 228.5 GB
disk_used | 71.6 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | AdGuard Home (4.7.5), Home Assistant Google Drive Backup (0.109.2), MariaDB (2.5.1), Mosquitto broker (6.1.3), NGINX Home Assistant SSL proxy (3.2.0), Node-RED (14.0.0), Node-RED Dashboard (1.3.0), Plex Media Server (3.1.2), Studio Code Server (5.5.0), Transmission (dev), Zigbee2MQTT (1.28.4-1), Glances (0.17.1), InfluxDB (4.5.0), Grafana (8.1.0), phpMyAdmin (0.8.2), pigpio (1.5.2), Nuki Bridge (0.0.23), room-assistant (beta) (3.0.0-beta.5), ESPHome (2022.12.0), RTSPtoWeb - WebRTC (1.4.0), Duck DNS (1.15.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 31
views | 9
mode | storage

</details>

<details><summary>MiWifi</summary>

version | 2.8.2
-- | --
192.168.31.1 (R2350) | ok

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | December 12, 2022 at 07:16
-- | --
current_recorder_run | December 14, 2022 at 23:29
estimated_db_size | 423.89 MiB
database_engine | mysql
database_version | 10.6.8

</details>

agners, 2022-12-20:
I cannot reproduce this, after every reboot I get this:

```
# bluetoothctl scan on
AdvertisementMonitor path registered
```

Is maybe something else interacting with WiFi/Bluetooth in your system?

regevbr, 2022-12-20:
I have an addon that uses bluetooth but it doesn't turn it off... maybe there is a race condition that causes the Bluetooth controller to get stuck? How can I check that?

agners, 2022-12-20:
Maybe the host log have some information.

Can you try to disable the add-on, just to see if that makes a difference in behavior?

regevbr, 2022-12-20:
disabled and the issue is gone. Maybe a race condition? in any case I don't need that addon anymore so losing the issue. Thanks for the assistance :0)

closed 2022-12-20 by regevbr