home-assistant/operating-system #2268: Raspberry Pi 3B+ reboot fails with Startech USB adapter + SATA SSD

madbrain76, 2022-12-14:
### Describe the issue you are experiencing

Whenever I go into Home Assistant / Hardware / Reboot , the Raspberry Pi 3B+ fails to boot from my SATA SSD attached to a Startech SATA/USB bridge. The boot loader skips the USB device, and then tries to do a PXE network boot from my Comcast router (sigh). If I cut off power to the Pi and restore it, it then boots properly from USB.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Power up Raspberry Pi 3B+
2. Boot Home Assistant from USB attached SSD
3. Go to Settings / System / Hardware
4. Go to vertical menu on the top right
5. Click "Reboot system"
6. Wait for system to reboot
7. Observe that the Pi 3B+ skips the USB device during the boot, and tries to do a PXE boot
8. Power off Pi 3B+
9. Power up Pi 3B+
10. Observe that HAOS boots properly from USB attached SSD

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

N/A

### Additional information

Issue does not occur when booting from SD card.
The USB / SATA SSD bridge I'm using is a Startech model USB3S2SAT3CB .
The SSD itself is a Patriot Burst Elite 120GB.

agners, 2022-12-20:
Unfortunately, we do see these type of errors a lot with Raspberry Pi & USB SSD. It is likely power or timing issue at startup of the USB SSD.

Can you try a powered USB hub between the Raspberry Pi and your USB SSD? What also would be interesting if this is reproducible with Raspberry Pi OS (on the very same hardware).

madbrain76, 2023-01-30:
I tried using a powered SATA dock from Inateck, and so far haven't had an issue with reboots. Looks like you are right and it's some sort of power/timing problem.

closed 2023-05-08 by github-actions[bot]