home-assistant/operating-system #2270: HA OS switching input on TV aggressively

mmalina, 2022-12-15:
### Describe the issue you are experiencing

Today I upgraded Home Assistant OS on my RPi 3b+ to the latest version (9.4) and now it keeps switching my TV to it (HDMI).

This is similar to what I reported in the past here: https://github.com/home-assistant/core/issues/54127

I have an Apple TV connected to another HDMI port on the TV and if I activate it via the Apple TV remote control, it will briefly switch to it, but then HA OS will switch it back to itself.

I'm pretty sure it must be HA OS doing this. I tried rebooting it, but that didn't change anything.

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install Home Assistant OS on RPi 3b+
2. Connect it to a TV
3. Connect some other device, e.g. an Apple TV to the other HDMI port on the TV
4. Start the TV and Home Assistant - it will automatically switch the input on the TV
5. Start Apple TV via their remote control - it will switch the TV input, but then switch back to show HA OS

### Anything in the Supervisor logs that might be useful for us?

```txt
No, nothing related to hdmi.
```

### Anything in the Host logs that might be useful for us?

```txt
No, nothing related to hdmi.
```

### System information

## System Information

version | core-2022.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.76-v7
arch | armv7l
timezone | Europe/Prague
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.11.2
agent_version | 1.4.1
docker_version | 20.10.19
disk_total | 109.3 GB
disk_used | 16.6 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.0), File editor (5.4.2), Dropbox Sync (1.3.0), Mosquitto broker (6.1.3), Terminal & SSH (9.6.1), deCONZ (6.17.1), Home Assistant Google Drive Backup (0.109.2), PI CEC (0.3), zabiny-temp (0.10)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 4
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | December 11, 2022 at 6:35 PM
-- | --
current_recorder_run | December 15, 2022 at 9:23 AM
estimated_db_size | 108.32 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

mmalina, 2022-12-15:
Note that I stopped using the pi cec add-on a month or so ago, it was just sitting there stopped. But ok, I deleted it. Afterwards, I rebooted HA OS once more. The issue persists.

The workaround for me is either unplug hdmi from raspberry (because I don't need it most of the time). Or if I switch the input using TV Remote, it stays on apple tv. But if I then turn the tv off using apple tv remote and turn it back on using the same, it will switch back to raspberry pi.

It's also possible that these two devices would always fight to get the focus, but somehow apple tv always won until now. Hard to say. Any way I can help debug this? I have ssh root access enabled, so maybe I can try to dig up some logs or something...

agners, 2022-12-15:
Hm, I see.

Did the behavior of this change with a new OS release? You can try going back to the last release using `ha os update --version 9.3`.

Another interesting test would be if the system behaves the same on Raspberry Pi OS (using the same hardware/cabling. We use Raspberry Pi firmware/kernel, I think the latest Raspberry Pi OS and latest HAOS use the very same version as well. I'd expect that the system should behave the same. If so, it might be worth checking with the Raspberry Pi folks.

mmalina, 2022-12-15:
This is really weird. First I started Apple TV just to confirm it's still happening. And it wasn't... So my thinking was: "Hmm, maybe this will only happen for a period of time after HA OS startup." So I rebooted HA OS. First boot hung on this: https://github.com/home-assistant/operating-system/issues/1712 OK, that hasn't happen in a while... Anyway, I unplugged power for a few seconds and tried again. Now it booted... And the hdmi switching issue is gone.

To be clear, I'm still on 9.4 - I haven't gotten to the downgrade yet, I wanted to confirm the issue once again before doing that.

Just for the fun of it, I did the downgrade. And then I did the upgrade to HA OS 9.4 again. And I still can't reproduce this :(

But before I filed this issue, I did like 3 reboots and it was happening constantly. And it did happen to me in the past, but also disappeared. Weird.

So I'm not sure there's anything to be done here. Maybe close this and I will reopen/create a new one if this happens again consistently?

agners, 2022-12-15:
Uh interesting.

Maybe a loose contact/bad cable?

> So I'm not sure there's anything to be done here. Maybe close this and I will reopen/create a new one if this happens again consistently? 

Yeah sounds good, I will close then for now, just ping again and we can reopen.

closed 2022-12-15 by agners

mmalina, 2022-12-15:
So I just came back home, I turn on Apple TV via the remote (which turns on the TV) and bam, I'm looking at HA console. I press the On button on the Apple TV remote again - it switches to Apple TV and then immediately to HA console. And again and again.

Anyway, with my luck I'm sure it will stop happening the moment I try debugging again. If HA OS is the same as Raspberry PI OS for the most part, then I assume it might be similar, but I'm not sure I want to reinstall the OS now especially since it seems a bit random.

> Maybe a loose contact/bad cable?

Doesn't seem likely, but who knows.