home-assistant/operating-system #2271: Add-on in privileged mode not sending data to GPIO

LeoCal, 2022-12-17:
### Describe the issue you are experiencing

Hello,

I'm facing an issue with HA OS to send a signal from an add-on container (running [Pilight](https://manual.pilight.org)) to the GPIO of a Raspberry PI 4. Although the add-on seems to be working fine (Pilight logs show the command is well received from the daemon running in the add-on), it seems the OS is somehow preventing the signal to get to the GPIO (I measured with a multimeter directly on the GPIO data PIN).

Suspecting an issue with the 433MHz module or the RPI itself, I've done the following experiments:
1) I installed Raspberry OS (Bullseye version) on the same RPI4, then installed Pilight directly on the OS via "apt install" -> all working fine
2) I installed Docker on Raspberry PI OS, then installed Pilight container (pretty much same Dockerfile content) -> all working fine 

In both cases, the 433MHz transmitter is correctly turning on/off 433MHz devices; moreover, my multimeter shows the signal makes it to the data PIN via GPIO.

But as I would really like to move to HA OS (instead of HA core), I loaded again HA OS and ran a few other experiments:
1) I've disabled protection mode for the Add-On, giving pretty much permission to access the entire system (yes, I know it's not a good practice, but I need to start troubleshooting somewhere) -> nothing, it doesn't work
2) I've manually run the "docker run ... --privileged ..." command that was working directly on Docker, and again, it doesn't work.

Is it a bug in the HA OS?
Or perhaps is that a known limitation?
I do you suggest going about it?

I'd really like to move to run HA OS instead of HA Core, but this is one of the last things I need to sort out before the move.

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Please see my full description of how to reproduce the issue above.

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

## System Information

version | core-2022.12.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.76-v7l
arch | armv7l
timezone | Europe/Zurich
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.11.2
agent_version | 1.4.1
docker_version | 20.10.19
disk_total | 219.4 GB
disk_used | 3.0 GB
healthy | true
supported | true
board | rpi4
supervisor_api | ok
version_api | ok
installed_addons | Pilight Server (1.7.2), File editor (5.4.2), Terminal & SSH (9.6.1), pilight (1.0.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 15 dicembre 2022, 20:09
-- | --
current_recorder_run | 17 dicembre 2022, 16:52
estimated_db_size | 1.54 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

agners, 2022-12-20:
HAOS uses the same kernel as Raspberry Pi OS, so it seems odd that this does work on Raspberry Pi OS and does not on HAOS.

Can you check the Linux kernel version you are using on Raspberry Pi OS (using `uname -a`?).

Are you using 32-bit version of Raspberry Pi OS as well?

What is surprising to me is that there is no error on Pilight. I would expect if something is preventing access to get some type of error from whichever process is trying to access the GPIO pins.

Over time, there have been different API's available to control the GPIO. Do you know what Pilight is using?

LeoCal, 2022-12-20:
Hi,

first of all, thanks for your reply.

Linux version of Raspberry PI OS:
Linux 5.15.76-v7l+ #1597 SMP Fri Nov 4 12:14:58 GMT 2022 armv7l GNU/Linux

Yes, it's a 32 bit version.

Pilight uses WiringX to access the GPIO of the device.

Cheers

LeoCal, 2023-03-21:
Sure, I’ll give it a try and get back 👍🏻

LeoCal, 2023-03-21:
I have just tried again, after updating to the latest OS, Supervisore and HA Core and I still see the issue.
What is weird is that the logs show transmission was fine, but the signal was not actually sent to the PI's GPIO.

Using regular Raspberry PI OS and running the same pilight (config, etc) in a docker container works just fine.

Could anyone help understanding how to troubleshoot this further?

LeoCal, 2023-06-16:
After fiddling a lot with this, I managed to find out the issue, so reporting here in case it might help other people in future. 

It turns out Pilight container in non-privileged mode also needs SYS_NICE on top of SYS_RAWIO. That, along with device "/dev/mem" did the trick for me!

closed 2023-06-16 by LeoCal