home-assistant/operating-system #2272: Stuck waiting for home assistant cli to be ready

mbroadst, 2022-12-18:
### Describe the issue you are experiencing

I flashed an Intel NUC (6th gen) with haos_generic-x86-64-9.4.img.xz this morning, and it has been stuck "Waiting for home assistant CLI to be ready..." for about an hour. The initial bios printout seems to indicate it's stuck trying to synchronize time (might be a DNS issue?),  but I'm unable to enter any sort of shell that gives me access to additional information.  I'm not able to switch tty's, and its not dropping me into the emergency shell after some time. The only thing I appear to be able to do is press "ctrl+c" which will result in a black screen for a few minutes before displaying the message again. I just tried reflashing the NUC with v9.3 of the operating-system, and it has the same behavior. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.4, 9.3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Boot up NUC with lubuntu live distro
2. Install balena etcher, and flash https://github.com/home-assistant/operating-system/releases/download/9.4/haos_generic-x86-64-9.4.img.xz
3. Start up NUC

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't access any logs
```

### Anything in the Host logs that might be useful for us?

```txt
Can't access any logs
```

mbroadst, 2022-12-18:
Some new details as I've been trying to debug this issue today. After a fresh install the boot process seems to stall out at two tasks:
- time synchronization
- something to do with ../by-label/haoss-data (maybe resizing?)

I thought maybe originally this was an issue with DNS and not being able to resolve `time.cloudflare.com`, but I've ensured my router is providing `1.1.1.1` and `8.8.8.8` as DNS servers now so there shouldn't be any curious blacklisting, or rate limiting. Another weird network-related clue is that when the machine first boots the ethernet appears to be functional (lights up on the router and the back of the NUC), when it goes through the init process the network shuts off, and when I eventually get to "Waiting for home assistant CLI to be ready..." the network is back on again.

I now think the issue is actually the initial step to set up `haoss-data`. When I boot the machine back on a usb stick and view the partition table for the disk, its showing me that `haoss-data` exists but its unformatted - is that expected?

ipm79, 2022-12-27:
I am having the exact same issue - start job for hassos-data is timing out and system is stuck on "waiting for CLI to be ready". I don't use a NUC but a Lenovo ThinkPad T540p laptop. Booting from a live Ubuntu USB drive shows a 255GiB partition at the end of the disk but without a filesystem.

McNamee-E, 2023-01-03:
Same situation. Install on Lenovo laptop from live Ubuntu using Balena-Etcher. Same failure, same log error.

almalino, 2023-01-13:
Same problem. Doing a fresh install on Intel Celeron mini PC and same problem. I tried to configure DNS manually but still same problem

wannes-ds, 2023-01-14:
Same issue on a Gigabyte BRIX GB-BMPD-6005 mini PC. Fan at full speed seems to indicate stuck in infinite loop. Sometimes the system does boot completely but after at most a couple of hours it freezes up completely.

mac-city, 2023-01-18:
same issue on proxmox

mbroadst, 2023-01-22:
@mac-city I'm not sure we're facing the same issue. I have no prompt to type "login" at, it's just stuck at a black screen which says "Waiting for the Home Assistant CLI to be ready...". The system time messages appear to be a red herring, the main issue seems to be that the `hassos-data` is never set up. 

@ipm79 It seems we have the same issue, have you made any further progress?

McNamee-E, 2023-01-22:
So I tried installing on 3 different Lenovo machines, all failed. Intel and AMD. Different firmware versions, every permutation I could muster.
Tried an HP laptop and bingo. No errors and up-and-running first time.

mbroadst, 2023-01-22:
Thanks @McNamee-E, those are helpful details.

I was able to take a picture of the boot logs before it drops into "Waiting for the Home Assistant CLI to be ready..." showing the failed systemd tasks:
[image]

You can see it fails to start the "Wait Until Kernel Time Sychronized" task, but I think the real problem is the next line:
```
Timed out waiting for device Samsung SSD 950 PRO 256GB hassos-data.
```

I found a few forum posts online ([here](https://bbs.archlinux.org/viewtopic.php?id=213548), and [here](https://unix.stackexchange.com/questions/544104/kubuntu-not-starting-up-after-cloning-hdd-to-ssd-dual-boot)) with people talking about a similar issue timing out waiting for a device, could there possibly be a corrupted fstab for this installation? 

Is it possible to drop into a single user shell with this image? I can't seem to do anything functional once "Waiting for the Home Assistant CLI to be ready..." is displayed.

hermanops, 2023-02-16:
Same issue on HA Yellow with a Samsung SSD 980 NVME. The filesystems on there didn't get mounted. After reinserting the NVME drive it worked again.

agners, 2023-02-16:
The operating system uses the file system label `hassos-data` to find the data partition.

One thing which can explain issue is if the Linux kernel is not able to communicate with the NVMe (missing driver/driver bug etc). The emergency console can help to debug this (check kernel logs `dmesg`, and check disk devices in `/dev`, probably `nvme*` in this case).

@hermanops so in your case it seems that the NVMe bad a bad contact for some reason. I hope it is resolved from here onwards.

hermanops, 2023-02-16:
Sadly it's back...
```
[  OK  ] Finished Load/Save Random Seed.
[  OK  ] Finished Apply Kernel Variables.
[  OK  ] Finished Retrigger udev rules after user rules mounted.
[ TIME ] Timed out waiting for device ▒▒dev/disk/by-label/hassos-data.
[DEPEND] Dependency failed for HassOS data resizing.
[DEPEND] Dependency failed for Home…t OS data partition migration.
[DEPEND] Dependency failed for HassOS data partition.
[DEPEND] Dependency failed for HassOS AppArmor.
[DEPEND] Dependency failed for HassOS supervisor.
[DEPEND] Dependency failed for Dock… Application Container Engine.
[DEPEND] Dependency failed for Docker persistent data.
[DEPEND] Dependency failed for Systemd persistent logging.
[DEPEND] Dependency failed for Flus…Journal to Persistent Storage.
[DEPEND] Dependency failed for Grow File System on /mnt/data.
[DEPEND] Dependency failed for File…dev/disk/by-label/hassos-data.
[  OK  ] Reached target HassOS overlay targets.
[  OK  ] Reached target Local File Systems.
```

Kernel says:
```
[Thu Feb 16 17:24:06 2023] nvme nvme0: Device not ready; aborting reset, CSTS=0x1
[Thu Feb 16 17:24:06 2023] nvme nvme0: Removing after probe failure status: -19
```
I'm running kernel:

Linux homeassistant 5.15.84-v8 #1 SMP PREEMPT Tue Jan 24 22:45:16 UTC 2023 aarch64 HAOS

viktor-c, 2023-02-25:
Got same on a Raspi4 with 2Gb. Tried both 64bit and 32 bit HAOS 9.5.

lstep, 2023-03-08:
> Some new details as I've been trying to debug this issue today. After a fresh install the boot process seems to stall out at two tasks:
> 
> * start job is running for Wait Until Kernel Time Synchronized
> * start job is running for /dev/disk/by-label/hassos-data
> 
> I thought maybe originally this was an issue with DNS and not being able to resolve `time.cloudflare.com`, but I've ensured my router is providing `1.1.1.1` and `8.8.8.8` as DNS servers now so there shouldn't be any curious blacklisting, or rate limiting. Another weird network-related clue is that when the machine first boots the ethernet appears to be functional (lights up on the router and the back of the NUC), when it goes through the init process the network shuts off, and when I eventually get to "Waiting for home assistant CLI to be ready..." the network is back on again.
> 
> I now think the issue is actually the initial step to set up `hassos-data`. Eventually the timeout says that start job timed out waiting for device. When I boot the machine back on a usb stick and view the partition table for the disk, its showing me that `hassos-data` exists but its unformatted - is that expected?

Exactly same problem/behaviour. I'm on a noname Intel NUC, with a SSD drive. Ethernet lights are on until I get the UEFI SLOT (A) prompt, and just after, it goes dark, and fails that NTP thing. After a long time (15 or so minutes) I get to a prompt (emergency as ha cli didn't manage to start).

agners, 2023-03-15:
I've come across a problem in the data disk move script, if triggered twice it renders the system unusable which causes exactly the behavior described here (see #2412). However, triggering the data disk move script should not be possible through Supervisor. But maybe there is a bug :man_shrugging:  The code on Supervisor side is currently refactured, so I hope this won't be possible in the future anymore. Otherwise, #2412 will avoid issues on OS side.

That said, this can also happen when renaming data partitions manually, removing data disks or maybe also in the face of data corruption. Generally, check if one (and only one!) partition has `LABEL="hassos-data"` using `blkid` from the OS shell.

bobinoz1, 2023-04-26:
Get the same problem "Waiting for home assistant CLI to be ready..." after shifting data to a different drive via Settings/System/Storage
OS: haos_generic-x86-64-11.0.dev20230416.img.xz from location https://os-builds.home-assistant.io/11.0.dev20230416/ (installed to test/confirm Broadcom tigon firmware added in this version).
Version fully functional with Broadcom firmware and no problems over several weeks installing multiple add-ons and integrations  including performing associated restarts/reboots and adding devices.
The "Waiting for home assistant CLI to be ready..." problem occured after the data disk was moved to another internal drive in the host PC. HA continued to run without problems after the data drive shift for just over a day until the first hard boot after a planned power down.

agners, 2023-04-26:
What you can try is checking what exactly is running in a separate virtual console. To do this, on the system console (using a physical keyboard and display)  press Ctrl+Alt+F2. Enter `root` to get access to a shell. In the shell, you can check which parts of the Home Assistant stack are running using `docker ps`. You may also want to check on Supervisor specifically using `docker logs hassio_supervisor` and `systemctl status hassos-supervisor`.

bobinoz1, 2023-04-27:
Hi Stefan,

 

Can log-in as root to shell ok but docker ps commands hang.

 

# systemctl status hassos-supervisor command returns:

               Hassos-supervisor.service – HassOS supervisor

               Loaded: loaded (/usr/lib/system/system/hassos-supervisor.service; enabled;

              Apr 27 00:22:15 homeassistant system[1]: Dependency failed for HassOS supervisor 

 

 

HA Data drive details show (active) as:

hassos-data-old          /dev/sda8

hassos-data-dis          /dev/sdb1

where /dev/sda is the SSD drive with HA OS and /dev/sdb is the HDD drive in the same server

 

 

Stefan Agner ***@***.***> 
Sent: Thursday, 27 April 2023 12:59 AM
To: home-assistant/operating-system ***@***.***>
Cc: bobinoz1 ***@***.***>; Comment ***@***.***>
Subject: Re: [home-assistant/operating-system] Stuck waiting for home assistant cli to be ready (Issue #2272)

 

What you can try is checking what exactly is running in a separate virtual console. To do this, on the system console (using a physical keyboard and display) press Ctrl+Alt+F2. Enter root to get access to a shell. In the shell, you can check which parts of the Home Assistant stack are running using docker ps. You may also want to check on Supervisor specifically using docker logs hassio_supervisor and systemctl status hassos-supervisor.

dodsonw, 2023-05-07:
I encountered the exact same issue when updating from HA 2023.5.0 -> 2023.5.1 and found that my `hassos-data` partition was labeled as `LABEL="hassos-data-old"`.  Removing the "-old" from the label and rebooting fixed the issue.

Maybe the root cause lies somewhere in the data disk move script, as @agners stated above.

rfortunatov, 2023-05-24:
Had the exact same issue, my problem was that I had an ubuntu installed on ssd I'm using and leftover zfs metadata was left on an ssd. Formatting it clean to unallocated with gparted solved it for me.

himslm01, 2023-05-31:
After a Homeassistant update yesterday I'm stuck at the "waiting for home assistant cli to be ready" prompt.
Docker ps just hangs too - which I guess is why the /usr/sbin/hassos-cli script is hanging.
In fact - all docker commands which talk to the docker back-end hang.

I've tried booting to the slot A and slot B on the boot menu but get the same result.
Any advice?

himslm01, 2023-06-08:
[willsargent](https://github.com/willsargent) I don't have the command "supervisor"

himslm01, 2023-06-11:
In the end I relabeled `/dev/sda8` from `hassos-data-backup` to `hassos-data`, rebooted, and Home-Assistant came back.

agners, 2023-06-12:
We use partition label to find the data partition. I wonder what labeled the partition to `hassos-data-backup` in first place?

None of the official data disk features relabel to `hassos-data-backup`, so it must have been something else? :thinking:

mpneuried, 2023-06-21:
**I had the same problem.**
My system  got stuck with the message **`Waiting for the Home Assistant CLI to be ready...`**.
During boot it run into a 1:30 min. timeout when accessing the `/dev/disk/by-label/hassos-data`.

**My solution** after 2-3 hours research was relabeling the `/dev/disk/by-label/hassos-data-old` into `/dev/disk/by-label/hassos-data`.

**But How?!**

1. Within the open blocked console with the text "Waiting for the Home Assistant CLI to be ready..." hit Ctrl+Alt+F2 to open a second terminal.
2. here you have to login to HassOS in my case the user was "root"
3. Now you should have a terminal to work with.
4. Here you should check the label with `e2label /dev/sda8`. Im my case this returned `hassos-data-old`.
    _Probably in your case the device is different._  
    _If it returns the correct `hassos-data` then this solution is probably caused by another reason._
6. To rename the label execute `e2label /dev/sda8 hassos-data`
7. Finally reboot HassOS by executing `reboot` in the terminal.

> **My Environment:** _HassOS in an LXC container on Proxmox running in a Celeron Intel NUC_

luka6000, 2023-07-01:
> **I had the same problem.** My system got stuck with the message **`Waiting for the Home Assistant CLI to be ready...`**. During boot it run into a 1:30 min. timeout when accessing the [...]

this solution worked for me on rPi. My label was hassos-data-dis

enoola, 2023-07-21:
Hello, I have a Raspberry Pi2, SDCard 128Gb, I tried with the following images :

- hassos_rpi2-5.7.img.xz
- hassos_rpi2-7.2.img.xz
- haos_rpi2-11.0.dev20230720.img.xz. 

Consistantly failing with the same error "Waiting for home assistant CLI to be ready...", 

on haos_rpi2-11.0.dev20230720.img.xz, I found /dev/disk/by-label/hassos-data (no missnamed folder)
I am not yet used to this platform so I don't have much more to share.

dodsonw, 2023-07-22:
> I encountered the exact same issue when updating from HA 2023.5.0 -> 2023.5.1 and found that my `hassos-data` partition was labeled as `LABEL="hassos-data-old"`. Removing the "-old" from the label [...]

I believe I have found the root cause for my issue, if it helps anyone:

_-- For clarity, I am running HAOS in a VM on a Debian server._

After waiting for the issue to arise again, I checked HA's supervisor logs and found nothing.  Next I tried gaining access to the host for HAOS via SSH [(ssh add-on),](https://community.home-assistant.io/t/add-on-hassos-ssh-port-22222-configurator/264109).  Once there, I was able to check the host logs and could see that I had memory issues.  After some more digging I found that I had some ffmpeg_noise sensors that were slowly eating up my ram over a matter of weeks.  Turning off those sensors has temporarily solved the issue (although I still need to figure out why those memory leaks are occurring).

So... my guess is that when I went to do an update, the supervisor tried to make a backup before updating but hung because it was waiting for some available RAM - which would never be available.  After restarting the host, the RAM constraints subsided, but HA wasn't looking for the "backup" partition and would just "wait for the CLI to be ready".

Two possible improvement I see are:
1) Check for both partitions on startup and load "hassos-data-old" if "hassos-data" is missing.
2) Make the supervisor aware of these critical issues on the host - For logging/notification purposes and for checks during app updates.

I'll need more time to see how feasible those code changes are.

wagnerfontes2, 2023-07-29:
> **But How?!**
>
> 1. Within the open blocked console with the text "Waiting for the Home Assistant CLI to be ready..." hit Ctrl+Alt+F2 to open a second terminal.
> 2. here you have to login to HassOS in [...]

I'm running HA in a VM under a Windows host in a desktop computer and I have the same error message, but probably my case is a little different.
I did follow the suggested steps, but in my case the label returns 'hassos-data'. Any suggestions to move on from here?

nicolas-sora, 2023-10-14:
> **I had the same problem.** My system got stuck with the message **`Waiting for the Home Assistant CLI to be ready...`**. During boot it run into a 1:30 min. timeout when accessing the `/dev/disk/by-label/hassos-data`.
> 
> **My solution** after 2-3 hours research was relabeling the `/dev/disk/by-label/hassos-data-old` into `/dev/disk/by-label/hassos-data`.
> 
> **But How?!**
> 
> 1. Within the open blocked console with the text "Waiting for the Home Assistant CLI to be ready..." hit Ctrl+Alt+F2 to open a second terminal.
> 2. here you have to login to HassOS in my case the user was "root"
> 3. Now you should have a terminal to work with.
> 4. Here you should check the label with `e2label /dev/sda8`. Im my case this returned `hassos-data-old`.
>    _Probably in your case the device is different._
>    _If it returns the correct `hassos-data` then this solution is probably caused by another reason._
> 5. To rename the label execute `e2label /dev/sda8 hassos-data`
> 6. Finally reboot HassOS by executing `reboot` in the terminal.
> 
> I hope this step by step guide helps the not so experience users.
> 
> > **My Environment:** _HassOS in an LXC container on Proxmox running in a Celeron Intel NUC_

This helped 100%. Thank you also for the step by step. My setup: Rpi4 and this happened after an update

rvdbergh, 2023-10-27:
> **I had the same problem.** My system got stuck with the message **`Waiting for the Home Assistant CLI to be ready...`**. During boot it run into a 1:30 min. timeout when accessing the [...]

Thank you for this!!  This fixed my issue on HassOS running in a Virtualbox VM. Today I updated my HassOS to 11.0 and before that I also got notified that one of the /dev/sdaXX devices would need to be renamed because it contained no data or something peculiar like that (can't remember the exact wording). Afterwards the VM would get stuck on the prompt "Waiting for home assistant CLI to be ready..." everytime.

r2korz, 2023-11-01:
Gentlemen, thanks for the tips above and I got mine to start working by:
pressing Alt-Ctrl-F2  then username: root    
then I entered the command   ha supervisor options
On my windows machine browser I enter the address   http://homeassistant.local:8123/
That brought up the page and 
"Preparing Home Assistant
This may take up to 20 minutes"

Good luck to you.

ichbinsteffen, 2023-11-23:
@mbroadst Have you been able to resolve this issue? 

I believe I have the same issue. Which is not fixable by the re-labeling because hassos-data has never been set up.
I think that just helps for people that faced this issue after upgrading to a new version. 

I filmed my boot and found that I have the same errors you showed in this screenshot 
https://github.com/home-assistant/operating-system/issues/2272#issuecomment-1399543806

By the comment history it looks like your issue has been hijacked by users that have a different issue. Because you said:
> _The system time messages appear to be a red herring, the main issue seems to be that the hassos-data is never set up._

So were you able to resolve the issue?

mbroadst, 2023-12-17:
I finally got time to come back to this project, and seem to have got it working now. What worked for me was to do a full (not fast) wipe of the drive, and I suspect what really worked was that all labels were removed from all devices during this process (as @miles75 said).  Hope this helps somebody, I'm going to close the issue for now.

Synergytechteam, 2023-12-30:
Finally, to solve the problem of the screen being stuck on 'waiting for the Home Assistant CLI to be ready', I had to make sure all partitions on the drive were removed. This was necessary because Balena Etcher doesn't completely wipe the disk. Then, I connected the target disk to an external reader and used Balena Etcher to write the latest Home Assistant OS onto the disk. This solution worked like a charm

liwenyip, 2024-02-07:
I believe I had the same issue as @mbroadst. Full (not fast) format of the drive worked for me. (wipefs did NOT work, and there was no hassos-data-old partition to rename)

What I did:

clean install of [haos_generic-x86-64-11.5.img.xz](https://github.com/home-assistant/operating-system/releases/download/11.5/haos_generic-x86-64-11.5.img.xz) on an Intel NUC DN2820FYKH (3rd gen) with a 120GB SSD, using the ubuntu Disks utility as described [here](https://www.home-assistant.io/installation/generic-x86-64#method-1-installing-haos-via-ubuntu-booting-from-a-usb-flash-drive)

Initially it was getting stuck on 
`[***   ] (1 of 2) A start job is running for KINGSTON_SA400S37120G hassos-data (__s / no limit)`
and
`[***   ] (2 of 2) A start job is running for HassOS data resizing (__s / no limit)`
for about 2 minutes, then proceeded to
`Waiting for the Home Assistant CLI to be ready...`
and got stuck there forever.

Based on @miles75's post I tried `wipefs -a /dev/sda` (where sda is the target drive) and then tried again, this time it got stuck on
`[***   ] A start job is running for /dev/disk/by-label/hassos-data (__s / no limit)`
for about 2 minutes, then proceeded to
`Waiting for the Home Assistant CLI to be ready...`
and got stuck there forever.

I videoed the boot sequence this time (https://www.youtube.com/watch?v=YPv7ZeiRO3I) and noticed I got the same "Dependency failed" errors as @hermanops.

Did a full (not fast) format of the target drive, then it worked.

I wonder why the full format worked, but wipefs -a did not.

athuler, 2024-03-19:
> Virtual

After getting a `system unhealthy` error within home assistant (which led to [this error page](https://www.home-assistant.io/more-info/unhealthy/setup)). After rebooting the system, it was stuck on `Waiting for Home Assistant CLI to be Ready`. After following instructions from @rvdbergh, it was able to successfully start.

Richyhx1, 2024-03-20:
Is there a way to rename the hassos-data file without the terminal? I can enter the terminal but any commands are ignored and not responded to (maybe im doing it wrong i am a novice at this, but ive not had problems with terminals before). 

Im sure that this is the issue i am having as HA said it was unhealthy, complained it had more than 1 file of the same name after an update, i hit OK and it said file has been renamed and after a restart i was receiving the waiting CLI error and can see during startup its missing the hassos-data file.

cody82, 2024-03-30:
I also had this problem on a fresh install. Previously there was TrueNAS installed on this computer so I think the problem was some remains of TrueNAS, maybe ZFS. Completely wiping all disks before installing Home Assistant OS fixed the problem.

LionRelaxe, 2024-04-25:
**Same issue:**
**Fresh install** on a Lenovo M90n-IoT with an i3, 4gb ram and a 128gb low-performance NVMe SSD.
Computer had windows before on the drive.
Booted Ubuntu Live, and used Disk to restore and image (HAOS 12.2) as per HA tutorial.
After 1m30s, timeout to resize the data partition:
`[DEPEND] Dependency failed for HassOS data resizing.`

Command line (From CTRL-ALT-F2) with blkid showed part p8 to be correctly labeled (hassos-data)
Ping to 8.8.8.8 and local network worked.

**What worked:**
Booted Ubuntu Live, used Disk to FORMAT DISK, with option "empty no partitions" and "slow write zeroes everywhere". Then used Disk to restore and image (HAOS 12.2) as per HA tutorial.

On bootup, resizing the SWAP took 30s, resizing data was so fast I did not see going trough (so, less than 2s).
...
and now it works.

I'm enclined to say it's a bug, as strictly following instructions give a dead-end.
Maybe the data resizing needs more time on a non-zeroed disk to perform the duty?
Why put a timeout anyway, if the whole thing fails if it fails??
If zeroing the disk is critical, probably something important is incorrectly defined in the disk image? Like not zeroing a trailing sector after a partition table that can add garbage to it?

Anyway, that is my first welcome to HA, and it's already a bug hunt.
Hope this helps other users and devs! :)

agners, 2024-05-13:
There are definitely multiple issues collected in here: The original poster posted the issue with HAOS 9.4, which did not had the swap initialization logic (as it was using zram back then, not zswap). For some users it seems that the data partition got renamed to `hassos-data-old`, this seems related to data disk feature.

In any case, please let's not continue growing this issue, it has become way too long at this point :see_no_evil: 

I do agree with @LionRelaxe investigation, depending on the system the resize might take longer. We probably should bump that timeout a bit. I created a new issue for this particular comment with https://github.com/home-assistant/operating-system/issues/3365.