home-assistant/operating-system #2273: Yellow Disable Ethernet LEDs

maforshaw, 2022-12-18:
### Describe the issue you are experiencing

I've attempted to disable the ethernet LEDs on the HA Yellow using the procedure for the Pi4, i.e. edit `/mnt/boot/config.txt` and add the following:

```
dtparam=eth_led0=4
dtparam=eth_led1=4
```
After reboot, the green ethernet led goes off but the orange one keeps flashing. It appears that led0 is orange and led1 is green, reversed from a Pi4 arrangement. Looking through the core source code I can see a patch for HA yellow that has the following override:

```
eth_led0 = <&phy1>,"led-modes:0";
eth_led1 = <&phy1>,"led-modes:4";
```

Not sure if that is related or nothing to do with this issue. I also tried a value of 14 (the Pi3 setting for off) and that didn't work either. Why can one of the LEDs be turned off but not the other.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Disable ethernet LEDs via /mnt/boot/config.txt
2. Reboot
3. Green ethernet LED goes off. Orange ethernet LED flashes.

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

agners, 2022-12-20:
> Looking through the core source code I can see a patch for HA yellow that has the following override:

The override is taken straight from the CM4 device tree provided by the Raspberry Pi Linux repository.

The problem is reproducible with Raspberry Pi OS, on a CM4 with the CM4 IO board. I opened an issue in the Raspberry Pi Linux repository: https://github.com/raspberrypi/linux/issues/5289

agners, 2023-01-24:
This is resolved with HAOS 9.5 with the latest kernel update (#2312).

closed 2023-01-24 by agners