home-assistant/operating-system #2276: [9.4] Web-Application Not Binding to :8123 on New Installs

codeshaw, 2022-12-19:
### Describe the issue you are experiencing

Hi all,

Odd issue here - I recently installed 9.2 on my TrueNAS installation in VMWare environment, then didn't bother configuring it as I was too lazy to migrate from my Pi until tonight. When trying to log in to the welcome console I was unable to connect, yet the observer was reporting availability.

Upon OS investigation it appears that there were no exposed ports from the `homeassistant` container, which is managed by the supervisor, I believe. After investigating the content of `/mnt/config` there was no default configuration present, which I wouldn't expect on an instance that had not been configured.

I reinstalled 9.4 and obtained the same results. I have now gone back to 9.2 and disabled internet access during initial startup to prevent unwanted updates until after my config is loaded,

While I'm not familiar with the architecture of the managed OS / supervisor setup it appears that the initial web-end configuration is not being loaded from a default, and the absence of a pre-existing configuration means there is no port to boot to.

Howver, I also note two recent networking changes:
970bec6ac3a63d08c6a31fec4835fb176ad83331 
ce16ee5d494a2a0140d94869bb62e5ecf1007867

The first listed is interesting to me, as it moves the supervisor's network range to the same as the web front end. My knowledge of docker networking is fairly limited, but it it possible that this, coupled with my shared-networking-card-using-dhcp setup was making life... strange.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install hassos_generic-x86-64_9.4 on Virtualised Hardware
2. Wait for CLI to be ready (Via VNC or similar)
3. Observe homeassistant.local:8123 not connecting
4. In CLI, drop to login shell and observe `docker inspect homeassistant` - no port is exposed

### Anything in the Supervisor logs that might be useful for us?

```txt
Couldn't start the front-end
```

### Anything in the Host logs that might be useful for us?

```txt
Couldn't start the front-end
```

codeshaw, 2022-12-20:
As an update, I have just upgraded the supervisor from 9.2-> 9.4 and the problem is back - no exposed ports for the web application.

[image]

codeshaw, 2022-12-20:
OK - I have _mostly_ resolved this, although I am at a loss as to what caused it (and this issue is in the completely wrong place).

After a dive into the core logs the front-end container was starting, but the internal process was stuck in a restart loop as the weather integration was not able to hit its endpoint. Based on an old post I set the DNS for the hass instance to 8.8.8.8 and that seems to have resolved it. I'm not sure why, as my router is also set to use Google DNS...

agners, 2022-12-20:
Home Assistant Core does run in its own container, however, it has full host network access. So whatever the Core is binding to, will be accessible from the outside.

> OK - I have mostly resolved this, although I am at a loss as to what caused it (and this issue is in the completely wrong place).

Yes, it probably belongs to the Core repository.

> After a dive into the core logs the front-end container was starting, but the internal process was stuck in a restart loop as the weather integration was not able to hit its endpoint. Based on an old post I set the DNS for the hass instance to 8.8.8.8 and that seems to have resolved it. I'm not sure why, as my router is also set to use Google DNS...After a dive into the core logs the front-end container was starting, but the internal process was stuck in a restart loop as the weather integration was not able to hit its endpoint. Based on an old post I set the DNS for the hass instance to 8.8.8.8 and that seems to have resolved it. I'm not sure why, as my router is also set to use Google DNS...

We use CoreDNS in our Supervisor DNS plug-in which provides DNS services to all add-ons and the Core itself. CoreDNS uses the DHCP provided DNS server. As fallback 1.1.1.1 is used (Cloudflare). But this should not become active as long as the DHCP provided DNS is working. You can also check `ha network info` to see what DNS server got picked up.

I'll close this issue. Feel free to create a new issue here in case something OS related pops up.

closed 2022-12-20 by agners