home-assistant/operating-system #2280: Recommended Wifi Dongle not recognized rtl8192cu/eu

lundsola, 2022-12-20:
### Describe the issue you are experiencing

My USB Wifi dongle with the chipset rtl8192eu is not recognized by HASSIO, 
Dmesg output: 
[image]

 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### System information

-

### Additional information

-

agners, 2022-12-20:
The driver should be enabled. It seems that your system is experience some type of USB issue. Can you try a different port and/or a Hub between your device and the WiFi dongle?

What type of device are you using?

Can you share the full host logs?

agners, 2022-12-20:
Just checked on my Generic x86-64 installation, the WiFi dongle is recognized properly:

```
[936517.154827] usb 3-1: new high-speed USB device number 5 using xhci_hcd
[936517.284928] usb 3-1: New USB device found, idVendor=0bda, idProduct=8176, bcdDevice= 2.00
[936517.284953] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[936517.284962] usb 3-1: Product: 802.11n WLAN Adapter
[936517.284970] usb 3-1: Manufacturer: Realtek
[936517.284977] usb 3-1: SerialNumber: 00e04c000001
[936517.371416] rtl8192cu: Chip version 0x10
[936518.216993] rtl8192cu: Board Type 0
[936518.218046] rtl_usb: rx_max_size 15360, rx_urb_num 8, in_ep 1
[936518.218143] rtl8192cu: Loading firmware rtlwifi/rtl8192cufw_TMSC.bin
[936518.218399] ieee80211 phy0: Selected rate control algorithm 'rtl_rc'
[936518.248885] usbcore: registered new interface driver rtl8192cu
[936518.267252] rtl8192cu 3-1:1.0 wlp0s16u1: renamed from wlan0
[936518.293309] rtl8192cu: MAC auto ON okay!
[936518.504526] rtl8192cu: Tx queue select: 0x05
```

lundsola, 2022-12-20:
The issue is probably that I have the eu-version then. The passthrough though proxmox is working fine for all my other dongels (BT, Zigbee, Z-wave etc)

This is the one I'm using: https://www.amazon.se/gp/product/B0088TKTY2/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1

Hmm, I don't get anything out of the host logs unfortunately :/

agners, 2022-12-20:
Hm I see, [this device](https://linux-hardware.org/?id=usb:2357-0109&page=1) seems to require the `CONFIG_RTL8XXXU` driver. Besides that, the device needs the `rtlwifi/rtl8192eu_nic.bin` firmware, but that one is already enabled with the Buildroot `BR2_PACKAGE_LINUX_FIRMWARE_RTL_81XX` package.

I'll add the required driver.

lundsola, 2022-12-20:
Super! Thanks a lot!

lundsola, 2023-02-14:
Everything works! Can be closed!

closed 2023-02-14 by lundsola