home-assistant/operating-system #2283: Intel AX210NGW Wifi Firmware issue

robotastic-io, 2022-12-21:
### Describe the issue you are experiencing

Hi,

I tried installing a Intel AX210NGW Wifi Card it my HA machine. After bootup bluetooth was detected fine but no Wifi. After checing dmesg I see the following log messages for iwlwifi.

> [    2.027870] iwlwifi 0000:02:00.0: api flags index 2 larger than supported by driver
[    2.027929] iwlwifi 0000:02:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.63.2.2
[    2.029140] iwlwifi 0000:02:00.0: loaded firmware version 66.f1c864e0.0 ty-a0-gf-a0-66.ucode op_mode iwlmvm
[    2.154663] iwlwifi 0000:02:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
[    2.306033] iwlwifi 0000:02:00.0: Detected RF GF, rfid=0x10d000
[    2.307027] iwlwifi 0000:02:00.0: Microcode SW error detected. Restarting 0x0.
[    2.307155] iwlwifi 0000:02:00.0: Start IWL Error Log Dump:
[    2.307162] iwlwifi 0000:02:00.0: Transport status: 0x0000004A, valid: 6
[    2.307169] iwlwifi 0000:02:00.0: Loaded firmware version: 66.f1c864e0.0 ty-a0-gf-a0-66.ucode
[    2.307178] iwlwifi 0000:02:00.0: 0x00000071 | NMI_INTERRUPT_UMAC_FATAL
[    2.307199] iwlwifi 0000:02:00.0: 0x002002F0 | trm_hw_status0
[    2.307206] iwlwifi 0000:02:00.0: 0x00000000 | trm_hw_status1
[    2.307213] iwlwifi 0000:02:00.0: 0x004DA722 | branchlink2
[    2.307219] iwlwifi 0000:02:00.0: 0x004D0CCE | interruptlink1
[    2.307225] iwlwifi 0000:02:00.0: 0x004D0CCE | interruptlink2
[    2.307231] iwlwifi 0000:02:00.0: 0x004D94DA | data1
[    2.307238] iwlwifi 0000:02:00.0: 0x00000010 | data2
[    2.307244] iwlwifi 0000:02:00.0: 0x00000000 | data3
[    2.307249] iwlwifi 0000:02:00.0: 0x00000000 | beacon time
[    2.307255] iwlwifi 0000:02:00.0: 0x00011AD6 | tsf low
[    2.307261] iwlwifi 0000:02:00.0: 0x00000000 | tsf hi
[    2.307267] iwlwifi 0000:02:00.0: 0x00000000 | time gp1
[    2.307273] iwlwifi 0000:02:00.0: 0x00023244 | time gp2
[    2.307279] iwlwifi 0000:02:00.0: 0x00000001 | uCode revision type
[    2.307285] iwlwifi 0000:02:00.0: 0x00000042 | uCode version major
[    2.307291] iwlwifi 0000:02:00.0: 0xF1C864E0 | uCode version minor
[    2.307298] iwlwifi 0000:02:00.0: 0x00000420 | hw version
[    2.307304] iwlwifi 0000:02:00.0: 0x18C89002 | board version
[    2.307311] iwlwifi 0000:02:00.0: 0x8008FF05 | hcmd
[    2.307317] iwlwifi 0000:02:00.0: 0x00020000 | isr0
[    2.307323] iwlwifi 0000:02:00.0: 0x60000000 | isr1
[    2.307330] iwlwifi 0000:02:00.0: 0x48F00002 | isr2
[    2.307335] iwlwifi 0000:02:00.0: 0x00C0000C | isr3
[    2.307342] iwlwifi 0000:02:00.0: 0x00000000 | isr4
[    2.307347] iwlwifi 0000:02:00.0: 0x00000000 | last cmd Id
[    2.307354] iwlwifi 0000:02:00.0: 0x004D94DA | wait_event
[    2.307360] iwlwifi 0000:02:00.0: 0x00000000 | l2p_control
[    2.307366] iwlwifi 0000:02:00.0: 0x00000000 | l2p_duration
[    2.307373] iwlwifi 0000:02:00.0: 0x00000000 | l2p_mhvalid
[    2.307379] iwlwifi 0000:02:00.0: 0x00000000 | l2p_addr_match
[    2.307385] iwlwifi 0000:02:00.0: 0x00000009 | lmpm_pmg_sel
[    2.307393] iwlwifi 0000:02:00.0: 0x00000000 | timestamp
[    2.307399] iwlwifi 0000:02:00.0: 0x00000024 | flow_handler
[    2.307452] iwlwifi 0000:02:00.0: Start IWL Error Log Dump:
[    2.307458] iwlwifi 0000:02:00.0: Transport status: 0x0000004A, valid: 7
[    2.307465] iwlwifi 0000:02:00.0: 0x2010070D | ADVANCED_SYSASSERT
[    2.307473] iwlwifi 0000:02:00.0: 0x00000000 | umac branchlink1
[    2.307479] iwlwifi 0000:02:00.0: 0x8045DFC6 | umac branchlink2
[    2.307485] iwlwifi 0000:02:00.0: 0x010910FE | umac interruptlink1
[    2.307491] iwlwifi 0000:02:00.0: 0x00000000 | umac interruptlink2
[    2.307497] iwlwifi 0000:02:00.0: 0x00000005 | umac data1
[    2.307502] iwlwifi 0000:02:00.0: 0xDEADBEEF | umac data2
[    2.307508] iwlwifi 0000:02:00.0: 0xDEADBEEF | umac data3
[    2.307514] iwlwifi 0000:02:00.0: 0x00000042 | umac major
[    2.307520] iwlwifi 0000:02:00.0: 0xF1C864E0 | umac minor
[    2.307527] iwlwifi 0000:02:00.0: 0x0002323C | frame pointer
[    2.307534] iwlwifi 0000:02:00.0: 0xC0885E88 | stack pointer
[    2.307540] iwlwifi 0000:02:00.0: 0x00010C00 | last host cmd
[    2.307546] iwlwifi 0000:02:00.0: 0x00000000 | isr status reg
[    2.307565] iwlwifi 0000:02:00.0: IML/ROM dump:
[    2.307571] iwlwifi 0000:02:00.0: 0x00000B03 | IML/ROM error/state
[    2.307590] iwlwifi 0000:02:00.0: 0x00007279 | IML/ROM data1
[    2.307608] iwlwifi 0000:02:00.0: 0x00000080 | IML/ROM WFPM_AUTH_KEY_0
[    2.307622] iwlwifi 0000:02:00.0: Fseq Registers:
[... 277 lines cut ...]
dashboards | 2
-- | --
resources | 3
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 10. November 2022 um 22:53
-- | --
current_recorder_run | 21. Dezember 2022 um 14:30
estimated_db_size | 494.54 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

agners, 2022-12-21:
Hm, this looks like a Linux kernel/firmware bug.

This bug report seems to match your firmware version and the issue, but the issue has been resolved and the patch is part of Home Assistant OS already:
https://bugzilla.kernel.org/show_bug.cgi?id=215523

Maybe it is related to this issue. It might be resolved in newer kernel versions:
https://bugzilla.kernel.org/show_bug.cgi?id=208425

I plan to update to 6.1 which presumably will be the next LTS kernel in January. Probably worth giving development builds  try once that is done.

robotastic-io, 2022-12-21:
Okay, I'll wait for the january update and report back then. Got a spare stick flying around I can use in the meantime.

dirk77, 2023-01-03:
I have exaclty the same issue with exactly the same AX210 card. Hopefully something will be done to the kernel in the near future. 
Thanks to your explanation I was able to find out that it was obviously not me that the wifi was not working ;-)

el1-x, 2023-03-03:
I have just recently picked up one of these wireless cards for my mini PC that's running Proxmox and HAOS amongst a host of other things.

I will admit I didn't expect there to be issues with this card, but have since learned otherwise after installing and running into issues with HAOS and the Wi-Fi PCI and Bluetooth USB devices being passed through.

I am eager to see the 6.1 kernel being used for HAOS.

agners, 2023-03-07:
@el1-x you can try latest dev builds from https://os-builds.home-assistant.io/ already which come with Linux 6.1:

el1-x, 2023-04-28:
I am currently at a loss for what direction to take to get the AX210 WiFi and Bluetooth working in HA.
I am running the latest version of HAOS 10.1 using the 6.1.x kernel. HA is running as a VM under Proxmox. WiFI (PCI) and BT (USB) devices have been passed through from the host to the VM. 

Apologies for the screenshots as opposed to plain text, but without being able to SSH into HAOS this is the best I can do at the moment using the Proxmox console. This is all the information I can see via dmesg regarding both the WiFi and BT devices.

WiFi - No 'iwlwifi'' marked entries, this is all that I can find referencing the PCI device.
[image]

BT - FW download error recovery failed?
[image]

agners, 2023-04-28:
Hm, -19 means "No such device", I wonder if the device somehow disappeared again.

el1-x, 2023-04-28:
A little more digging with dmesg | grep 00:01 reveals the below.
```
[    0.437994] pci 0000:00:1a.0: can't claim BAR 4 [io  0xd300-0xd31f]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.438499] pci 0000:00:1a.1: can't claim BAR 4 [io  0xd2e0-0xd2ff]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.439229] pci 0000:00:1a.2: can't claim BAR 4 [io  0xd2c0-0xd2df]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.439751] pci 0000:00:1d.0: can't claim BAR 4 [io  0xd2a0-0xd2bf]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.440486] pci 0000:00:1d.1: can't claim BAR 4 [io  0xd280-0xd29f]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.440499] pci 0000:00:1d.2: can't claim BAR 4 [io  0xd260-0xd27f]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.441302] pci 0000:00:1f.2: can't claim BAR 4 [io  0xd240-0xd25f]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.441499] pci 0000:00:1f.3: can't claim BAR 4 [io  0xd200-0xd23f]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.442618] pci 0000:00:01.0: vgaarb: setting as boot VGA device
[    0.442890] pci 0000:00:01.0: vgaarb: bridge control possible
[    0.443259] pci 0000:00:01.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[    0.511892] pci_bus 0000:01: resource 0 [io  0xd000-0xdfff]
[    0.512281] pci_bus 0000:01: resource 1 [mem 0xc2000000-0xc21fffff]
[    0.512689] pci_bus 0000:01: resource 2 [mem 0x700000000000-0x700fffffffff 64bit pref]
```

el1-x, 2023-04-29:
I don't believe this to be the case. I quickly fired up an Alpine Linux VM, passed though the PCI device and everything operates as expected running kernel version 5.15.109-0-lts

```
01:00.0 Network controller: Intel Corporation Wi-Fi 6 AX210/AX211/AX411 160MHz (rev 1a)
        Subsystem: Intel Corporation Wi-Fi 6 AX210 160MHz
        Physical Slot: 0
        Flags: bus master, fast devsel, latency 0, IRQ 16
        Memory at fe800000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: [c8] Power Management version 3
        Capabilities: [d0] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [40] Express Endpoint, MSI 00
        Capabilities: [80] MSI-X: Enable+ Count=16 Masked-
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [14c] Latency Tolerance Reporting
        Kernel driver in use: iwlwifi

[    2.227672] iwlwifi 0000:01:00.0: api flags index 2 larger than supported by driver
[    2.227707] iwlwifi 0000:01:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.63.2.2
[    2.227964] iwlwifi 0000:01:00.0: loaded firmware version 66.f1c864e0.0 ty-a0-gf-a0-66.ucode op_mode iwlmvm
[    2.250611] iwlwifi 0000:01:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
[    2.426518] iwlwifi 0000:01:00.0: loaded PNVM version 0xdbd9582f
[    2.441278] iwlwifi 0000:01:00.0: Detected RF GF, rfid=0x10d000

                                    Devices                                    
--------------------------------------------------------------------------------
  Name                  Address               Powered     Adapter     Mode      
--------------------------------------------------------------------------------
  wlan0                 xx:xx:xx:xx:xx:xx     on          phy0        station
  
  --------------------------------------------------------------------------------
  Settable  Property              Value                                          
--------------------------------------------------------------------------------
            Scanning              no                                               
            State                 connected                                        
            Connected network     BNTO                                             
            IPv6 address        
            IPv6 address               
            ConnectedBss          00:11:32:b1:81:1f                                
            Frequency             5785                                             
            Security              WPA2-Personal                                    
            RSSI                  -20 dBm                                          
            AverageRSSI           -23 dBm                                          
            TxMode                802.11ac                                         
            TxMCS                 3                                                
            TxBitrate             234000 Kbit/s                                    
            RxBitrate             6000 Kbit/s 
```

el1-x, 2023-04-29:
> Hm, -19 means "No such device", I wonder if the device somehow disappeared again.

Thank you for this little bit of information. It pointed me in the right direction and I was able to get the Bluetooth portion of the AX210 working. 

For anyone else that may stumble across this issue, The host was 'taking control' of the Bluetooth device causing the (-19) error. The fix was to create a blacklist file within `/etc/modprobe.d/` on the Proxmox host. 

While the filename doesn't really matter, I named mine disable-bt.conf so that future me knows/remembers what I did and why. The file itself simply needs to contain the following line:
`blacklist btusb`
This will blacklist the bluetooth-usb driver from being loaded by the host. From there I was successfully able to pass-through the BT device and HA immediately detected the new device once a reboot of the VM was complete.

el1-x, 2023-05-31:
@agners 
The update to HA OS 10.2 has resolved my issue. Firmware is loaded and the device is now accessible via nmcli and within the Home Assistant UI. Hopefully everyone else has the same outcome.

Cheers.

agners, 2023-05-31:
Thanks for the update. Probably the Linux kernel or firmware update fixed the issue then.

closed 2023-05-31 by agners