home-assistant/operating-system #2285: Patching the BlueZ included with HA OS

bdr99, 2022-12-22:
### Describe the issue you are experiencing

BlueZ 5.63, the version included in HA OS, has a bug that causes the Bluetooth GATT services for certain devices to not be visible. I encountered the issue with an Eufy T9148 smart scale, and it might be relevant to other devices as well, like in https://github.com/Jc2k/aiohomekit/issues/135, https://github.com/home-assistant/core/issues/76380, and https://github.com/bluez/bluez/issues/30.

I reported this issue to BlueZ, here is the GitHub issue:
https://github.com/bluez/bluez/issues/438

After I reported the issue, BlueZ fixed it in their latest master branch. These are the two commits which fixed the issue:
https://github.com/bluez/bluez/commit/b91d9213d951277896845830d0680061c7643828
https://github.com/bluez/bluez/commit/da203f5dbc7edaa15b55d1efcea3fda5e43cfc2e

I backported this change to 5.63, and confirmed that it still fixes the issue in that version. Minor changes were needed in order to make the patch work with 5.63 due to BlueZ having recently changed their approach to debug logging. Here is the patch for the change needed to fix the issue in BlueZ 5.63:
[0001-patch.patch](https://github.com/home-assistant/operating-system/files/10283357/0001-patch.patch)

I'd like to create an integration for my smart scale, but currently I can't do that, since HA OS uses BlueZ 5.63 which has the bug. Would it be possible to apply the above patch to the BlueZ package included in HA OS? This could potentially improve compatibility with several Bluetooth devices. I'm willing to help with the implementation, but would need some guidance since I've never contributed to an OS before.

Any help would be greatly appreciated. Thanks!

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

N/A

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

agners, 2022-12-22:
Interesting issue and nice you've been able to track it down!

We use the upstream buildroot package to build BlueZ. Hence this requires changing our buildroot fork: https://github.com/home-assistant/buildroot/tree/2022.02.x-haos.

So in this case the path would be to create a PR for that repository to add the patches, and then bump it in a PR on this repository.

Adding two patches is easy enough, so I went ahead and backported them to BlueZ 5.63 (which needed a third patch to make it compile), see https://github.com/home-assistant/buildroot/commit/41bd49dba4c1301f5613b9410763c0f70f57e785.

The bump of the buildroot git submodule is then a PR to this repository, see #2286. With this, the problem should be addressed with the next major and minor release.

bdr99, 2022-12-22:
@agners Great, thanks for creating that PR! I was close to figuring out how to apply the patches on my own, but didn't realize patches were stored in a separate repo.

Dvalin21, 2022-12-27:
I believe I've been having a similar issue. What I noticed the most is that "bleak" will freak out. I think the only part that may not be similar is that HA OS after a day or two will no longer see my bluetooth adapter attached to the host. Even after restarting HAOS it still doesn't see it. I've already verified that the host system has the device detected and also in the VM (Synology VM) still shows the device attached. In order for it to have it come back I have to unplug the usb adapter from the host system, re attach to the VM and restart HAOS.

bdr99, 2022-12-27:
@Dvalin21 That sounds like it might be a different issue. This issue was related to missing services for a Bluetooth device, I never had an issue where HA OS wasn't seeing the Bluetooth adapter.

Chaosflo, 2022-12-28:
I leave a pin here for [Awox HACS](https://github.com/fsaris/home-assistant-awox/issues)
Which currently doesn't work for a few people
Maybe this will fix the issue of timeouting when connecting

I hope this will get released soon =/

agners, 2022-12-28:
If I build another 9.x release, it will be sometime in January.

You can also try it out by updating to dev builds, it is quite safe.

```
ha supervisor --channel dev
ha supervisor reload
ha os update --version 10.0.dev20221222
ha supervisor --channel stable
```

Chaosflo, 2023-01-01:
> I leave a pin here for [Awox HACS](https://github.com/fsaris/home-assistant-awox/issues) Which currently doesn't work for a few people Maybe this will fix the issue of timeouting when connecting
> 
> I hope this will get released soon =/

Okay, I got time to test this.
Sadly it doesn't change anything.
Still timeouts but idk, maybe its the length of the timeout or something else
Will research further

BUT!
Things got faster on my machine regarding bluetooth 👍

agners, 2023-01-25:
HAOS 9.5 contains the BlueZ bugfix.

closed 2023-01-25 by agners

bdr99, 2023-01-25:
Thanks again!

tramsay, 2024-06-05:
Bluez Maintainers
Here is a fix for attribute writes not working in src/shared/gatt-db.c
https://github.com/tramsay/bluez-tramsay/commit/246bc960629dff34e744c728f048e9f50f1a005d