home-assistant/operating-system #2287: make generic_x86_64 fails while patching systemd

voltagex, 2022-12-26:
### Describe the issue you are experiencing

When building from 93656904def4438b043b693304abde077e5a7ccc
```
>>> systemd 247.3 Extracting
gzip -d -c /tmp/dl/systemd/systemd-247.3.tar.gz | tar --strip-components=1 -C /home/voltagex/src/operating-system/output/build/systemd-247.3   -xf -
>>> systemd 247.3 Patching

Applying 0001-time-wait-sync-log-errors-trying-to-watch-run-system.patch using patch:
can't find file to patch at input line 16
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 2214c1f039143caaf4ed99eb0f3fc4551b322148 Mon Sep 17 00:00:00 2001
|Message-Id: <2214c1f039143caaf4ed99eb0f3fc4551b322148.1635377266.git.stefan@agner.ch>
|From: Aman Gupta Karmani <aman@tmm1.net>
|Date: Mon, 12 Oct 2020 13:39:26 -0700
|Subject: [PATCH 1/2] time-wait-sync: log errors trying to watch
| /run/systemd/timesync
|
|---
| src/timesync/wait-sync.c | 2 +-
| 1 file changed, 1 insertion(+), 1 deletion(-)
|
|diff --git a/src/timesync/wait-sync.c b/src/timesync/wait-sync.c
|index 2a9b113ff4..62fa13c509 100644
|--- a/src/timesync/wait-sync.c
|+++ b/src/timesync/wait-sync.c
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
make[1]: *** [package/pkg-generic.mk:209: /home/voltagex/src/operating-system/output/build/systemd-247.3/.stamp_patched] Error 1
make[1]: Leaving directory '/home/voltagex/src/operating-system/buildroot'
make: *** [Makefile:39: generic_x86_64] Error 2
```

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

dev

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Build HAOS from source

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

voltagex, 2022-12-26:
At a guess, https://github.com/home-assistant/operating-system/commit/93656904def4438b043b693304abde077e5a7ccc updated a submodule but did not update `.gitmodules`, my checkout was still on the older buildroot branch.

agners, 2022-12-26:
> >>> systemd 247.3 Extracting

According to `package/systemd/systemd.mk` is 250.8, so something seems to be outdated on your end.

agners, 2022-12-26:
> At a guess, [9365690](https://github.com/home-assistant/operating-system/commit/93656904def4438b043b693304abde077e5a7ccc) updated a submodule but did not update `.gitmodules`, my checkout was still on the older buildroot branch.

Yes, git submodules need manual updating. See: https://developers.home-assistant.io/docs/operating-system/getting-started#check-out-source-code

voltagex, 2022-12-26:
Yes, but https://github.com/home-assistant/operating-system/blob/dev/.gitmodules#L4 is pointing to an old branch.

agners, 2022-12-26:
Right, created a PR to fix this: https://github.com/home-assistant/operating-system/pull/2288.

A `git submodule update --init --force` should still checkout the right git hash though.

voltagex, 2022-12-26:
Many thanks - I'll know how to create a PR for it if I come across something like this in the future. Should I leave this issue open until that PR is merged?

agners, 2022-12-26:
If the PR looks good to you, I will merge it and close this issue.

closed 2022-12-27 by agners