home-assistant/operating-system #2290: Experimental Features lead to Bluetooth errors

Chaosflo, 2022-12-28:
### Describe the issue you are experiencing

I want to use the [Awox HACS](https://github.com/fsaris/home-assistant-awox/) integration, which does currently timeout when connecting.

What I noticed but cannot really try in Hassos (Cant disable the experimental features of BlueZ [Config in HassOs](https://github.com/home-assistant/operating-system/blob/bfb665bb85d07c23d1ba1fdc1d93f2bba476425e/buildroot-external/rootfs-overlay/etc/bluetooth/main.conf#L2) ), is that when enabling these I get errors when connecting.

Tried locally on my linux laptop
bluetoothctl connect XXXXX
Failed to connect: org.bluez.Error.Failed le-connection-abort-by-local

Disabling these makes the errors go away completly.
Also speed is much better

To note
I used the same Awox Eglo Lamp when trying this.
On booth, Hassos via the Shell and on the laptop I get this same error.

What would be nice
A small list of things to do to disable this Feature to try the integration above.
As HassOS is readonly I cannot edit the main.conf at all.

Issues with this problem
https://github.com/fsaris/home-assistant-awox/issues/72

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Have a paired EGLO Lamp with the Awox APP
2. Try with HassOs Shell and bluetoothctl to connect to the lamp
3. 
...

### Anything in the Supervisor logs that might be useful for us?

```txt
See linked issue!
```

### Anything in the Host logs that might be useful for us?

```txt
See linked issue!
```

### System information

## System Information

version | core-2022.12.8
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.80
arch | x86_64
timezone | Europe/Vienna
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4881
Installed Version | 1.29.0
Stage | running
Available Repositories | 1199
Downloaded Repositories | 19

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.4
-- | --
update_channel | beta
supervisor_version | supervisor-2022.11.2
agent_version | 1.4.1
docker_version | 20.10.19
disk_total | 457.7 GB
disk_used | 9.2 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.5.0), Samba share (10.0.0), Zigbee2MQTT (1.28.4-1), Mosquitto broker (6.1.3), Let's Encrypt (4.12.7), SSH & Web Terminal (13.0.0), Log Viewer (0.14.0), Vaultwarden (Bitwarden) (0.18.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 13
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 23 December 2022 at 18:13
-- | --
current_recorder_run | 28 December 2022 at 08:26
estimated_db_size | 286.38 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

HassOS runs on a MSI GE60 Apache2
The bluetooth adapter works fine with other devices.

agners, 2022-12-28:
Experimental APIs have been enabled in HAOS 9.4, before it was disabled. You can downgrade relatively easily to test the behavior without the APIs enabled by downgrading:

```
ha os update --version 9.3
```

Honestly, I don't think that the experimental APIs influence `connect` behavior. This is more likely a problem with the software otherwise (different firmware maybe?) or hardware (HAOS running on different hardware, different setup which causes interference or similar.. Try to use an extension cable and put the Bluetooth dongle as far away as possible from USB 3.0 ports/devices).

Chaosflo, 2022-12-28:
Sry, you misunderstood me.

I tried connecting on booth to the same device.
Booth get the same error messages.

Thing is I can disable it on the non HassOs device which I used for testing. There it works flawlessly.

I will test the downgrade and will report back soon!

Thanks

Chaosflo, 2022-12-28:
So I downgraded to 9.3 as you said.
Interestingly, the sucessrate is now much higher.

But you are right the timeout doesn't go away completly
So this doesnt make really a change. 

Could you say me the time range when these fixes will be released? #2285

Thank u =)

agners, 2022-12-28:
Probably mid January. You can also run the dev version (as mentioned in that issue report), the current dev is not that far from 9.x and quite stable.

closed 2023-04-04 by github-actions[bot]