home-assistant/operating-system #2291: OS 9.3 to 9.4 upgrade breaks Pi-hole itegration

MaxVonEvil, 2022-12-28:
### Describe the issue you are experiencing

When upgrading HAOS from 9.3 to 9.4, the Pi-Hole integration is partially broken, specifically all Pi-hole entities become unavailable, with the exception of 3 upgrade.* entities.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install and configure Pi-hole integration on OS 9.3 or earlier (will need actual pi-hole and it's API key)
2. Set up integration, go to the device, verify all entities are alive and kicking
3. Upgrade to OS 9.4, go back to pi-hole device's entities. Verify they are now Unavailable

Alternative, if pi-hole already installed, upgrade to OS 9.4 the remove Pi-hole integration then re-add.

### Anything in the Supervisor logs that might be useful for us?

```txt
Two interesting entries

1) =================================

Logger: homeassistant.components.binary_sensor
Source: components/pi_hole/sensor.py:71
Integration: Binary sensor (documentation, issues)
First occurred: 11:24:54 AM (1 occurrences)
Last logged: 11:24:54 AM

pi_hole: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/pi_hole/sensor.py", line 69, in native_value
    return round(self.api.data[self.entity_description.key], 2)
TypeError: list indices must be integers or slices, not str

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 493, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 715, in async_device_update
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 359, in async_update
    await self.coordinator.async_request_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 158, in async_request_refresh
    await self._debounced_refresh.async_call()
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 82, in async_call
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 110, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 348, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 545, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 583, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 551, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 669, in state
    value = self.native_value
  File "/usr/src/homeassistant/homeassistant/components/pi_hole/sensor.py", line 71, in native_value
    return self.api.data[self.entity_description.key]
TypeError: list indices must be integers or slices, not str

2) ========================

Logger: homeassistant.components.switch
Source: components/pi_hole/switch.py:76
Integration: Switch (documentation, issues)
First occurred: 11:24:54 AM (2 occurrences)
Last logged: 11:24:54 AM

Error adding entities for domain switch with platform pi_hole
Error while setting up pi_hole platform for switch
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 428, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 692, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 789, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 545, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 583, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 551, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 982, in state
    if (is_on := self.is_on) is None:
  File "/usr/src/homeassistant/homeassistant/components/pi_hole/switch.py", line 76, in is_on
    return self.api.data.get("status") == "enabled"  # type: ignore[no-any-return]
AttributeError: 'list' object has no attribute 'get'
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing appears relevant.
```

### System information

## System Information

version | core-2022.11.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.74
arch | x86_64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4901
Installed Version | 1.28.4
Stage | running
Available Repositories | 1201
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.11.2
agent_version | 1.4.1
docker_version | 20.10.18
disk_total | 30.8 GB
disk_used | 14.7 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.4.1), Samba share (10.0.0), Terminal & SSH (9.6.1), Duck DNS (1.15.0), Node-RED (13.5.3), Nginx Proxy Manager (0.12.3), MariaDB (2.5.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 8
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | November 27, 2022 at 8:40 PM
-- | --
current_recorder_run | November 27, 2022 at 12:40 PM
estimated_db_size | 868.21 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Additional information

One of the entities which are broken:
[image]

All the broken entities. Notice the update.* entities at the bottom still seem to work:
[image]

Mosher23, 2022-12-29:
I experience same issue

bcutter, 2023-01-29:
I dont think this is related to HA OS. There has been a change in Pi-Hole itself in terms of authentication. HA Core 2023.1 was the first version adapting to that change and fixing the unavailable entities. Check that first.

closed 2023-05-06 by github-actions[bot]