home-assistant/operating-system #2297: Bluetooth keeps throwing errors, disconnecting devices

Dvalin21, 2023-01-05:
### Describe the issue you are experiencing

Bluetooth devices keep disconnecting. Never reconnecting until the BT Adapter is disconnected and then reconnected after the VM is restarted. 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

In order to connect back to the devices (for the 5 minutes they are connected)
I have to disconnect the BT Adapter from the VM
Restart the HAOS
Reconnect BT Adapter back to VM
Then everything is connected for a short period of time.

### Anything in the Supervisor logs that might be useful for us?

```txt
23-01-05 12:57:30 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
23-01-05 12:57:30 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
23-01-05 12:57:30 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
23-01-05 12:57:30 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
23-01-05 12:57:30 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
23-01-05 12:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
23-01-05 12:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
23-01-05 12:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.NO_CURRENT_BACKUP/ContextType.SYSTEM
23-01-05 12:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
23-01-05 12:57:31 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-01-05 12:57:31 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
23-01-05 12:57:31 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-01-05 12:57:31 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
23-01-05 12:57:31 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-01-05 12:59:22 INFO (MainThread) [supervisor.backups.manager] Found 24 backup files
23-01-05 12:59:37 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-01-05 12:59:56 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
23-01-05 12:59:56 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
23-01-05 13:00:59 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-01-05 13:31:02 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.NO_CURRENT_BACKUP/ContextType.SYSTEM
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
23-01-05 13:57:31 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-01-05 13:59:29 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
23-01-05 13:59:29 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/thomasmauerer/hassio-addons repository
23-01-05 13:59:29 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
23-01-05 13:59:29 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository
23-01-05 13:59:29 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
23-01-05 13:59:30 INFO (MainThread) [supervisor.store] Loading add-ons from store: 70 all - 0 new - 0 remove
23-01-05 13:59:30 INFO (MainThread) [supervisor.store] Loading add-ons from store: 70 all - 0 new - 0 remove
23-01-05 13:59:58 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
23-01-05 13:59:58 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
[... 239 lines cut ...]
2:40:15 PM – (ERROR) /usr/local/lib/python3.10/site-packages/somecomfort/client.py - message first occurred at January 4, 2023 at 10:04:44 PM and shows up 13 times
Connection Error occurred.
2:40:15 PM – (ERROR) /usr/local/lib/python3.10/site-packages/somecomfort/client.py - message first occurred at January 4, 2023 at 10:04:44 PM and shows up 13 times
Error fetching roku data: Invalid response from API: Timeout occurred while connecting to device
9:58:41 AM – (ERROR) Roku - message first occurred at 4:17:12 AM and shows up 2 times
[TVision HUB(x.x.x.x):8009] Failed to connect to service ServiceInfo(type='host', data=('x.x.x.x', 8009)), retrying in 5.0s
4:17:50 AM – (ERROR) /usr/local/lib/python3.10/site-packages/pychromecast/socket_client.py
[TVision HUB(x.x.x.x):8009] Heartbeat timeout, resetting connection
4:17:20 AM – (WARNING) /usr/local/lib/python3.10/site-packages/pychromecast/socket_client.py
Error doing job: Task was destroyed but it is pending!
2:59:46 AM – (ERROR) runner.py - message first occurred at 2:59:46 AM and shows up 2 times
Error fetching AP-91054500ACE0 data: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error)
1:36:38 AM – (ERROR) LED BLE - message first occurred at January 4, 2023 at 9:04:41 PM and shows up 3 times
Unexpected error while attempting USB reset of hci0: [Errno 19] No such device
1:35:59 AM – (ERROR) /usr/local/lib/python3.10/site-packages/bluetooth_auto_recovery/recover.py
AP-91054500ACE0: Device unexpectedly disconnected; RSSI: -54
1:35:53 AM – (WARNING) runner.py - message first occurred at January 4, 2023 at 5:11:24 PM and shows up 29 times
Failed to cast media http://x.x.x.x:8123/api/frigate/frigate/vod/event/1672891693.120824-vxnjsp/index.m3u8?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJmZjQ3ZDQxMzNlN2Q0OGUxOGU5Njc5OTdjNDI5N2QxYiIsInBhdGgiOiIvYXBpL2ZyaWdhdGUvZnJpZ2F0ZS92b2QvZXZlbnQvMTY3Mjg5MTY5My4xMjA4MjQtdnhuanNwL2luZGV4Lm0zdTgiLCJwYXJhbXMiOnt9LCJpYXQiOjE2NzI5MDM5MjksImV4cCI6MTY3Mjk5MDMyOX0.JCoWACx_O4Spe5IrGBUM4ETwPUQ0rxynUAV2agDtycI from internal_url (http://x.x.x.x:8123). Please make sure the URL is: Reachable from the cast device and either a publicly resolvable hostname or an IP address
1:32:25 AM – (ERROR) Google Cast - message first occurred at 1:32:14 AM and shows up 2 times
Detected integration that calls media_source.async_resolve_media without passing an entity_id. Please report issue for media_source using this method at homeassistant/components/media_source/__init__.py, line 147: report("calls media_source.async_resolve_media without [... cut ...]
1:29:59 AM – (WARNING) helpers/frame.py
Config entry '08:BE:AC:30:D6:60' for bluetooth integration not ready yet: Bluetooth adapter None with address 08:BE:AC:30:D6:60 not found; Retrying in background
January 4, 2023 at 4:59:37 PM – (WARNING) config_entries.py - message first occurred at January 4, 2023 at 4:59:25 PM and shows up 4 times
Setup of config is taking over 10 seconds.
January 4, 2023 at 4:58:59 PM – (WARNING) runner.py - message first occurred at January 4, 2023 at 4:58:58 PM and shows up 2 times

mcc05, 2023-01-14:
Same setup VM(Oracle Virtual Box 7) on Windows,  This is not just bluetooth devices it happens on my Sonoff Zigbee 3, usb device also. I have tried to downgrade to HAS OS 9.0 same problem but less frequent, gone back to 8.5 and appears to be rock solid.

Kaisarion, 2023-03-05:
can reproduce on latest.

Dvalin21, 2023-03-06:
> can reproduce on latest.

Same

Update of light.lorri_s_lamp is taking over 10 seconds
1:54:45 PM – (WARNING) helpers/entity.py - message first occurred at 12:33:44 PM and shows up 51 times
AP-91054500ACE0: Device unexpectedly disconnected; RSSI: -48
1:54:30 PM – (WARNING) runner.py - message first occurred at 12:33:48 PM and shows up 10 times
Connection: BleakError: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error)
1:54:15 PM – (ERROR) Yeelight bluetooth (custom integration) - message first occurred at 12:35:12 PM and shows up 19 times
Logger: homeassistant.components.light
Source: helpers/entity_platform.py:822
Integration: Light (documentation, issues)
First occurred: 12:41:34 PM (5 occurrences)
Last logged: 1:54:04 PM

Updating yeelight_bt light took longer than the scheduled update interval 0:00:30

It did however last for a full day or so before disconnecting. Thats a new record!

Dvalin21, 2023-06-05:
I'm going to close this as bluetooth has been running AMAZING!!!  Even with the Edimax bluetooth 5.

BoneheadFraggle, 2023-09-01:
> I'm going to close this as bluetooth has been running AMAZING!!! Even with the Edimax bluetooth 5.

I am seeing the same problems but with Proxmox. 
Bluetooth stops working after a few days. Restarting HA doesn't help, but restarting the VM does.
I never need to restart Proxmox or the host computer, so it seems related to HA Supervisor?

What did you do to solve the issue?

Dvalin21, 2023-09-01:
> I am seeing the same problems but with Proxmox. Bluetooth stops working after a few days. Restarting HA doesn't help, but restarting the VM does. I never need to restart Proxmox or the host [...]

I've actually been having the same issue again after this last update.

sairon, 2023-10-09:
When the issue happens, please share `dmesg` output from HAOS, and if the host system is Linux as well, please do share the same output from the host system too. Adding the `lsusb` output would help too, so we know what adapter exactly are you using.

When you're experiencing generic disconnects of USB devices, I have a suspicion whether something in the host system isn't interfering - generally it's a good idea to have the host system as plain as possible to avoid it. I'm not familiar with the Windows/Virtualbox USB stack though to give any hints, but a random search shows [it may happen](https://forums.virtualbox.org/viewtopic.php?p=520723#p520723).

BoneheadFraggle, 2023-10-09:
I have activated a push notification when the state changes on one of the devices that lose contact when Bluetooth disconnects. Hopefully, this will make it possible to get fresh logs next time.

Since the disconnects seems kind of random, does anyone know if it is possible to do an automation that runs dmesg and saves the output?

closed 2024-01-08 by Dvalin21