home-assistant/operating-system #2303: HA OS 9.4 won't boot with HDMI monitor display connected to Raspberry PI 4

benjwaechter, 2023-01-15:
### Describe the issue you are experiencing

When connecting a screen to my Raspberry PI 4, the system start booting and crashes after scanning for bus xhci_pci. Then system reboots and it's an infinite loop. Note that connecting a keyboard to USB 2 or 3 port doesn't produce reboot, only a display to USB C port (any of them). 
System logs during boot below :
>> Net:   eth0: ethernet@7d580000
>> PCIe BRCM:  link up, 5.0 Gbps x1 (SSC)
>> starting USB...
>> Bus xhci: Register 5000420 NbrPorts 5
>> Starting the controller
>> USB XHCI 1.00
>> scanning bus xhci_pci for devices... Unexpected XHCI event TRB, skipping...  (3affc6e0 00000000 01000000 01008401)
>> BUG at drivers/usb/host/xhci-ring.c:530/abort_td()!
>> BUG!

If I disconnect the monitor, the system boots properly.

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Connect a monitor display to Raspberry PI 4 with a SSD card containing HA OS 9.4
2. Power on the PI
3. See boot messages on screen
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No as superviser is not strated at this stage
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2023.1.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.76-v7l
arch | armv7l
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4961
Installed Version | 1.29.1
Stage | running
Available Repositories | 1264
Downloaded Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.12.1
agent_version | 1.4.1
docker_version | 20.10.19
disk_total | 58.0 GB
disk_used | 6.1 GB
healthy | true
supported | true
board | rpi4
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.4.2), Terminal & SSH (9.6.1), Node-RED (14.0.1), Z-Wave JS UI (1.4.2), motionEye (0.18.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 1
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 14, 2023 at 18:57
-- | --
current_recorder_run | January 15, 2023 at 11:33
estimated_db_size | 260.06 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

[image]
[image]
[image]

agners, 2023-01-17:
> Note that connecting a keyboard to USB 2 or 3 port doesn't produce reboot

So you are saying connecting a keyboard makes the system boot normally?

Do you have anything else connected on one of the USB ports?

benjwaechter, 2023-01-19:
Hi,

No, I'm saying that connecting a display screen on micro HDMI makes system to reboot.
The only peripheral connected is a keyboard on USB port.
If I unplug the display screen, the system boots normally.

Thanks

agners, 2023-01-19:
Can you try the 64-bit version?

It could be power related, what kind of power supply are you using?

benjwaechter, 2023-01-19:
My PI is 32-bit...
I use the power supllier supplied with the PI

benjwaechter, 2023-01-20:
Yes connecting a keyboard makes the system boot normally. But if I connect a screen display on micro HDMI, it fails...
So having a PI with a keyboard but without screen is useless :-)

agners, 2023-01-25:
> My PI is 32-bit...

All Raspberry Pi 4 support 32-bit or 64-bit.

Do you have a different monitor or HDMI cable you could try?

wbarber69, 2023-01-29:
sounds like a power issue.

benjwaechter, 2023-01-30:
> 

I've tried same HDMI cable, same monitor, same power supply with another PI, no issue. System boots properly...

wbarber69, 2023-01-30:
When you say same power supply, are you talking about the supply to the pi or the monitor?

benjwaechter, 2023-01-31:
I say same PI power supply.

closed 2023-05-08 by github-actions[bot]