home-assistant/operating-system #2306: Support for ACPI shutdown is missing

nazar-pc, 2023-01-17:
### Describe the issue you are experiencing

ACPI shutdown in VM doesn't work

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Create a VM with libvirt or other virtualization tool
2. Install Some Assistant OS
3. Click shutdown on VM itself
4. Observe nothing happening, while expectation is that OS will start shutting down gracefully, at least by default

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
Jan 17 16:30:11 home-assistant systemd-logind[284]: Power key pressed short.
Jan 17 16:30:19 home-assistant systemd-logind[284]: Power key pressed short.
```

Yes, I clicked twice. But nothing happens after that.
```

### System information

## System Information

version | core-2023.1.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.80
arch | x86_64
timezone | Europe/Kiev
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4992
Installed Version | 1.29.1
Stage | running
Available Repositories | 1196
Downloaded Repositories | 1

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.12.1
agent_version | 1.4.1
docker_version | 20.10.19
disk_total | 5.2 GB
disk_used | 3.0 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (13.0.1), File editor (5.4.2), SQLite Web (3.6.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 15 січня 2023 р. о 01:30
-- | --
current_recorder_run | 15 січня 2023 р. о 03:30
estimated_db_size | 20.55 MiB
database_engine | sqlite
database_version | 3.38.5

agners, 2023-01-17:
The `generic-x86-64` images are meant for actual x86-64 hardware. We had issues where accidental power button press would shutdown the HA instance, hence we set it to long-press only.

For virtual machines, please use the `ova` images. In those, ACPI shutdown is configured to be immediate.

closed 2023-01-17 by agners

nazar-pc, 2023-01-17:
Hm... Home Assistant OS is the first OS (from those I used) that behaves like this and cares whether it is physical or virtual installation. Is there a way to upgrade from "bare metal" to "KVM" version for me? I wasn't able to find it in the docs.

agners, 2023-01-17:
> Hm... Home Assistant OS is the first OS (from those I used) that behaves like this and cares whether it is physical or virtual installation.

Our build system allows that rather easily, so we decided to go that route at the beginning. It avoids guest agents to be installed unnecessarily on the `generic-x86-64` images. Creating a single image would be possible too, but at this point it would mean a lot of migration works to merge it.

> Is there a way to upgrade from "bare metal" to "KVM" version for me? I wasn't able to find it in the docs.

No, there is no support to cross-upgrade the image. However, you can rather easily create a full backup, download that and restore it on a new installation.

nazar-pc, 2023-01-17:
> It avoids guest agents to be installed unnecessarily on the generic-x86-64 images.

At least for KVM I don't think any guest agent is necessary. In fact `systemd-logind` already caught the event, but chose to ignore it.

> However, you can rather easily create a full backup, download that and restore it on a new installation.

Annoying, but at least not starting from scratch, thanks! BTW installation guide could benefit from warning that in case of VM user might want to prefer a different version of the base image.

agners, 2023-01-18:
> Annoying, but at least not starting from scratch, thanks! BTW installation guide could benefit from warning that in case of VM user might want to prefer a different version of the base image.

The guides should link the appropriate images (see https://www.home-assistant.io/installation/linux#install-home-assistant-operating-system). At what site/place did you end up choosing the wrong image?

nazar-pc, 2023-01-18:
I clicked on "Generic x86-64 (e.g. Intel NUC)". In every other case before with KVM (libvirt) I used generic x86-64 OS images.
It genuinely didn't occur to me that I need to click "Linux" because I don't install it under Linux (as a container with Docker/Podman/Systemd or directly), I dedicate the whole machine, even though virtual, for this purpose, hence I chose "Generic x86-64".

agners, 2023-01-18:
Yeah I see what you are saying and agree that this can be misunderstood. We should be more explicit that this is meant for bare metal only. We plan on revamp getting started guides soon.